TrunkRecorder/trunk-recorder #249: System modulation separation

wvt34525, 2019-07-21:
Im currently decoding two P25 systems. One is a phase 2 system the other phase one, both systems cover almost the same area of spectrum in the 770MHZ band, the P2 system uses qpsk while the P1 system uses fsk4 modulation. My problem is either system will at times select the wrong SDR dongle for voice decode as it sees an available dongle covering the freq it needs, resulting in garbled voice decodes. Is there a way to fix this? Currently the only way I can see to fix this problem is to use a separate computer for one of the systems which is how I have it setup now.
https://openmhz.com/system/comb

Thanks

gofaster, 2019-08-09:
If you are using separate dongles for the 2 systems, you can run another instance of trunk-recorder on the same machine.

robotastic, 2019-09-12:
Unfortunately the only way to do it right now is run 2 separate instances of Trunk Recorder. The good news is that you can do that on a single computer! Fixing that would take a pretty major re-write.

closed 2019-09-12 by robotastic