TrunkRecorder/trunk-recorder #251: Raspberry Pi 4?

rpdale, 2019-08-08:
I did a fresh install on a Pi4, even upgraded the firmware, and am failing on the very last install step. 

/home/pi/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h: In member function ‘virtual float gr::op25_repeater::gardner_costas_cc::get_freq_error()’:
/home/pi/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h:51:43: warning: no return statement in function returning non-void [-Wreturn-type]
       virtual float get_freq_error(void) {}
                                           ^
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/recorder.dir/build.make:63: CMakeFiles/recorder.dir/trunk-recorder/main.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:74: C

imagesafari, 2019-08-08:
I had to add swap to a rpi4 with only 1GB of ram, otherwise the build runs
out of memory.

rpdale, 2019-08-09:
Thanks - I've done zram and swap and no change. The entire process works fine until I do the very last "make" command.

warning: no return statement in function returning non-void [-Wreturn-type] virtual float get_freq_error(void) {}

robotastic, 2019-08-09:
That is just a Warning statement - it shouldn't stop compilation. Maybe
check a little higher up for an Error

imagesafari, 2019-08-09:
Agreed - the OOM error was not obvious when I was chasing it down. Make
sure you're properly adding swap.

closed 2019-08-09 by rpdale

imagesafari, 2019-08-09:
What was the resolution?

rpdale, 2019-08-09:
Multiple combinations of zram and swapfiles until I found one that works. Just don't ask me to duplicate it :)

imagesafari, 2019-08-09:
Cool.

All I did was create a swap file and turn it on, run the build, and then
turn off and remove the file. I didn’t bother trying to do it on a ram
drive or anything fancy.

rpdale, 2019-08-10:
Being a little OCD ;) I had to start from scratch to duplicate. I found just this was adequate.

https://www.tecmint.com/create-a-linux-swap-file/