TrunkRecorder/trunk-recorder #253: What am I learning from these error messages?

rpdale, 2019-08-10:
Monitoring MPSCS Tower 1-001: https://www.radioreference.com/apps/db/?sid=100

{
    "sources": [{
        "center": 857000000.0,
        "rate": 8000000.0,
        "error": 23300,
        "gain": 33,
        "driver": "osmosdr",
        "device": "rtl=00000001",
        "modulation": "qpsk",
        "digitalRecorders": 4
    }],
    "systems": [{
        "control_channels": [853475000],
        "type": "p25",
        "talkgroupsFile": "ChanList.csv",
	"shortName": "mpscs"
    }],
    "captureDir": "/home/pi/radio/"
}

[2019-08-10 00:03:48.846062] (info)   
-------------------------------------
SYSTEMS
-------------------------------------
0
[2019-08-10 00:03:48.846184] (info)   

System Number: 1
-------------------------------------
1
[2019-08-10 00:03:48.846233] (info)   Short Name: mpscs
[2019-08-10 00:03:48.846262] (info)   System Type: p25
[2019-08-10 00:03:48.846293] (info)   Control Channels: 
[2019-08-10 00:03:48.846418] (info)   8.53475e+08 
[2019-08-10 00:03:48.846482] (info)   API Key: 
[2019-08-10 00:03:48.846510] (info)   Upload Script: 
[2019-08-10 00:03:48.846536] (info)   Call Log: true
[2019-08-10 00:03:48.846570] (info)   Audio Archive: true
[2019-08-10 00:03:48.846596] (info)   Loading Talkgroups...
[2019-08-10 00:03:48.865871] (info)   Read 1898 talkgroups.
[2019-08-10 00:03:48.865963] (info)   Talkgroups File: ChanList.csv
[2019-08-10 00:03:48.865997] (info)   Record Unknown Talkgroups: true
[2019-08-10 00:03:48.866050] (info)   Talkgroup Display Format: Id
[2019-08-10 00:03:48.866084] (info)   Hide Encrypted Talkgroups: false
[2019-08-10 00:03:48.866112] (info)   Hide Unknown Talkgroups: false
[2019-08-10 00:03:48.866139] (info)   Minimum Call Duration (in seconds): 0
[2019-08-10 00:03:48.866178] (info)   
[2019-08-10 00:03:48.866212] (info)   

-------------------------------------
SOURCES
-------------------------------------

[2019-08-10 00:03:48.866309] (info)   Driver: osmosdr
[2019-08-10 00:03:48.866336] (info)   Center: 8.570000e+08
[2019-08-10 00:03:48.866399] (info)   Rate: 8000000
[2019-08-10 00:03:48.866443] (info)   Error: 23300
[2019-08-10 00:03:48.866480] (info)   PPM Error: 0
[2019-08-10 00:03:48.866509] (info)   Gain: 33
[2019-08-10 00:03:48.866544] (info)   IF Gain: 0
[2019-08-10 00:03:48.866573] (info)   BB Gain: 0
[2019-08-10 00:03:48.866602] (info)   LNA Gain: 0
[2019-08-10 00:03:48.866630] (info)   PGA Gain: 0
[2019-08-10 00:03:48.866658] (info)   TIA Gain: 0
[2019-08-10 00:03:48.866686] (info)   MIX Gain: 0
[2019-08-10 00:03:48.866714] (info)   VGA1 Gain: 0
[2019-08-10 00:03:48.866742] (info)   VGA2 Gain: 0
[2019-08-10 00:03:48.866769] (info)   Squelch: 0
[2019-08-10 00:03:48.866797] (info)   Idle Silence: false
[2019-08-10 00:03:48.866823] (info)   Digital Recorders: 4
[2019-08-10 00:03:48.866855] (info)   Debug Recorders: 0
[2019-08-10 00:03:48.866880] (info)   SigMF Recorders: 0
[2019-08-10 00:03:48.866906] (info)   Analog Recorders: 0
[2019-08-10 00:03:48.866935] (info)   Modulation: qpsk
[2019-08-10 00:03:48.866972] (info)   Source Device: rtl=00000001
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp 
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Detached kernel driver
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[2019-08-10 00:03:49.567949] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2019-08-10 00:03:49.568110] (info)   Setting sample rate to: 8000000
Invalid sample rate: 8000000 Hz
[2019-08-10 00:03:49.568329] (info)   Actual sample rate: 1024000
[2019-08-10 00:03:49.568435] (info)   Tunning to 8.570233e+08
[2019-08-10 00:03:49.623959] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6 
[2019-08-10 00:03:49.624185] (info)   Max Freqency: 8.610000e+08
[2019-08-10 00:03:49.624328] (info)   Min Freqency: 8.530000e+08
[2019-08-10 00:03:49.649690] (info)   LNA Gain set to: 0
[2019-08-10 00:03:49.675638] (info)   TIA Gain set to: 0
[2019-08-10 00:03:49.700653] (info)   PGA Gain set to: 0
[2019-08-10 00:03:49.725029] (info)   Gain set to: 32.8
[2019-08-10 00:03:49.725164] (info)   Setting antenna to [RX]
[2019-08-10 00:03:49.747359] (info)   	 P25 Recorder Initial Rate: 500000 Resampled Rate: 31250 Initial Decimation: 16 Decimation: 16 System Rate: 30000 ARB Rate: 0.96
[2019-08-10 00:03:49.747514] (info)   P25 Recorder Taps - initial: 1213 channel: 1515 ARB: 757 Total: 3485
Project 25 IMBE Encoder/Decoder Fixed-Point implementation
Developed by Pavel Yazev E-mail: pyazev@gmail.com
Version 1.0 (c) Copyright 2009
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; see the file ``LICENSE'' for details.
[2019-08-10 00:03:49.777053] (info)   	 P25 Recorder Initial Rate: 500000 Resampled Rate: 31250 Initial Decimation: 16 Decimation: 16 System Rate: 30000 ARB Rate: 0.96
[2019-08-10 00:03:49.777259] (info)   P25 Recorder Taps - initial: 1213 channel: 1515 ARB: 757 Total: 3485
[2019-08-10 00:03:49.800236] (info)   	 P25 Recorder Initial Rate: 500000 Resampled Rate: 31250 Initial Decimation: 16 Decimation: 16 System Rate: 30000 ARB Rate: 0.96
[2019-08-10 00:03:49.800361] (info)   P25 Recorder Taps - initial: 1213 channel: 1515 ARB: 757 Total: 3485
[2019-08-10 00:03:49.825451] (info)   	 P25 Recorder Initial Rate: 500000 Resampled Rate: 31250 Initial Decimation: 16 Decimation: 16 System Rate: 30000 ARB Rate: 0.96
[2019-08-10 00:03:49.825549] (info)   P25 Recorder Taps - initial: 1213 channel: 1515 ARB: 757 Total: 3485
[2019-08-10 00:03:49.835755] (info)   
-------------------------------------

[2019-08-10 00:03:49.835825] (info)   

-------------------------------------
INSTANCE
-------------------------------------

[2019-08-10 00:03:49.835906] (info)   Capture Directory: /home/pi/radio
[2019-08-10 00:03:49.835936] (info)   Upload Server: 
[2019-08-10 00:03:49.835962] (info)   Status Server: 
[2019-08-10 00:03:49.835987] (info)   Instance Key: 
[2019-08-10 00:03:49.836012] (info)   Instance Id: 
[2019-08-10 00:03:49.836037] (info)   Default Mode: digital
[2019-08-10 00:03:49.836062] (info)   Call Timeout (seconds): 3
[2019-08-10 00:03:49.836089] (info)   Log to File: false
[2019-08-10 00:03:49.836116] (info)   Control channel warning rate: 10
[2019-08-10 00:03:49.836143] (info)   Control channel retune limit: 0
[2019-08-10 00:03:49.836172] (info)   Frequency format: 0
[2019-08-10 00:03:49.836197] (info)   Status as String: false
[2019-08-10 00:03:49.836235] (info)   [mpscs]	Started with Control Channel: 8.534750e+08
[2019-08-10 00:03:49.894887] (info)   	 P25 Trunking - SysNum: 0
[2019-08-10 00:03:49.894970] (info)   	 P25 Trunking Initial Rate: 500000 Resampled Rate: 50000 Initial Decimation: 16 Decimation: 10 System Rate: 48000 ARB Rate: 0.96
[2019-08-10 00:03:49.895201] (info)   P25 Trunking Taps - initial: 1213 channel: 1137 ARB: 757 Total: 3107
Allocating 15 zero-copy buffers
[2019-08-10 00:03:52.005621] (error)   [mpscs]	There is only one control channel defined
[2019-08-10 00:03:52.005780] (error)   [mpscs]	 Control Channel Message Decode Rate: 0.333333/sec, count:  1
[2019-08-10 00:03:55.002754] (error)   [mpscs]	There is only one control channel defined
[2019-08-10 00:03:55.002904] (error)   [mpscs]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2019-08-10 00:03:58.004963] (error)   [mpscs]	There is only one control channel defined
[2019-08-10 00:03:58.005122] (error)   [mpscs]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2019-08-10 00:04:01.007708] (error)   [mpscs]	There is only one control channel defined
[2019-08-10 00:04:01.007870] (error)   [mpscs]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2019-08-10 00:04:04.003635] (error)   [mpscs]	There is only one control channel defined
[2019-08-10 00:04:04.003791] (error)   [mpscs]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2019-08-10 00:04:07.009689] (error)   [mpscs]	There is only one control channel defined
[2019-08-10 00:04:07.009803] (error)   [mpscs]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2019-08-10 00:04:10.002099] (error)   [mpscs]	There is only one control channel defined
[2019-08-10 00:04:10.002365] (error)   [mpscs]	 Control Channel Message Decode Rate: 1/sec, count:  3

rpdale, 2019-08-10:
I let it run all night - nothing was recorded, and scrolling back I would occasionally see:

[2019-08-10 08:09:36.007553] (info)   Currently Active Calls: 0
[2019-08-10 08:09:36.007762] (info)   Recorders: 
[2019-08-10 08:09:36.007965] (info)   [ rtl=00000001 ]  
[2019-08-10 08:09:36.008171] (info)   [ D0 ] State: 2
[2019-08-10 08:09:36.008382] (info)   [ D1 ] State: 2
[2019-08-10 08:09:36.008588] (info)   [ D2 ] State: 2
[2019-08-10 08:09:36.008790] (info)   [ D3 ] State: 2

rpdale, 2019-08-11:
Running gqrx led me a different error number - I'm using 1000 now. No change. Am I missing something?

[image]

mattames, 2019-08-11:
What type of dongle is it? If the offset keeps changing then it is probably
thermal drift in the XO.

Cheers,
Matt

frameshift18, 2019-08-11:
That site has 700mhz freqs too so youre going to need two rtl dongles to be able to hear all the voice traffic.

Your center freq for the 800mhz channels looks good, although centered would be 852.3625

If you are using a rtl sdr the max rate you can use without problems is 2.4mhz, not 8mhz as you have set.

Looks like trunk-recorder is setting the sample rate to 1.024mhz cause the 8mhz is incorrect, then its setting the center freq to ~857.02mhz which is out of the range of your control channel at 853.475 when using the fallback 1.024mhz rate.

I would set the center to 852.3625mhz, set rate to 2.4mhz. See if that works.

The signal strength on gqrx doesnt look that high either, in my experience trunk-recorder needs a fairly good SNR to decode.

Which sdr are you using?

rpdale, 2019-08-11:
Thanks all. Using RTL-SDR v3. I edited the center and rate to no avail. I swapped out for a NooElec R820T, ran gqrx to find the new error, and ran it again with the same outcome.

Using these dongles I can hear the system fairly well (not perfect) running Unitrunker on Windows but I am in a basement - I'll move the setup outside later today where I have a clear shot to the tower and see if that changes anything.

rpdale, 2019-08-11:
I'm showing the output of Unitrunker on the Windows 10 box that is sitting right next to my RPI. All I did was unplug the RTLSDRv3 from the RPI and plug it in to Windows. Should I change any settings?

{
    "sources": [{
        "center": 852362500.0,
        "rate": 2400000.0,
        "error": 1875,
        "gain": 30,
        "driver": "osmosdr",
        "device": "rtl",
        "modulation": "qpsk",
        "digitalRecorders": 4
    }],
    "systems": [{
        "control_channels": [853475000],
        "type": "p25",
        "talkgroupsFile": "ChanList.csv",
	"shortName": "mpscs"
    }],
    "captureDir": "/home/pi/radio/"
}

[image]

rpdale, 2019-08-11:
For the fun of it, I installed op25 on this RPI4. Audio sounds weird, but it appears to be tracking it just fine?

./rx.py --args 'rtl' -N 'LNA:47' -S 2400000 -f 853.4750e6 -o 25000 -q -2 -T trunk.tsv -V -2 -U 2> stderr.2
[image]

robotastic, 2019-08-11:
The only thing could think of changing is the error, try -1875

Also - maybe try fsk4 for the modulation instead.

gofaster, 2019-08-11:
Actually it looks like your tuned frequency is slightly off the signal (high). 
If you want, install [boatbod's fork](https://github.com/boatbod/op25) of op25.  It has a nice web interface and you can flip between different plots to help optimize.

If required, set the -q (ppm) value to get close.  Then use the keyboard keys to fine tune.
Look at the rate of TSBKs (trunk signalling blocks)  and the various plots...mixer centered; constellation, datascope and symbol nice & tight.  Transfer your "fine tune" error value (hz) to rx.py's -d parameter and run it again. 
Iterate until things looks as good as possible, then transfer the -q parameter's value to trunk-recorder's 'ppm' entry for that source, and the fine tune value to the 'error' entry.

[op25 rx.py command line arguments](https://github.com/boatbod/op25/blob/master/op25/gr-op25_repeater/apps/README)

rpdale, 2019-08-11:
robotastic is the winner :) fsk4 worked. Here are a few files:

http://skywatch.org/11/

I'm almost fully operational now. But my TGIDs are not showing up anywhere, here is a sample of my ChanList.csv if anyone can find an issue with the file.

1006,1,D,MSP1,MSP District 1,Law Talk,State Police  Statewide
1007,1,D,MSP3,MSP District 3,Law Talk,State Police  Statewide
1008,1,D,STATW1,Statewide District  1,Interop,MPSCS Statewide
1009,1,D,STATW3,Statewide District  3,Interop,MPSCS Statewide

closed 2019-08-11 by rpdale