TrunkRecorder/trunk-recorder #255: My Needs

dh47, 2019-08-12:
I have been looking high and low for a system, that allows me to record a radio frequency, and then be able to scrobble back live to listen to past transmissions.  Is this what I'm looking for or must I look elsewhere?  If so can someone point me in the right direction?  Thanks.

bctrainers, 2019-08-12:
Hi @dh47, yeah - TrunkRecorder will do the recording part. It records just about all types of common digital trunking systems and analog systems. It will not record audio that is encrypted of any type.

You will need to check out [TrunkPlayer](https://github.com/ScanOC/trunk-player) if you're looking for a robust Recorder+Playback combo.

dh47, 2019-08-12:
Could you show me a simple config file I can use for an analog system, monitoring 1 frequency?

dh47, 2019-08-13:
@bctrainers hi, I'm having issues trying to set up a few analog systems.  I'm not sure what to put for type. 

My config
`{
    "sources": [{
        "center": 14143750,
        "rate": 2048000,
        "error": 1000,
        "gain": 39,
        "debugRecorders": 0,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=0"
    }, {
        "center": 486500000,
        "rate": 2048000,
        "error": 0,
        "gain": 39,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=1"
    }, {
        "center": 157950000,
        "rate": 2048000,
        "error": 0,
        "gain": 39,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=2"
    }],
    "systems": [{
        "type": "analog"
    }],
    "captureDir": "/home/pi/radio"
}`

dh47, 2019-08-13:
Not sure why I closed this, any help would be grateful.

gofaster, 2019-08-13:
Hello -

From the [README.md file](https://github.com/robotastic/trunk-recorder/blob/master/README.md):
"type - the type of trunking system. The options are smartnet, p25, conventional & conventionalP25."

There are example configs in the example subdirectory
[https://github.com/robotastic/trunk-recorder/tree/master/examples](https://github.com/robotastic/trunk-recorder/tree/master/examples)

dh47, 2019-08-14:
Doesn't look like support for just analog then, unfortunately, this will need suite my needs.

imagesafari, 2019-08-14:
Analog conventional is supported.

dh47, 2019-08-14:
So I would put conventional for the type?

closed 2019-08-14 by dh47