TrunkRecorder/trunk-recorder #257: Not Receiving Anything

dh47, 2019-08-14:
I don't seem to be picking anything up.  There doesn't seem to be any errors so I'm not sure.
`{
    "sources": [{
        "center": 1414000000,
        "rate": 2048000,
        "error": 0,
		"squelch": -50,
        "gain": 49,
        "debugRecorders": 0,
        "digitalRecorders": 4,
		"modulation": "qpsk",
        "driver": "osmosdr",
		"device": "rtl=0"
    }],
    "systems": [
        {
            "type": "conventional",
            "channels": [1414000000],
            "shortName": "EAS",
            "callLog": true,
            "audioArchive": true
}
],
  "captureDir": "/home/pi/radio/"
}`

`[2019-08-14 09:58:54.958252] (info)   Capture Directory: /home/pi/radio
[2019-08-14 09:58:54.958294] (info)   Upload Server:
[2019-08-14 09:58:54.958328] (info)   Status Server:
[2019-08-14 09:58:54.958362] (info)   Instance Key:
[2019-08-14 09:58:54.958395] (info)   Instance Id:
[2019-08-14 09:58:54.958427] (info)   Default Mode: digital
[2019-08-14 09:58:54.958461] (info)   Call Timeout (seconds): 3
[2019-08-14 09:58:54.958497] (info)   Log to File: false
[2019-08-14 09:58:54.958531] (info)   Control channel warning rate: 10
[2019-08-14 09:58:54.958567] (info)   Control channel retune limit: 0
[2019-08-14 09:58:54.958604] (info)   Frequency format: 0
[2019-08-14 09:58:54.958638] (info)   Status as String: false
[2019-08-14 09:58:54.958693] (info)   [EAS]     Monitoring Conventional Channel: 1.414000e+09 Talkgroup: 1
[2019-08-14 09:58:54.984126] (info)   Analog Recorder Taps - initial: 311 channel: 2327 ARB: 775 Total: 3413
[2019-08-14 09:58:54.987870] (info)   [EAS]     TG: 1   Freq: 1.414000e+09      Listening on Src: rtl=0
Allocating 15 zero-copy buffers
[2019-08-14 09:58:53.880083] (info)   Using Config file: config.json

[2019-08-14 09:58:53.881616] (info)
-------------------------------------
SYSTEMS
-------------------------------------
0
[2019-08-14 09:58:53.881783] (info)

System Number: 1
-------------------------------------
1
[2019-08-14 09:58:53.881842] (info)   Short Name: EAS
[2019-08-14 09:58:53.881889] (info)   System Type: conventional
[2019-08-14 09:58:53.881931] (info)   Conventional Channels:
[2019-08-14 09:58:53.882094] (info)   1.414e+09
[2019-08-14 09:58:53.882212] (info)   Alpha Tags:
[2019-08-14 09:58:53.882248] (info)   API Key:
[2019-08-14 09:58:53.882281] (info)   Upload Script:
[2019-08-14 09:58:53.882347] (info)   Call Log: true
[2019-08-14 09:58:53.882402] (info)   Audio Archive: true
[2019-08-14 09:58:53.882437] (info)   Loading Talkgroups...
[2019-08-14 09:58:53.882539] (error)   Error Opening TG File:

[2019-08-14 09:58:53.882584] (info)   Talkgroups File:
[2019-08-14 09:58:53.882618] (info)   Record Unknown Talkgroups: true
[2019-08-14 09:58:53.882671] (info)   Talkgroup Display Format: Id
[2019-08-14 09:58:53.882713] (info)   Hide Encrypted Talkgroups: false
[2019-08-14 09:58:53.882749] (info)   Hide Unknown Talkgroups: false
[2019-08-14 09:58:53.882783] (info)   Minimum Call Duration (in seconds): 0
[2019-08-14 09:58:53.882829] (info)
[2019-08-14 09:58:53.882871] (info)

-------------------------------------
SOURCES
-------------------------------------

[2019-08-14 09:58:53.883024] (info)   Driver: osmosdr
[2019-08-14 09:58:53.883063] (info)   Center: 1.414000e+09
[2019-08-14 09:58:53.883169] (info)   Rate: 2048000
[2019-08-14 09:58:53.883235] (info)   Error: 0
[2019-08-14 09:58:53.883351] (info)   PPM Error: 0
[2019-08-14 09:58:53.883392] (info)   Gain: 49
[2019-08-14 09:58:53.883441] (info)   IF Gain: 0
[2019-08-14 09:58:53.883480] (info)   BB Gain: 0
[2019-08-14 09:58:53.883519] (info)   LNA Gain: 0
[2019-08-14 09:58:53.883554] (info)   PGA Gain: 0
[2019-08-14 09:58:53.883590] (info)   TIA Gain: 0
[2019-08-14 09:58:53.883625] (info)   MIX Gain: 0
[2019-08-14 09:58:53.883661] (info)   VGA1 Gain: 0
[2019-08-14 09:58:53.883697] (info)   VGA2 Gain: 0
[2019-08-14 09:58:53.883733] (info)   Squelch: -50
[2019-08-14 09:58:53.883781] (info)   Idle Silence: false
[2019-08-14 09:58:53.883815] (info)   Digital Recorders: 4
[2019-08-14 09:58:53.883861] (info)   Debug Recorders: 0
[2019-08-14 09:58:53.883900] (info)   SigMF Recorders: 0
[2019-08-14 09:58:53.883933] (info)   Analog Recorders: 0
[2019-08-14 09:58:53.883971] (info)   Modulation: qpsk
[2019-08-14 09:58:53.884039] (info)   Source Device: rtl=0
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[2019-08-14 09:58:54.474761] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2019-08-14 09:58:54.474941] (info)   Setting sample rate to: 2048000
[R82XX] PLL not locked!
[2019-08-14 09:58:54.590515] (info)   Actual sample rate: 2048000
[2019-08-14 09:58:54.590765] (info)   Tunning to 1.414000e+09
[2019-08-14 09:58:54.641855] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6
[2019-08-14 09:58:54.642027] (info)   Max Freqency: 1.415024e+09
[2019-08-14 09:58:54.642158] (info)   Min Freqency: 1.412976e+09
[2019-08-14 09:58:54.667611] (info)   LNA Gain set to: 0
[2019-08-14 09:58:54.692905] (info)   TIA Gain set to: 0
[2019-08-14 09:58:54.718064] (info)   PGA Gain set to: 0
[2019-08-14 09:58:54.743342] (info)   Gain set to: 49.6
[2019-08-14 09:58:54.743561] (info)   Setting antenna to [RX]
[2019-08-14 09:58:54.800892] (info)      P25 Recorder Initial Rate: 512000 Resampled Rate: 30117.6 Initial Decimation: 4 Decimation: 17 System Rate: 30000 ARB Rate: 0.996094
[2019-08-14 09:58:54.801177] (info)   P25 Recorder Taps - initial: 311 channel: 1551 ARB: 731 Total: 2593
Pr`

frameshift18, 2019-08-14:
"center": 1414000000, is 1.414Ghz, do you have an extra 0? are you recording 2m analog amateur radio?

dh47, 2019-08-14:
Ops, it's meant to be 141.400 MHz.

frameshift18, 2019-08-14:
If you want to record analog conventional you need to use "analogRecorders" instead of "digitalRecorders" and don't put the modulation line. Since you only have 1 freq in the channels array you just need 1 Recorder.

dh47, 2019-08-14:
What 141.400 MHz be for the config? 1414000?

frameshift18, 2019-08-14:
what you had minus a 0, so 141400000, its in hertz

frameshift18, 2019-08-14:
Also fair warning the analog recording on this is a little finicky. Your'e mostly likely gonna have to mess with your squelch setting or you'll get a bunch of random static recordings. Also the analog recorder works a little different than the digital so its going to act like its constantly recording but when the squelch breaks and there is traffic it will just save the audio after the squelch broke. Also the timestamps on the analog recordings seem to be offset by one recording.

dh47, 2019-08-14:
Roger that.

dh47, 2019-08-14:
What squelch setting do you recommend?

frameshift18, 2019-08-14:
It all depends on your gain and strength of the signal you're wanting to receive.  Roughly between -30 and -60 but it varies. You'll just have to mess with it.

dh47, 2019-08-14:
Ok now I have added another dongle and freq, the second channel is working but not the first is picking up nothing but static.  The squelch for both is set, am I missing something?

dh47, 2019-08-14:
{
    "sources": [{
        "center": 141400000,
        "rate": 2048000,
        "error": 0,
		"squelch": -50,
        "gain": 49,
        "debugRecorders": 0,
        "analogRecorders": 4,
        "driver": "osmosdr",
		"device": "rtl=0"
    },{
        "center": 486500000,
        "rate": 2048000,
        "error": 0,
		"squelch": -50,
        "gain": 49,
        "analogRecorders": 4,
        "driver": "osmosdr",
        "device": "rtl=1"
    }],
    "systems": [
        {
            "type": "conventional",
            "channels": [141400000, 486500000],
            "shortName": "FTC",
            "callLog": true,
            "audioArchive": true
}
],
  "captureDir": "/home/pi/radio/"
}

closed 2019-08-15 by dh47