TrunkRecorder/trunk-recorder #258: Digital Recorders

rpdale, 2019-08-14:
I'm getting some occasionally high CPU usage now that I have two SDRs and the resulting audio file sounds really bad. Is that due to having multiple SDRs or too many recoders just pushing the RPI4 past its limit? I'm not clear if it's the overall digital recorder count, or count divided by SDRs

 Should I add another SDR and reduce the digital recorder count per SDR or would that not help?

imagesafari, 2019-08-14:
I'd say you're probably hitting the limit of # of recorders. Are you
recording a trunk system? How many recorders are typically active?

I've got a rpi4 running 7 conventional recorders, and that's about all
it'll handle.

rpdale, 2019-08-14:
Thanks - I have 9 recorders now. I'll drop that down to 7, but the issue is that the three 700MHz channels are always full. I'll need to mess with the csv priorities I think... (Un)fortunately the tower nearby is where the main radio programming office also resides, so those freq's are FULL of random stuff from around the state :)

frameshift18, 2019-08-14:
I'd also look at your sdr bandwidth and lower it if possible. Less sdr bandwidth is less decimation each recorder has to do.

rpdale, 2019-08-14:
Where would that be?

{
    "sources": [{
        "center": 852362500.0,
        "rate": 2400000.0,
        "error": 1000,
        "gain": 47,
        "driver": "osmosdr",
        "device": "rtl=1",
        "modulation": "fsk4",
        "digitalRecorders": 4
    },
{
        "center": 770500000.0,
        "rate": 2400000.0,
        "error": -32000,
        "gain": 47,
        "driver": "osmosdr",
        "device": "rtl=2",
        "modulation": "fsk4",
        "digitalRecorders": 3
    }],

frameshift18, 2019-08-14:
rate. but just make sure the center freq +/- half the rate cover what you
need.

imagesafari, 2019-08-14:
Yeah, you can edit the talkgroup file to lower priority or skip talkgroups
you don't care about.

rpdale, 2019-08-15:
With 4 digital recorders on 800Mhz SDR and 3 on the 700MHz, I have this in my ChanList.csv file:

11906,1,D,23EATRAN,Eaton County Transportation Authority,Transportation,Eaton County 23,20

Am I doing something wrong because it's still being recorded, even though I never have 20 recorders available?

frameshift18, 2019-08-15:
I have that issue too. Could try and set it to E instead of D maybe trunk recorder would skip over it in that case? I've been meaning to try that as a workaround.

gofaster, 2019-08-15:
Set priority greater than 99 to disable recording for that talkgroup.

UnderwaterHockey, 2019-08-15:
I use:
 "recordunknown":false 
in my systems and only include the talkgroups I want to record in my tg list.

imagesafari, 2019-08-16:
Set to 100.

rpdale, 2019-08-19:
Setting > 100 seemed to have done the trick - thanks!

And for framshift18 - I had to make the rate larger than just the 1/2. But that seemed to work too...

closed 2019-08-19 by rpdale