TrunkRecorder/trunk-recorder #262: gr::log :DEBUG: correlate_access_code_tag_bb0

bradbenton, 2019-08-21:
I now have it up & running, and I'm capturing some recordings. However, I'm getting a LOT of these messages in the output. Over 20 per second.

gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 204197
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 204281
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 204365
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 204449
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 204533

classicyuppie, 2019-09-15:
I'm getting the same thing...

mdozier9, 2019-10-04:
What SDR device are you using? Post your config as well.

bradbenton, 2019-10-04:
Thanks mdozier. I've got 2 NESDR Smart (https://www.amazon.com/gp/product/B01HA642SW) running on an R Pi 4 with Raspian Buster.

Config
```
{
    "sources": [
        {
            "center": 489600000.0,
            "rate": 2048000.0,
            "error": 0,
            "gain": 40,
            "digitalRecorders": 3,
            "modulation": "fsk4",
            "driver": "osmosdr",
            "device": "rtl=00000001"
        }, {
            "center": 482737000.0,
            "rate": 2048000.0,
            "error": 0,
            "gain": 40,
            "digitalRecorders": 3,
            "modulation": "fsk4",
            "driver": "osmosdr",
            "device": "rtl=00000002"
        }
    ],
    "systems": [{
        "control_channels": [488700000,489075000],
        "type": "smartnet",
        "bandplan": "400_custom",
        "bandplanBase": 482000000,
        "bandplanHigh": 490937500,
        "bandplanSpacing": 12500,
        "bandplanOffset": 380,
        "talkgroupsFile": "ChanList.csv",
        "shortName": "MERA"
    }],
    "captureDir": "/home/pi/trunk-recorder/capture/"
}
```

As I noted, I do capture recordings. I know there is at least one additional problem with custom 400 bandplan for this system, but I thought I'd resolve that after I get the log to quiet down a bit.

mdozier9, 2019-10-04:
On issue #267, this error seemed to go away after the rate was set from 2000000.0 to 2048000.  I would avoid the decimal point in the config (rate and center) as well and enter "2048000" with no decimal points.  Perhaps that could be causing the issue?  Also, to confirm your RTL serial numbers are 00000001 and 00000002 (you can confirm with rtl_test)?

bradbenton, 2019-10-04:
Sadly, dropping the decimal did not help in this case.

`[2019-10-04 14:26:43.816027] (info)   Using Config file: ./config.json

[2019-10-04 14:26:43.817071] (info)
-------------------------------------
SYSTEMS
-------------------------------------
0
[2019-10-04 14:26:43.817196] (info)

System Number: 1
-------------------------------------
1
[2019-10-04 14:26:43.817247] (info)   Short Name: MERA
[2019-10-04 14:26:43.817277] (info)   System Type: smartnet
[2019-10-04 14:26:43.817310] (info)   Control Channels:
[2019-10-04 14:26:43.817433] (info)   4.887e+08
[2019-10-04 14:26:43.817530] (info)   4.89075e+08
[2019-10-04 14:26:43.817569] (info)   API Key:
[2019-10-04 14:26:43.817595] (info)   Upload Script:
[2019-10-04 14:26:43.817621] (info)   Call Log: true
[2019-10-04 14:26:43.817655] (info)   Audio Archive: true
[2019-10-04 14:26:43.817682] (info)   Loading Talkgroups...
[2019-10-04 14:26:43.819714] (info)   Read 174 talkgroups.
[2019-10-04 14:26:43.819771] (info)   Talkgroups File: ChanList.csv
[2019-10-04 14:26:43.819800] (info)   Record Unknown Talkgroups: true
[2019-10-04 14:26:43.819840] (info)   Talkgroup Display Format: Id
[2019-10-04 14:26:43.819902] (info)   Smartnet bandplan: 400_custom
[2019-10-04 14:26:43.819928] (info)   Smartnet band: 400
[2019-10-04 14:26:43.819953] (info)   Smartnet bandplan base freq: 482
[2019-10-04 14:26:43.819989] (info)   Smartnet bandplan high freq: 490.938
[2019-10-04 14:26:43.820019] (info)   Smartnet bandplan spacing: 0.0125
[2019-10-04 14:26:43.820048] (info)   Smartnet bandplan offset: 380
[2019-10-04 14:26:43.820073] (info)   Hide Encrypted Talkgroups: false
[2019-10-04 14:26:43.820100] (info)   Hide Unknown Talkgroups: false
[2019-10-04 14:26:43.820125] (info)   Minimum Call Duration (in seconds): 0
[2019-10-04 14:26:43.820154] (info)
[2019-10-04 14:26:43.820185] (info)

-------------------------------------
SOURCES
-------------------------------------

[2019-10-04 14:26:43.820266] (info)   Driver: osmosdr
[2019-10-04 14:26:43.820292] (info)   Center: 4.896000e+08
[2019-10-04 14:26:43.820351] (info)   Rate: 2048000
[2019-10-04 14:26:43.820395] (info)   Error: 0
[2019-10-04 14:26:43.820430] (info)   PPM Error: 0
[2019-10-04 14:26:43.820458] (info)   Gain: 40
[2019-10-04 14:26:43.820492] (info)   IF Gain: 0
[2019-10-04 14:26:43.820520] (info)   BB Gain: 0
[2019-10-04 14:26:43.820548] (info)   LNA Gain: 0
[2019-10-04 14:26:43.820575] (info)   PGA Gain: 0
[2019-10-04 14:26:43.820603] (info)   TIA Gain: 0
[2019-10-04 14:26:43.820630] (info)   MIX Gain: 0
[2019-10-04 14:26:43.820657] (info)   VGA1 Gain: 0
[2019-10-04 14:26:43.820685] (info)   VGA2 Gain: 0
[2019-10-04 14:26:43.820712] (info)   Squelch: 0
[2019-10-04 14:26:43.820739] (info)   Idle Silence: false
[2019-10-04 14:26:43.820764] (info)   Digital Recorders: 3
[2019-10-04 14:26:43.820795] (info)   Debug Recorders: 0
[2019-10-04 14:26:43.820820] (info)   SigMF Recorders: 0
[2019-10-04 14:26:43.820846] (info)   Analog Recorders: 0
[2019-10-04 14:26:43.820876] (info)   Modulation: fsk4
[2019-10-04 14:26:43.820913] (info)   Source Device: rtl=00000001
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfsp                                                                       ace airspy airspyhf soapy redpitaya freesrp
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[... 98 lines cut ...]
[2019-10-04 14:26:45.859666] (info)   Control channel: 4.887000e+08
[2019-10-04 14:26:45.873042] (info)      SmartNet Recorder Initial Rate: 512000                                                                        Resampled Rate: 36571.4 Initial Decimation: 4 Decimation: 14 System Rate: 36000                                                                        ARB Rate: 0.984375
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: ac
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ff
Allocating 15 zero-copy buffers
Allocating 15 zero-copy buffers
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 99
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 183
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 267
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 351
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 435
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 519
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 603
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 687
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 771
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 855
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 939
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1023
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1064
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1107
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1191
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1275
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1316
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1359
`

mdozier9, 2019-10-05:
rtl_test output does not look good!  It is showing lots of loss.  Ideally, you want no output at all.  I would expect a Pi4 to be able to handle the rtl_test without issue.   Make sure your RTL dongles are either directly connected to the Pi, or on a good quality powered hub.  Also, USB cables can cause issues too.  For testing purposes, eliminate any USB cables and plug directly in to the Pi.  Those are good quality SDRs, so I doubt the dongle is an issue.  I assume you have already done a rpi-update to update the pi's firmware too.  ...not sure what is going on there, but I suspect loss in the USB communication.

bradbenton, 2019-10-05:
dongles are connected directly to the Pi. I tried moving them to two different USB ports and thought we were on to something as the number of lost bytes on rtl_test went way down. but testing a little more and moving them back to the original ports now shows identical results. I'm not sure why yesterday's rtl_test was so bad, but here's where I am today:

```
pi@raspberrypi4:~ $ rtl_test
Found 2 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000001
  1:  Realtek, RTL2838UHIDIR, SN: 00000002

Using device 0: Generic RTL2832U OEM
Detached kernel driver
Found Rafael Micro R820T tuner
Supported gain values (29): 0.0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28.0 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48.0 49.6
[R82XX] PLL not locked!
Sampling at 2048000 S/s.

Info: This tool will continuously read from the device, and report if
samples get lost. If you observe no further output, everything is fine.

Reading samples in async mode...
Allocating 15 zero-copy buffers
lost at least 2172 bytes
lost at least 276 bytes
^CSignal caught, exiting!

User cancel, exiting...
Samples per million lost (minimum): 18
Reattached kernel driver
pi@raspberrypi4:~ $
```
rpi-update had be done as part of the install. (this is a fresh R Pi 4 just for this project). But I ran it again and did have a new update to apply. also did apt update/upgrade to bring everything current again.

trunk-recorder results are still the same, however:

```
pi@raspberrypi4:~/trunk-recorder $ ./recorder
[2019-10-05 09:37:13.196773] (info)   Using Config file: ./config.json

[2019-10-05 09:37:13.197781] (info)
-------------------------------------
SYSTEMS
-------------------------------------
0
[2019-10-05 09:37:13.197900] (info)

System Number: 1
-------------------------------------
1
[2019-10-05 09:37:13.197948] (info)   Short Name: MERA
[2019-10-05 09:37:13.197976] (info)   System Type: smartnet
[2019-10-05 09:37:13.198012] (info)   Control Channels:
[2019-10-05 09:37:13.198129] (info)   4.887e+08
[2019-10-05 09:37:13.198223] (info)   4.89075e+08
[2019-10-05 09:37:13.198262] (info)   API Key:
[2019-10-05 09:37:13.198289] (info)   Upload Script:
[2019-10-05 09:37:13.198315] (info)   Call Log: true
[2019-10-05 09:37:13.198351] (info)   Audio Archive: true
[2019-10-05 09:37:13.198378] (info)   Loading Talkgroups...
[2019-10-05 09:37:13.200311] (info)   Read 174 talkgroups.
[2019-10-05 09:37:13.200365] (info)   Talkgroups File: ChanList.csv
[2019-10-05 09:37:13.200395] (info)   Record Unknown Talkgroups: true
[2019-10-05 09:37:13.200436] (info)   Talkgroup Display Format: Id
[2019-10-05 09:37:13.200497] (info)   Smartnet bandplan: 400_custom
[2019-10-05 09:37:13.200523] (info)   Smartnet band: 400
[2019-10-05 09:37:13.200553] (info)   Smartnet bandplan base freq: 482
[2019-10-05 09:37:13.200589] (info)   Smartnet bandplan high freq: 490.938
[2019-10-05 09:37:13.200620] (info)   Smartnet bandplan spacing: 0.0125
[2019-10-05 09:37:13.200651] (info)   Smartnet bandplan offset: 380
[2019-10-05 09:37:13.200679] (info)   Hide Encrypted Talkgroups: false
[2019-10-05 09:37:13.200706] (info)   Hide Unknown Talkgroups: false
[2019-10-05 09:37:13.200732] (info)   Minimum Call Duration (in seconds): 0
[... 163 lines cut ...]

gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1175
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1259
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1343
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1427
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1511
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1595
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1679
[2019-10-05 09:37:15.940513] (info)   Patching Command: 340 Freq: 0.000000e+00 Talkgroup: 6960

gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1752
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1763
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1847
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1931
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 1969
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 2015
[2019-10-05 09:37:15.991254] (info)   Patching Command: 340 Freq: 0.000000e+00 Talkgroup: 6960

^C
 Signal caught!
[2019-10-05 09:37:16.092212] (info)   stopping flow graph
pi@raspberrypi4:~/trunk-recorder $
```

one other thing I tried was to take the decimals off the center freq as well. didn't make a difference.

mdozier9, 2019-10-06:
Well, I'm out of ideas here.  Since this appears to be a gnu radio debug output, you might find some info in the gnu radio forums on what it means, and how to to possibly turn off debug for that module.    I'm not too familiar with the smartnet stuff...all mine are either P25, or conventional.  It does look like you are getting successful control channel decoding, as it's seeing talk groups.  Are you getting recordings dropped in your media location?

bradbenton, 2019-10-06:
no worries. thanks for the help.

i found reference to a similar problem on another unrelated project. the developer there indicated it was normal and could be overlooked. i'd be fine doing that but was minimally hoping to turn off the debug to make the output easier to view. but i'm not familiar with gnu radio and when stepping through the code I have yet to locate where to set that debug level. i don't see that there's anything in the trunk-radio config to facilitate that.

i am decoding the control channel and i am generating recordings. the one issue I have with that, however, is that with 400 custom system i'm using there's a split bandplan with 3 different channel ranges. i don't think trunk-recorder can handle that currently (at least based on the config options). i should be able to add that to the code, but here again was waiting to see if i could make things a little less noisy before jumping in to that.

gofaster, 2019-10-06:
This file? 
/usr/local/etc/gnuradio/conf.d/gnuradio-runtime.conf

# This file contains system wide configuration data for GNU Radio.
# You may override any setting on a per-user basis by editing
# ~/.gnuradio/config.conf
...
[LOG]
# Levels can be (case insensitive):
#       DEBUG, INFO, WARN, TRACE, ERROR, ALERT, CRIT, FATAL, EMERG
log_level = debug
debug_level = emerg
...

bradbenton, 2019-10-07:
ooohhh. yeah, that's the file. i was thinking that trunk-recorder was responsible for how GR was configured, instantiating it as an internal object or something. it didn't occur to me that GR might still read it's own config, but sure enough changing log_level to info there instead of debug does the trick for quieting down the output. thanks a bunch.

closed 2021-04-28 by robotastic