TrunkRecorder/trunk-recorder #265: Shared libraries error when running recorder (error 0x22)

classicyuppie, 2019-09-10:
Originally when running `./recorder`, I was getting seg faults, however, after a few restarts of my raspi 4 running Buster, I managed to get a more cogent error message:

```
pi@raspberrypi:~/trunk-build $ ./recorder
./recorder: error while loading shared libraries: /usr/lib/arm-linux-gnueabihf/libboost_unit_test_framework.so.1.67.0: unexpected reloc type 0x22
```
I've tried adding `export LD_LIBRARY_PATH="/usr/lib/"` to the .bashrc file and restarting, but that doesn't resolve the issue. Can't seam to figure this out and I'm at an impasse on actually running this software.

Please help.

robotastic, 2019-09-12:
I still haven't gotten my hands on a RasPi 4. My guess is that this is related to Buster, but I am not sure. Hopefully someone else can chime in.

rpdale, 2019-09-12:
I followed your directions on RasPi4 with Buster and am running smoothly... Just a stab - but the only time I've ever had RPI issues were due to SD cards going (gone) bad.

classicyuppie, 2019-09-15:
I'm going to try and recompile everything from scratch after reinstalling Buster. I'll let you know how I make out.

raygalasso, 2019-09-15:
Did you try a different SD card? Because I've had the same thing happen with a bad one.

raygalasso, 2019-09-15:
I've had the same thing happen with a bad SD card.  I know it's brand new, but it might still be bad. Did you try a different one?

classicyuppie, 2019-09-15:
@raygalasso I'm trying a fresh install and recompile of everything now. If that still produces the same error, I'll try the new SD card I purchased today. Thanks for the input!

classicyuppie, 2019-09-15:
So I'm getting closer. Just completed the reinstall of Buster and a recompile of all the necessary components and I'm getting this error:

```

pi@raspberrypi:~/trunk-build $ ./recorder
[2019-09-15 04:01:50.342042] (info)   Using Config file: ./config.json

[2019-09-15 04:01:50.353202] (error)   Failed parsing Config: ./config.json: cannot open file
[2019-09-15 04:01:50.353292] (error)   Unable to setup a System to record, exiting...
```

closed 2019-09-15 by classicyuppie