TrunkRecorder/trunk-recorder #266: Unable to Upload CSV to OpenMHZ

jkole13261, 2019-09-14:
Anybody else having an issue uploading their CSV file in admin side of openmhz?

classicyuppie, 2019-09-15:
Yup. Tried in multiple browsers, even one that has no plugins or extensions installed. No bueno.

photounit, 2020-02-08:
I have had this issue too and I was stumped. Everything looked ok in Excell or calc on the csv file but it still wouldn't upload. What I found is I had extra spaces between fields that didn't show up when checking the file in Excell or calc. What I did is opened the file using a text editor and the errors were readily apparent. I just made the fixes and saved the file and all was well. This is what your file should look like....

2002,7D2,T,911 PollingW,Police West Response,Law Dispatch,Countywide Law Enforcement,1
2007,7D7,T,911 Polling,911 Priority,Law Dispatch,Countywide Law Enforcement,1
3001,BB9,T,EMS Resp1,EMS Response 1,EMS Dispatch,Countywide EMS,1
3002,BBA,T,EMS Resp2,EMS Response 2,EMS Dispatch,Countywide EMS,1
3003,BBB,T,EMS 4,EMS 4,EMS-Tac,Countywide EMS,2
3004,BBC,T,EMS 5,EMS 5,EMS-Tac,Countywide EMS,2
3005,BBD,T,EMS 6,EMS 6,EMS-Tac,Countywide EMS,2
3006,BBE,T,EMS 7,EMS 7,EMS-Tac,Countywide EMS,2

mattmac553, 2020-05-01:
Try with the HEX column removed, and it may import.

photounit, 2020-05-26:
Yeah, I forgot to remove the hex column in my example here but the actual problem in my case was the spaces. But the hex column will probably cause you issues too....

amateur-rad, 2020-05-26:
I have been trying to upload a CSV file for quite some time... it has never worked.  Even after checking there were no spaces, and trying different browsers.

dmonies327, 2020-06-09:
I got this to work after A) making sure the columns were exact, with no header and B) using MS Edge.
Did not work in chrome.

closed 2020-07-31 by robotastic