TrunkRecorder/trunk-recorder #267: Unable to setup a System to record, exiting...

classicyuppie, 2019-09-15:
I'm getting an `Unable to setup a System to record, exiting...` error when running my Trunk Recorder config. Here is my config and output when I run `./recorder`:

### Config

```
{
    "sources": [

        {
            "center": 852340000,
            "rate": 2000000.0,
	    "error": 0,
	    "squelch": -48.9,
	    "analogRecorders": 4,
            "modulation": "fsk4",
	    "fskGain": "0.80",
            "analogLevels": 8,
            "driver": "osmosdr",
	    "device": "rtl=0"
	}

    ],
    "systems": [
	{
            "control_channels": [853762500,851887500,852262500,852375000],
            "type": "smartnet",
	    "bandplan": "800_reband",
       	    "talkgroupsFile": "montcopa.csv",
	    "recordUnknown": false,
            "shortName": "montcopa",
	    "apiKey": "..."
    	}
    ],
    "uploadServer": "https://api.openmhz.com",
    "captureDir": "/home/pi/trunk-build/montcopa"
}
```

### Output
```
pi@raspberrypi:~/trunk-build $ ./recorder
[2019-09-15 22:56:25.470713] (info)   Using Config file: ./config.json

[2019-09-15 22:56:25.471707] (info)
-------------------------------------
SYSTEMS
-------------------------------------
0
[2019-09-15 22:56:25.471823] (info)

System Number: 1
-------------------------------------
1
[2019-09-15 22:56:25.471885] (info)   Short Name: montcopa
[2019-09-15 22:56:25.471915] (info)   System Type: smartnet
[2019-09-15 22:56:25.471947] (info)   Control Channels:
[2019-09-15 22:56:25.472087] (info)   8.53762e+08
[2019-09-15 22:56:25.472185] (info)   8.51888e+08
[2019-09-15 22:56:25.472228] (info)   8.52262e+08
[2019-09-15 22:56:25.472278] (info)   8.52375e+08
[2019-09-15 22:56:25.472323] (info)   API Key: ...
[2019-09-15 22:56:25.472359] (info)   Upload Script:
[2019-09-15 22:56:25.472386] (info)   Call Log: true
[2019-09-15 22:56:25.472430] (info)   Audio Archive: true
[2019-09-15 22:56:25.472458] (info)   Loading Talkgroups...
[2019-09-15 22:56:25.472883] (error)   Malformed talkgroup entry at line 19.
[2019-09-15 22:56:25.472955] (error)   Malformed talkgroup entry at line 20.
[2019-09-15 22:56:25.473421] (error)   Malformed talkgroup entry at line 57.
[2019-09-15 22:56:25.473523] (error)   Malformed talkgroup entry at line 62.
[2019-09-15 22:56:25.473617] (error)   Malformed talkgroup entry at line 66.
[2019-09-15 22:56:25.473938] (error)   Warning: skipped 5 of 89 talkgroup entries! Invalid format?
[2019-09-15 22:56:25.473969] (error)   The format is very particular. See https://github.com/robotastic/trunk-recorder for example input.
[2019-09-15 22:56:25.474006] (info)   Talkgroups File: montcopa.csv
[2019-09-15 22:56:25.474066] (info)   Record Unknown Talkgroups: false
[2019-09-15 22:56:25.474114] (info)   Talkgroup Display Format: Id
[2019-09-15 22:56:25.474146] (info)   Smartnet bandplan: 800_reband
[2019-09-15 22:56:25.474217] (info)   Smartnet band: 800
[2019-09-15 22:56:25.474244] (info)   Hide Encrypted Talkgroups: false
[2019-09-15 22:56:25.474279] (info)   Hide Unknown Talkgroups: false
[2019-09-15 22:56:25.474305] (info)   Minimum Call Duration (in seconds): 0
[2019-09-15 22:56:25.474349] (info)
[2019-09-15 22:56:25.474418] (info)

-------------------------------------
SOURCES
-------------------------------------

[2019-09-15 22:56:25.474540] (info)   Driver: osmosdr
[2019-09-15 22:56:25.474599] (info)   Center: 8.523400e+08
[2019-09-15 22:56:25.474670] (info)   Rate: 2000000
[2019-09-15 22:56:25.474715] (info)   Error: 0
[2019-09-15 22:56:25.474758] (info)   PPM Error: 0
[2019-09-15 22:56:25.474787] (info)   Gain: 0
[2019-09-15 22:56:25.474824] (info)   IF Gain: 0
[2019-09-15 22:56:25.474875] (info)   BB Gain: 0
[2019-09-15 22:56:25.474905] (info)   LNA Gain: 0
[2019-09-15 22:56:25.474942] (info)   PGA Gain: 0
[2019-09-15 22:56:25.474971] (info)   TIA Gain: 0
[2019-09-15 22:56:25.475007] (info)   MIX Gain: 0
[2019-09-15 22:56:25.475036] (info)   VGA1 Gain: 0
[2019-09-15 22:56:25.475072] (info)   VGA2 Gain: 0
[2019-09-15 22:56:25.475124] (info)   Squelch: -48.9
[2019-09-15 22:56:25.475172] (info)   Idle Silence: false
[2019-09-15 22:56:25.475207] (info)   Digital Recorders: 0
[2019-09-15 22:56:25.475235] (info)   Debug Recorders: 0
[2019-09-15 22:56:25.475270] (info)   SigMF Recorders: 0
[2019-09-15 22:56:25.475295] (info)   Analog Recorders: 4
[2019-09-15 22:56:25.475338] (info)   Modulation: fsk4
[2019-09-15 22:56:25.475383] (info)   Source Device: rtl=0
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
Using device #0 Realtek RTL2838UHIDIR SN: 100
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[2019-09-15 22:56:26.057847] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2019-09-15 22:56:26.057972] (info)   Setting sample rate to: 2000000
Exact sample rate is: 2000000.052982 Hz
[R82XX] PLL not locked!
[2019-09-15 22:56:26.172702] (info)   Actual sample rate: 2000000
[2019-09-15 22:56:26.172842] (info)   Tunning to 8.523400e+08
[2019-09-15 22:56:26.223122] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6
[2019-09-15 22:56:26.223230] (info)   Max Freqency: 8.533400e+08
[2019-09-15 22:56:26.223332] (info)   Min Freqency: 8.513400e+08
[2019-09-15 22:56:26.248368] (info)   LNA Gain set to: 0
[2019-09-15 22:56:26.273298] (info)   TIA Gain set to: 0
[2019-09-15 22:56:26.298288] (info)   PGA Gain set to: 0
[2019-09-15 22:56:26.298390] (info)   Gain set to: 0
[2019-09-15 22:56:26.298492] (info)   Setting antenna to [RX]
[2019-09-15 22:56:47.418447] (info)   Analog Recorder Taps - initial: 303 channel: 2273 ARB: 757 Total: 3333
[2019-09-15 22:56:47.474049] (info)   Analog Recorder Taps - initial: 303 channel: 2273 ARB: 757 Total: 3333
[2019-09-15 22:56:47.529454] (info)   Analog Recorder Taps - initial: 303 channel: 2273 ARB: 757 Total: 3333
[2019-09-15 22:56:47.584461] (info)   Analog Recorder Taps - initial: 303 channel: 2273 ARB: 757 Total: 3333
[2019-09-15 22:56:47.586288] (info)
-------------------------------------

[2019-09-15 22:56:47.586339] (info)

-------------------------------------
INSTANCE
-------------------------------------

[2019-09-15 22:56:47.586417] (info)   Capture Directory: /home/pi/trunk-build/montcopa
[2019-09-15 22:56:47.586449] (info)   Upload Server: https://api.openmhz.com
[2019-09-15 22:56:47.586490] (info)   Status Server:
[2019-09-15 22:56:47.586517] (info)   Instance Key:
[2019-09-15 22:56:47.586542] (info)   Instance Id:
[2019-09-15 22:56:47.586578] (info)   Default Mode: digital
[2019-09-15 22:56:47.586605] (info)   Call Timeout (seconds): 3
[2019-09-15 22:56:47.586679] (info)   Log to File: false
[2019-09-15 22:56:47.586718] (info)   Control channel warning rate: 10
[2019-09-15 22:56:47.586755] (info)   Control channel retune limit: 0
[2019-09-15 22:56:47.586783] (info)   Frequency format: 0
[2019-09-15 22:56:47.586809] (info)   Status as String: false
[2019-09-15 22:56:47.586890] (info)   [montcopa]        Started with Control Channel: 8.537625e+08
[2019-09-15 22:56:47.586972] (error)   Unable to setup a System to record, exiting...
```

For reference, my talkgroups are listed on RadioReference at https://www.radioreference.com/apps/db/?sid=724&opt=all_tg#tgs. Not sure what it is I'm missing.

mdozier9, 2019-09-15:
It looks like your first control channel is outside the range of your RTL.  You might need to add a second RTL to cover your control channel range.

classicyuppie, 2019-09-15:
I was able to "solve" the error by adjusting the center channel to 852800000. However, I am now getting a bunch of errors as follows and no recordings:

```
[2019-09-15 23:27:55.007168] (info)   Pll Phase: 4.1148 min Freq: -3.141593e-01 Max Freq: 3.141593e-01
[2019-09-15 23:27:55.007323] (error)   [montcopa] Retuning to Control Channel: 8.537625e+08
[2019-09-15 23:27:55.007371] (info)      - System Source - Min Freq: 8.518000e+08 Max Freq: 8.538000e+08
[2019-09-15 23:27:55.007501] (info)   Offset set to: 962500 Freq: 8.537625e+08

[2019-09-15 23:27:55.007564] (error)   [montcopa]        Control Channel Message Decode Rate: 0/sec, count:  0
[2019-09-15 23:27:58.002616] (info)   Pll Phase: 5.67599 min Freq: -3.141593e-01 Max Freq: 3.141593e-01
[2019-09-15 23:27:58.002823] (error)   [montcopa] Retuning to Control Channel: 8.518875e+08
[2019-09-15 23:27:58.002896] (info)      - System Source - Min Freq: 8.518000e+08 Max Freq: 8.538000e+08
[2019-09-15 23:27:58.003126] (info)   Offset set to: -912500 Freq: 8.518875e+08

[2019-09-15 23:27:58.003204] (error)   [montcopa]        Control Channel Message Decode Rate: 0/sec, count:  0
[2019-09-15 23:28:01.001135] (info)   Pll Phase: 4.85253 min Freq: -3.141593e-01 Max Freq: 3.141593e-01
[2019-09-15 23:28:01.001427] (error)   [montcopa] Retuning to Control Channel: 8.522625e+08
[2019-09-15 23:28:01.001548] (info)      - System Source - Min Freq: 8.518000e+08 Max Freq: 8.538000e+08
[2019-09-15 23:28:01.001823] (info)   Offset set to: -537500 Freq: 8.522625e+08

[2019-09-15 23:28:01.001940] (error)   [montcopa]        Control Channel Message Decode Rate: 0/sec, count:  0
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 320213
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 320293
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 320729
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 320806
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 321057
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 321235
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 321689
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 321784
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 321797
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 321981
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 322336
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 322859
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 323145
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 323390
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 323723
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 323795
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 323837
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 324195
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 325024
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 325631
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 325780
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 326554
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 327086
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 327142
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 327159
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 327198
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 327279
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 327436
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 328291
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 328533
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 328550
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 328701
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 328925
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 329136
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 329156
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 329406
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 329646
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 329916
[2019-09-15 23:28:04.000749] (info)   Pll Phase: 5.77497 min Freq: -3.141593e-01 Max Freq: 3.141593e-01
[2019-09-15 23:28:04.000919] (error)   [montcopa] Retuning to Control Channel: 8.523750e+08
[2019-09-15 23:28:04.000971] (info)      - System Source - Min Freq: 8.518000e+08 Max Freq: 8.538000e+08
[2019-09-15 23:28:04.001137] (info)   Offset set to: -425000 Freq: 8.523750e+08
```

mdozier9, 2019-09-15:
Try using a rate of 2048000.  Most RTL dongles don't go much higher than that.  Sometimes you can push them to 3.2k, but I find the best performance at 2.048k.

classicyuppie, 2019-09-15:
@mdozier9 -- set, but no joy. Still ending up with errors. FWIW, I'm seeing a `PLL not locked!` error in the output as follows, with the same errors as indicate above:

```
Using device #0 Realtek RTL2838UHIDIR SN: 100
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[2019-09-15 23:40:19.808828] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2019-09-15 23:40:19.808949] (info)   Setting sample rate to: 2048000
[R82XX] PLL not locked!
[2019-09-15 23:40:19.923270] (info)   Actual sample rate: 2048000
[2019-09-15 23:40:19.923397] (info)   Tunning to 8.527980e+08
[2019-09-15 23:40:19.973528] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6
[2019-09-15 23:40:19.973634] (info)   Max Freqency: 8.538240e+08
[2019-09-15 23:40:19.973735] (info)   Min Freqency: 8.517760e+08
[2019-09-15 23:40:19.998659] (info)   LNA Gain set to: 0
[2019-09-15 23:40:20.023539] (info)   TIA Gain set to: 0
[2019-09-15 23:40:20.048425] (info)   PGA Gain set to: 0
[2019-09-15 23:40:20.073289] (info)   Gain set to: 49.6
[2019-09-15 23:40:20.073410] (info)   Setting antenna to [RX]
[2019-09-15 23:40:20.126406] (info)   Analog Recorder Taps - initial: 311 channel: 2327 ARB: 775 Total: 3413
[2019-09-15 23:40:20.205336] (info)   Analog Recorder Taps - initial: 311 channel: 2327 ARB: 775 Total: 3413
[2019-09-15 23:40:20.260295] (info)   Analog Recorder Taps - initial: 311 channel: 2327 ARB: 775 Total: 3413
[2019-09-15 23:40:20.313851] (info)   Analog Recorder Taps - initial: 311 channel: 2327 ARB: 775 Total: 3413
[2019-09-15 23:40:20.315673] (info)
```

mdozier9, 2019-09-15:
PLL not locked is a trivial message, and can be ignored.  I guess I should have asked, which SDR product are you using?  You can go over to the Osmo SDR page to see what the supported rates are.  But based on your last output, things are looking much better.  Have you validated that you are getting a good signal from from RTL/feedline/antenna?  You can stop trunk recorder, then start up a rtl_tcp session and check it in SDR# or GQRX.

classicyuppie, 2019-09-15:
My environment is Raspian Buster using an RTL-SDR R820T. I have two (one on the Raspi and another on my Win10 machine). The Win10 SDR receives without issue in SDR Console and I can hear traffic over the Fire Dispatch channel (for example) clearly. Everything is exactly the same between machines, except for OS.

mdozier9, 2019-09-15:
I'm not very familiar with SmartNet setups, but I just looked up that system on Radioreference, and I think the audio is both analog and P25 (System Type: | Motorola Type II SmartZone-- | --Analog and APCO-25 Common Air Interface).  I'm not sure if that makes a difference, but perhaps setting the type to P25 might produce something?   I also see the note about some sites being rebranded, but you might play around with those settings too.  If it's the smartnet side you are after, I'm afraid I don't have much experience decoding that.

classicyuppie, 2019-09-15:
@mdozier9 I didn't even notice that it was both analog and P25. I set the type to P25, and I'm getting different output now:

```
-------------------------------------
INSTANCE
-------------------------------------

[2019-09-16 00:06:57.132721] (info)   Capture Directory: /home/pi/trunk-build/montcopa
[2019-09-16 00:06:57.132775] (info)   Upload Server: https://api.openmhz.com
[2019-09-16 00:06:57.132822] (info)   Status Server:
[2019-09-16 00:06:57.132869] (info)   Instance Key:
[2019-09-16 00:06:57.132914] (info)   Instance Id:
[2019-09-16 00:06:57.132959] (info)   Default Mode: digital
[2019-09-16 00:06:57.133005] (info)   Call Timeout (seconds): 3
[2019-09-16 00:06:57.133051] (info)   Log to File: false
[2019-09-16 00:06:57.133097] (info)   Control channel warning rate: 10
[2019-09-16 00:06:57.133164] (info)   Control channel retune limit: 0
[2019-09-16 00:06:57.133220] (info)   Frequency format: 0
[2019-09-16 00:06:57.133270] (info)   Status as String: false
[2019-09-16 00:06:57.133333] (info)   [montcopa]        Started with Control Channel: 8.537625e+08
[2019-09-16 00:06:57.183581] (info)      P25 Trunking - SysNum: 0
[2019-09-16 00:06:57.183694] (info)      P25 Trunking Initial Rate: 512000 Resampled Rate: 51200 Initial Decimation: 4 Decimation: 10 System Rate: 48000 ARB Rate: 0.9375
[2019-09-16 00:06:57.183955] (info)   P25 Trunking Taps - initial: 311 channel: 1163 ARB: 775 Total: 2249
Allocating 15 zero-copy buffers
[2019-09-16 00:07:00.004149] (error)   [montcopa] Retuning to Control Channel: 8.518875e+08
[2019-09-16 00:07:00.004306] (info)      - System Source - Min Freq: 8.517760e+08 Max Freq: 8.538240e+08
[2019-09-16 00:07:00.004463] (error)   [montcopa]        Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2019-09-16 00:07:00.832242] (info)   [montcopa]        Decoding System ID 669 WACN: BEE00 NAC: 664
```

mdozier9, 2019-09-15:
This is a good sign: "Decoding System ID 669 WACN: BEE00 NAC: 664".  Now the million dollar question is are they using analog or digital!  Lol.  Your Pi is up and runing for P25 digital based on that output.  Don't forget to set your digitalRecorders to a reasonable number too.  One more thing is the modulation type...most all of my P25 systems use qpsk.  Hopefully you will pick up something, else you might need to play around with the analog stuff.

classicyuppie, 2019-09-15:
According to https://www.radioreference.com/apps/db/?sid=724, there's a bunch of channels that are broadcasting in Digital. Some channels like the county Event channels are in analog, but I'm not as concerned with those at the moment. I'm mainly concerned with the Fire Region East channels and Fireground East channels.

mdozier9, 2019-09-16:
Looking at Radioreference, it looks like you will need 3 RTL dongles (running at 2048000 rate).  851.2626 to 856.7375 is the range (about 5.5Mhz wide), and try not to put your desired channel at the very edge of the RTL envelope to avoid reduced sensitivity.  To keep things simple, I would center them as such: 852/854/856 (2048000 rate) for an effective coverage of 851-857.  With multiple RTLs, you will need to program the eeprom with serial numbers, and reference that serial number in each respective device statements in the config file.  You may also want to consider setting the buffers option.  Take a read through Luke's documentation for more info about using multiple RTLs.

All that said, using 3 RTLs is very likely going to stress a Raspberry Pi, both in terms of CPU, and USB power.  I doubt it will be able to keep up.  I've tried running Raspberry Pi trunkers before, and they work good for very small systems, but I'm pretty sure 3 RTLs will surpass the abilities of a Pi.  One option that I use in three of my setups is a HP T620 thin client ($35 off ebay) running Ubuntu.  They are quad-core, fanless, with a SSD.  And they have enough horsepower to handle multiple dongles (5 in one of mine) on a P25 system.  Basically, I look at them as a step up from a Pi, but still not a full-blown, energy-hogging PC.

JoeGilkey, 2020-01-14:
@classicyuppie did you get this to work? I’m wondering if the signals are using TPS (Tactical Public Safety). That’s analog audio with a P25 header. It’s Motorola’s solution you FDNY running out of available MDC-1200 id’s.

closed 2021-04-28 by robotastic