TrunkRecorder/trunk-recorder #268: Lengthy recordings with no audio

Pacmannion, 2019-09-18:
So after returning my feed online, I'm now getting recordings with actual length (not 0 second) however, OpenMHZ won't play them, similar to the 0 second recording bug.

UnderwaterHockey, 2019-09-18:
I have noticed this as well, calls with as much as 30s durations listed on the Openmhz page don't play, as if they were 0s recordings.  I haven't verified yet whether they play locally.

mattmac553, 2020-04-30:
I am having the same issue, it was working, I updated the TG file then when I ran it again no audio

[2020-04-29 21:37:25.533912] (info)   [jaxp25]	TG:       1113	Freq: 859.712524	Starting Recorder on Src: hackrf=266663
[2020-04-29 21:37:30.547215] (info)   [jaxp25]	TG:       1145	Freq: 859.462524	Not Recording: ENCRYPTED 
[2020-04-29 21:37:37.010629] (info)   [jaxp25]	TG:       1113	Freq: 859.712524	Ending Recorded Call - Last Update: 4s	Call Elapsed: 12
[2020-04-29 21:37:37.011187] (info)   	- Stopping P25 Recorder Num [0]	TG:       1113	Freq: 859.712524 	TDMA: false	Slot: 0
[2020-04-29 21:37:37.011541] (info)   [jaxp25]	Deleting this call as it has a duration less than minimum duration of 0	TG:       1113	Freq: 859.712524	Call Duration: 0s
[

it deletes some clips for being zero, I'm confused

closed 2020-08-04 by robotastic