TrunkRecorder/trunk-recorder #269: No Audio Uploading to OpenMhz

amateur-rad, 2019-09-22:
Hi, I am trying to monitor a convention single-frequency system with the following configuration, but it does not appear to be uploading to OpenMhz, and does not appear in the website's systems list.  I am also experiencing the inability to upload .csv files on the admin view, as mentioned in another thread.  Any ideas?

{
    "sources": [{
        "center": 158500000,
        "rate": 2048000,
        "error": 0,
        "gain": 28,
        "debugRecorders": 0,
        "digitalRecorders": 0,
        "squelch": -50,
        "driver": "osmosdr"
    }],
    "systems": [
        {
            "type": "conventional",
            "channels": [158500000],
            "shortName": "countyfire",
            "talkgroupsFile": "countyfire-conv.csv",
            "callLog": true,
            "audioArchive": true,
            "apiKey": "the correct apiKey"
}],
  "uploadServer": "https://api.openmhz.com"

}

mdozier9, 2019-09-22:
Are  you converting the .wav file to a .aac?  To do so, you need to compile FFMPEG allowing the "non-free" codec support.  If you attempt to send a .wav to OpenMhz, it will not work.  In your media directory you should see both .wav and .aac if your FFMPEG process is working.  Also, watching the output of TR would be helpful.  Are you seeing something like "upload successful" in the logs?  One off-topic more suggestion, I would recommend re-centering your RTL so that your desired channel frequency is not in the center.  Some RTLs will produce an undesireable artifact at center frequency that can be particularly interfering, especially to analog signals. Moving it just a few khz should prevent interference with your channel frequency.

amateur-rad, 2019-09-28:
Thanks - I had compiled FFMPEG, but it looks like there are only .wav files in the output.  No .aac.  And I do not see 'upload successful' while TR is running.  Do I need to do something/compile anything additional specific to allow the 'non-free' codec support?

Thanks for the suggestion on the RTL re-centering, I will tweak it once I get the upload issue resolved.

mdozier9, 2019-10-04:
When you compiled ffmpeg, you should have included the option, "--enable-nonfree". To validate, run ffmpeg -encoders and look for "AAC (Advanced Audio Coding)".  If it's not there, then you need to recompile.  Do not install the repo version (apt-get install, yum, deb, etc).  Those don't have the non-free encoders enabled by default, and hence will not work.

amateur-rad, 2020-04-08:
Still have not gotten this to work over the past six months.  Did a clean install of the Raspbian OS, and reinstalled Trunk Recorder.  Confirmed I have AAC, but am having the same issue - the system is not even appearing in the OpenMhz list.  Do I need to do something special to activate it in OpenMhz?

tdduarte, 2020-04-29:
I am having the same issue... I cant seem to get the AAC files to appear in the directory.. I rebuilt this thing 3 times following the directions.. The Wav files sound great just no AAC files.. Running on a Raspberry Pi 3 ... Any ideas what could be worng?

tdduarte, 2020-04-29:
pi@trunk-recorder:~ $ ffmpeg -encoders
ffmpeg version N-97508-g1cc3851b60 Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 8 (Raspbian 8.3.0-6+rpi1)
  configuration: --pkg-config-flags=--static --bindir=/home/pi/bin --enable-gpl --enable-libass --enable-libfdk-aac --enable-nonfree
  libavutil      56. 43.100 / 56. 43.100
  libavcodec     58. 82.100 / 58. 82.100
  libavformat    58. 42.101 / 58. 42.101
  libavdevice    58.  9.103 / 58.  9.103
  libavfilter     7. 79.100 /  7. 79.100
  libswscale      5.  6.101 /  5.  6.101
  libswresample   3.  6.100 /  3.  6.100
  libpostproc    55.  6.100 / 55.  6.100
Encoders:
 V..... = Video
 A..... = Audio
 S..... = Subtitle
 .F.... = Frame-level multithreading
 ..S... = Slice-level multithreading
 ...X.. = Codec is experimental
 ....B. = Supports draw_horiz_band
 .....D = Supports direct rendering method 1
 ------
 V..... a64multi             Multicolor charset for Commodore 64 (codec a64_multi)
 V..... a64multi5            Multicolor charset for Commodore 64, extended with 5th color (colram) (codec a64_multi5)
 V..... alias_pix            Alias/Wavefront PIX image
 V..... amv                  AMV Video
 V..... apng                 APNG (Animated Portable Network Graphics) image
 V..... asv1                 ASUS V1
 V..... asv2                 ASUS V2
 V..... avrp                 Avid 1:1 10-bit RGB Packer
 V..X.. avui                 Avid Meridien Uncompressed
 V..... ayuv                 Uncompressed packed MS 4:4:4:4
 V..... bmp                  BMP (Windows and OS/2 bitmap)
 V..... cinepak              Cinepak
 V..... cljr                 Cirrus Logic AccuPak
 V.S... vc2                  SMPTE VC-2 (codec dirac)
 VFS... dnxhd                VC3/DNxHD
 V..... dpx                  DPX (Digital Picture Exchange) image
 VFS... dvvideo              DV (Digital Video)
 V.S... ffv1                 FFmpeg video codec #1
 VF.... ffvhuff              Huffyuv FFmpeg variant
 V..... fits                 Flexible Image Transport System
 V..... flashsv              Flash Screen Video
 V..... flashsv2             Flash Screen Video Version 2
 V..... flv                  FLV / Sorenson Spark / Sorenson H.263 (Flash Video) (codec flv1)
 V..... gif                  GIF (Graphics Interchange Format)
 V..... h261                 H.261
 V..... h263                 H.263 / H.263-1996
 V..... h263_v4l2m2m         V4L2 mem2mem H.263 encoder wrapper (codec h263)
 V.S... h263p                H.263+ / H.263-1998 / H.263 version 2
 V..... h264_v4l2m2m         V4L2 mem2mem H.264 encoder wrapper (codec h264)
 V..... h264_vaapi           H.264/AVC (VAAPI) (codec h264)
 V..... hevc_v4l2m2m         V4L2 mem2mem HEVC encoder wrapper (codec hevc)
 V..... hevc_vaapi           H.265/HEVC (VAAPI) (codec hevc)
 VF.... huffyuv              Huffyuv / HuffYUV
 V..... jpeg2000             JPEG 2000
 VF.... jpegls               JPEG-LS
 VF.... ljpeg                Lossless JPEG
 VF.... magicyuv             MagicYUV video
 VFS... mjpeg                MJPEG (Motion JPEG)
 V..... mjpeg_vaapi          MJPEG (VAAPI) (codec mjpeg)
 V.S... mpeg1video           MPEG-1 video
 V.S... mpeg2video           MPEG-2 video
 V..... mpeg2_vaapi          MPEG-2 (VAAPI) (codec mpeg2video)
 V.S... mpeg4                MPEG-4 part 2
 V..... mpeg4_v4l2m2m        V4L2 mem2mem MPEG4 encoder wrapper (codec mpeg4)
 V..... msmpeg4v2            MPEG-4 part 2 Microsoft variant version 2
 V..... msmpeg4              MPEG-4 part 2 Microsoft variant version 3 (codec msmpeg4v3)
 V..... msvideo1             Microsoft Video-1
 V..... pam                  PAM (Portable AnyMap) image
[... 93 lines cut ...]
 A..... pcm_u32le            PCM unsigned 32-bit little-endian
 A..... pcm_u8               PCM unsigned 8-bit
 A..... pcm_vidc             PCM Archimedes VIDC
 A..... real_144             RealAudio 1.0 (14.4K) (codec ra_144)
 A..... roq_dpcm             id RoQ DPCM
 A..X.. s302m                SMPTE 302M
 A..... sbc                  SBC (low-complexity subband codec)
 A..X.. sonic                Sonic
 A..X.. sonicls              Sonic lossless
 A..X.. truehd               TrueHD
 A..... tta                  TTA (True Audio)
 A..X.. vorbis               Vorbis
 A..... wavpack              WavPack
 A..... wmav1                Windows Media Audio 1
 A..... wmav2                Windows Media Audio 2
 S..... ssa                  ASS (Advanced SubStation Alpha) subtitle (codec ass)
 S..... ass                  ASS (Advanced SubStation Alpha) subtitle
 S..... dvbsub               DVB subtitles (codec dvb_subtitle)
 S..... dvdsub               DVD subtitles (codec dvd_subtitle)
 S..... mov_text             3GPP Timed Text subtitle
 S..... srt                  SubRip subtitle (codec subrip)
 S..... subrip               SubRip subtitle
 S..... text                 Raw text subtitle
 S..... webvtt               WebVTT subtitle
 S..... xsub                 DivX subtitles (XSUB)

robotastic, 2020-05-01:
Hmm... the latest version of Trunk Recorder is using fdkaac and sox. do you have those installed?

'Note: v3.1.1 changes from using ffmpeg to fdkaac and sox for compressing audio for OpenMHz. Both utilities can be easily installed using apt-get'

tdduarte, 2020-05-01:
Thank you that worked!

-Tony

closed 2021-04-28 by robotastic