TrunkRecorder/trunk-recorder #272: Additional metadata

mdickw, 2019-10-08:
My local fire departments are switching from VHF to P25. I have a system currently where I check all the radio transmissions for tones and I track all the fire tone-outs (indicates which department the call is for). The transmissions over P25 do not include any tones. I'm guessing that the same information is conveyed digitally in the P25 transmission somehow.

Is there any additional metadata present in the P25 transmissions that doesn't get brought out in the JSON files? Is it possible to access it?

patrickfish, 2019-12-12:
@mdickw do you know if the tones are still dropping over VHF? Around here, those that have migrated to P25 still transmit tones on VHF where all the station alerting gear is still listening.

In cases where they no longer tone over VHF, the stations have moved over to IP-based solutions.

At any rate, it's probably not happening over P25.

mdickw, 2019-12-12:
The tones are still going out over VHF. Hopefully that keeps up. If they switch to something IP-based, that will be disappointing because it will make it much more difficult to pick out the alerts.

JoeGilkey, 2020-01-14:
@mdickw do you have any recordings of the P25 and VHF audio? I can take a look.

closed 2021-04-28 by robotastic