TrunkRecorder/trunk-recorder #274: Compile error ArchLinux looking for multiply_const_ff.h

vabiro, 2019-10-26:
Hi 
I updated gnuradio using pacman, and needed to recompile trunk-recorder, and it seems to be failing looking for multiply_const_ff.h

Here is the error I receive 
```    [ 68%] Linking CXX shared library libgnuradio-op25_repeater.so
    [ 68%] Built target gnuradio-op25_repeater
    Scanning dependencies of target recorder
    [ 69%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/main.cc.o
    In file included from /root/trunk-recorder/trunk-recorder/source.h:11,
                     from /root/trunk-recorder/trunk-recorder/main.cc:42:
    /root/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:23:10: fatal error: gnuradio/blocks/multiply_const_ff.h: No such file or directory
       23 | #include <gnuradio/blocks/multiply_const_ff.h>
          |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    compilation terminated.
    make[2]: *** [CMakeFiles/recorder.dir/build.make:63: CMakeFiles/recorder.dir/trunk-recorder/main.cc.o] Error 1
    make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/recorder.dir/all] Error 2
    make: *** [Makefile:130: all] Error 2
```
I have tried searching for it, but can't find it anywhere on the HD.

Pacman installed the following packages: gnuradio-3.8.0.0-3  gnuradio-osmosdr-0.1.5.r128.f3905d-3, and I do see the file mentioned in the docs for version 3.7.x of gnuradio, but I wonder if something changed in 3.8

Any thoughts?

vabiro, 2019-12-25:
I am continuing to have this problem.

This is happening around the 69% point, which seems to be a common place recently for problems.

Any thoughts? Anyone?

robotastic, 2020-08-04:
Is this still an issue? Since then I thing there have been a switch to GR3.8 that may have solved this.

closed 2021-04-28 by robotastic