TrunkRecorder/trunk-recorder #277: Build Trunk Recorder Fails at 69% Mac OS 10.13

blewbot, 2019-11-05:
I have tried multiple times to get the to work and I am hopelessly stuck. Im running 10.13 

If anyone can help or point me in the right direction it would be greatly appreciated! 

My steps that I followed. 
1. installed installed Xcode 9.4.1
2. Installed Apple's Command Line Developer Tools
3. agreed to Xcode license terms
4. installed macports
5. installed GNU radio with `sudo port install gnuradio uhd gr-osmosdr`
6. Installed tools to compile trunk recorder with `sudo port install cmake boost uhd libusb cppunit`

Now here is the issue when i enter the following commands to build trunk recorder...

```
mkdir trunk-build
git clone https://github.com/robotastic/trunk-recorder.git
cd trunk-build
cmake ../trunk-recorder
make
```
It stops at 

> [ 69%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/main.cc.o

and I get the following error: 

> In file included from /Users/josh/trunk-recorder/trunk-recorder/main.cc:42:
In file included from /Users/josh/trunk-recorder/trunk-recorder/source.h:11:
In file included from /Users/josh/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:42:
In file included from /Users/josh/trunk-recorder/trunk-recorder/recorders/recorder.h:44:
In file included from /Users/josh/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink.h:26:
In file included from /Users/josh/trunk-recorder/lib/../trunk-recorder/call.h:31:
In file included from /Users/josh/trunk-recorder/trunk-recorder/uploaders/call_uploader.h:4:
In file included from /Users/josh/trunk-recorder/trunk-recorder/uploaders/uploader.h:19:
In file included from /Users/josh/trunk-recorder/trunk-recorder/uploaders/../systems/system.h:8:
In file included from /Users/josh/trunk-recorder/trunk-recorder/uploaders/../systems/p25_trunking.h:41:
/Users/josh/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h:51:43: warning: control reaches end of non-void function
      [-Wreturn-type]
      virtual float get_freq_error(void) {}
                                          ^
In file included from /Users/josh/trunk-recorder/trunk-recorder/main.cc:42:
In file included from /Users/josh/trunk-recorder/trunk-recorder/uploaders/../source.h:11:
In file included from /Users/josh/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:42:
In file included from /Users/josh/trunk-recorder/trunk-recorder/recorders/recorder.h:44:
In file included from /Users/josh/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink.h:26:
In file included from /Users/josh/trunk-recorder/trunk-recorder/call.h:31:
In file included from /Users/josh/trunk-recorder/trunk-recorder/uploaders/call_uploader.h:4:
In file included from /Users/josh/trunk-recorder/trunk-recorder/uploaders/uploader.h:8:
In file included from /opt/local/include/boost/asio.hpp:24:
In file included from /opt/local/include/boost/asio/basic_datagram_socket.hpp:20:
In file included from /opt/local/include/boost/asio/basic_socket.hpp:27:
In file included from /opt/local/include/boost/asio/executor.hpp:338:
/opt/local/include/boost/asio/impl/executor.hpp:179:22: error: no member named 'context' in 'std::__1::reference_wrapper<boost::asio::io_context>'
    return executor_.context();
           ~~~~~~~~~ ^
/opt/local/include/boost/asio/impl/executor.hpp:142:3: note: in instantiation of member function
      'boost::asio::executor::impl<std::__1::reference_wrapper<boost::asio::io_context>, std::__1::allocator<void> >::context' requested here
  impl(const Executor& e, const Allocator& a) BOOST_ASIO_NOEXCEPT
  ^
/opt/local/include/boost/asio/impl/executor.hpp:137:30: note: in instantiation of member function
      'boost::asio::executor::impl<std::__1::reference_wrapper<boost::asio::io_context>, std::__1::allocator<void> >::impl' requested here
    impl* p = new (mem.ptr_) impl(e, a);
                             ^
/opt/local/include/boost/asio/impl/executor.hpp:333:50: note: in instantiation of member function
      'boost::asio::executor::impl<std::__1::reference_wrapper<boost::asio::io_context>, std::__1::allocator<void> >::create' requested here
  : impl_(impl<Executor, std::allocator<void> >::create(e))
                                                 ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:2057:18: note: in instantiation of function
      template specialization 'boost::asio::executor::executor<std::__1::reference_wrapper<boost::asio::io_context> >' requested here
[... 388 lines cut ...]
      >::__compressed_pair<std::__1::allocator<boost::asio::basic_stream_socket<boost::asio::ip::tcp, boost::asio::executor> > &,
      std::__1::reference_wrapper<boost::asio::io_context> &&>' requested here
            :  __data_(piecewise_construct, _VSTD::forward_as_tuple(__a),
               ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:4230:26: note: in instantiation of function
      template specialization 'std::__1::__shared_ptr_emplace<boost::asio::basic_stream_socket<boost::asio::ip::tcp, boost::asio::executor>,
      std::__1::allocator<boost::asio::basic_stream_socket<boost::asio::ip::tcp, boost::asio::executor> >
      >::__shared_ptr_emplace<std::__1::reference_wrapper<boost::asio::io_context> >' requested here
    ::new(__hold2.get()) _CntrlBlk(__a2, _VSTD::forward<_Args>(__args)...);
                         ^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/include/c++/v1/memory:4600:29: note: in instantiation of function
      template specialization 'std::__1::shared_ptr<boost::asio::basic_stream_socket<boost::asio::ip::tcp, boost::asio::executor>
      >::make_shared<std::__1::reference_wrapper<boost::asio::io_context> >' requested here
    return shared_ptr<_Tp>::make_shared(_VSTD::forward<_Args>(__args)...);
                            ^
/Users/josh/trunk-recorder/lib/websocketpp/transport/asio/security/none.hpp:171:25: note: in instantiation of function template specialization
      'std::__1::make_shared<boost::asio::basic_stream_socket<boost::asio::ip::tcp, boost::asio::executor>,
      std::__1::reference_wrapper<boost::asio::io_context> >' requested here
        m_socket = lib::make_shared<lib::asio::ip::tcp::socket>(
                        ^
1 warning and 7 errors generated.
make[2]: *** [CMakeFiles/recorder.dir/trunk-recorder/main.cc.o] Error 1
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
make: *** [all] Error 2
old-mbp:trunk-build josh$

robotastic, 2019-11-07:
> 'boost::asio::executor::impl<std::__1::reference_wrapperboost::asio::io_context, std::__1::allocator >::context' requested here
> 'boost::asio::executor::impl<std::__1::reference_wrapperboost::asio::io_context, std::_1::allocator >::impl' requested here
> 'boost::asio::executor::impl<std::__1::reference_wrapperboost::asio::io_context, std::_1::allocator >::create' requested here
> : impl(impl<Executor, std::allocator >::create(e))
> 'boost::asio::executor::impl<std::__1::reference_wrapperboost::asio::io_context, std::__1::allocator >::on_work_started' requested here
> 'boost::asio::executor::impl<std::__1::reference_wrapperboost::asio::io_context, std::__1::allocator >::on_work_finished' requested here
> 'boost::asio::executor::impl<std::__1::reference_wrapperboost::asio::io_context, std::__1::allocator >::dispatch' requested here
> 'boost::asio::executor::impl<std::__1::reference_wrapperboost::asio::io_context, std::__1::allocator >::post' requested here
> 'boost::asio::executor::impl<std::__1::reference_wrapperboost::asio::io_context, std::__1::allocator >::defer' requested here
> 'boost::asio::executor::impl<std::__1::reference_wrapperboost::asio::io_context, std::__1::allocator >::equals' requested here

Unfortuantely, I am having this problem too. I think is related to the version of Boost that Mac Ports is pulling in. I am going to see if I can use Brew to get everything running instead. Let me know if you figure anything out!

NoahJaehnert, 2019-11-07:
I get very similar errors using Homebrew/Brew as well. It appears based on a different project's analysis of these errors that there were changes to the ASIO/Boost code that has broken things: https://github.com/zaphoyd/websocketpp/issues/794

robotastic, 2019-11-08:
> I get very similar errors using Homebrew/Brew as well. It appears based on a different project's analysis of these errors that there were changes to the ASIO/Boost code that has broken things: zaphoyd/websocketpp#794 <https://github.com/zaphoyd/websocketpp/issues/794>

Doh - I think that means that I will have to just remove the WebSocket stuff or atleast make it a build option. ugh...

blewbot, 2019-11-08:
Thanks for looking into this! hopefully you can get it working 🤞🏼. Thank you for supplying this and your hard work.

vabiro, 2019-11-08:
I am having a very similar problem, at the same spot, with ArchLinux since recent upgrades.

I have rolled back boost and gnuradio, and I am still getting errors:

```
[ 68%] Linking CXX shared library libgnuradio-op25_repeater.so
[ 68%] Built target gnuradio-op25_repeater
Scanning dependencies of target recorder
[ 69%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/main.cc.o
In file included from /root/trunk-recorder/lib/../trunk-recorder/uploaders/../systems/p25_trunking.h:41,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/../systems/system.h:8,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/uploader.h:19,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/call_uploader.h:4,
                 from /root/trunk-recorder/lib/../trunk-recorder/call.h:31,
                 from /root/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink.h:26,
                 from /root/trunk-recorder/trunk-recorder/recorders/recorder.h:44,
                 from /root/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:42,
                 from /root/trunk-recorder/trunk-recorder/source.h:11,
                 from /root/trunk-recorder/trunk-recorder/main.cc:42:
/root/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h: In member function ‘virtual float gr::op25_repeater::gardner_costas_cc::get_freq_error()’:
/root/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h:51:43: warning: no return statement in function returning non-void [-Wreturn-type]
   51 |       virtual float get_freq_error(void) {}
      |                                           ^
[ 70%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/formatter.cc.o
[ 71%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/source.cc.o
In file included from /root/trunk-recorder/lib/../trunk-recorder/uploaders/../systems/p25_trunking.h:41,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/../systems/system.h:8,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/uploader.h:19,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/call_uploader.h:4,
                 from /root/trunk-recorder/lib/../trunk-recorder/call.h:31,
                 from /root/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink.h:26,
                 from /root/trunk-recorder/trunk-recorder/recorders/recorder.h:44,
                 from /root/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:42,
                 from /root/trunk-recorder/trunk-recorder/source.h:11,
                 from /root/trunk-recorder/trunk-recorder/source.cc:1:
/root/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h: In member function ‘virtual float gr::op25_repeater::gardner_costas_cc::get_freq_error()’:
/root/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h:51:43: warning: no return statement in function returning non-void [-Wreturn-type]
   51 |       virtual float get_freq_error(void) {}
      |                                           ^
[ 72%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/uploaders/uploader.cc.o
In file included from /root/trunk-recorder/trunk-recorder/uploaders/../systems/p25_trunking.h:41,
                 from /root/trunk-recorder/trunk-recorder/uploaders/../systems/system.h:8,
                 from /root/trunk-recorder/trunk-recorder/uploaders/uploader.h:19,
                 from /root/trunk-recorder/trunk-recorder/uploaders/uploader.cc:1:
/root/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h: In member function ‘virtual float gr::op25_repeater::gardner_costas_cc::get_freq_error()’:
/root/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h:51:43: warning: no return statement in function returning non-void [-Wreturn-type]
   51 |       virtual float get_freq_error(void) {}
      |                                           ^
[ 74%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/uploaders/call_uploader.cc.o
In file included from /root/trunk-recorder/trunk-recorder/uploaders/../systems/p25_trunking.h:41,
                 from /root/trunk-recorder/trunk-recorder/uploaders/../systems/system.h:8,
                 from /root/trunk-recorder/trunk-recorder/uploaders/uploader.h:19,
                 from /root/trunk-recorder/trunk-recorder/uploaders/call_uploader.h:4,
                 from /root/trunk-recorder/trunk-recorder/uploaders/call_uploader.cc:1:
/root/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h: In member function ‘virtual float gr::op25_repeater::gardner_costas_cc::get_freq_error()’:
/root/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h:51:43: warning: no return statement in function returning non-void [-Wreturn-type]
   51 |       virtual float get_freq_error(void) {}
      |                                           ^
[ 75%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/uploaders/stat_socket.cc.o
In file included from /root/trunk-recorder/lib/../trunk-recorder/uploaders/../systems/p25_trunking.h:41,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/../systems/system.h:8,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/uploader.h:19,
[... cut ...]
/usr/bin/ld: /usr/lib/gcc/x86_64-pc-linux-gnu/9.2.0/../../../../lib/libgnuradio-osmosdr.so: undefined reference to `gr::hier_block2::log_level[abi:cxx11]()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/recorder.dir/build.make:508: recorder] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/recorder.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
At this point I am using:
boost-1.69.0-2
gnuradio-3.7.13.5-2
libuhd-3.14.0
libvolk-1.4-3

At this point it will not compile, and previously compiled versions of recorder will not run.

Anyone have any suggestions?

Thanks

NoahJaehnert, 2019-11-21:
@robotastic Any thoughts on how we could get this working with the latest versions of libboost? I'd really like to get Trunk Recorder working so I can feed OpenMHZ and Broadcastify.

robotastic, 2019-12-08:
Ok - I have this fixed on OSX. Give this branch a try: https://github.com/robotastic/trunk-recorder/tree/remove-websocket
I removed the feature that was breaking the build. Let me know if this doesn't work.
If you are using the command line, do: `git checkout remove-websocket`

NoahJaehnert, 2019-12-10:
Still seems broken for me. I originally had issues with libssl. Error I was receiving was: 
```
[100%] Linking CXX executable recorder
"ld: library not found for -lssl"
```
I was able to solve the Brew libssl issue via these instructions: https://forum.crystal-lang.org/t/noob-error-in-brew-installation-on-mac/714/6?u=thbo

Now I'm having a different problem, still related to the Boost libraries - `ld: symbol(s) not found for architecture x86_64
`

Full error/output below: 
`[100%] Linking CXX executable recorder
Undefined symbols for architecture x86_64:
  "boost::log::v2_mt_posix::attributes::named_scope::named_scope()", referenced from:
      _main in main.cc.o
  "boost::log::v2_mt_posix::record_view::public_data::destroy(boost::log::v2_mt_posix::record_view::public_data const*)", referenced from:
      load_config(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >) in main.cc.o
      start_recorder(Call*, TrunkMessage, System*) in main.cc.o
      stop_inactive_recorders() in main.cc.o
      print_status() in main.cc.o
      retune_recorder(TrunkMessage, Call*) in main.cc.o
      handle_call(TrunkMessage, System*) in main.cc.o
      find_system(int) in main.cc.o
      ...
  "boost::log::v2_mt_posix::attribute_name::get_id_from_string(char const*)", referenced from:
      _main in main.cc.o
  "boost::log::v2_mt_posix::basic_formatter<char> boost::log::v2_mt_posix::parse_formatter<char>(char const*, char const*)", referenced from:
      boost::shared_ptr<boost::log::v2_mt_posix::sinks::synchronous_sink<boost::log::v2_mt_posix::sinks::text_file_backend> > boost::log::v2_mt_posix::aux::add_file_log<boost::parameter::aux::flat_like_arg_list<boost::parameter::aux::flat_like_arg_tuple<boost::log::v2_mt_posix::keywords::tag::file_name, boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::file_name, char const [27]>, std::__1::integral_constant<bool, true> >, boost::parameter::aux::flat_like_arg_tuple<boost::log::v2_mt_posix::keywords::tag::auto_flush, boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::auto_flush, bool const>, std::__1::integral_constant<bool, true> >, boost::parameter::aux::flat_like_arg_tuple<boost::log::v2_mt_posix::keywords::tag::time_based_rotation, boost::parameter::aux::tagged_argument_rref<boost::log::v2_mt_posix::keywords::tag::time_based_rotation, boost::log::v2_mt_posix::sinks::file::rotation_at_time_point>, std::__1::integral_constant<bool, true> >, boost::parameter::aux::flat_like_arg_tuple<boost::log::v2_mt_posix::keywords::tag::format, boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::format, char const [25]>, std::__1::integral_constant<bool, true> >, boost::parameter::aux::flat_like_arg_tuple<boost::log::v2_mt_posix::keywords::tag::rotation_size, boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::rotation_size, int const>, std::__1::integral_constant<bool, true> > > >(boost::parameter::aux::flat_like_arg_list<boost::parameter::aux::flat_like_arg_tuple<boost::log::v2_mt_posix::keywords::tag::file_name, boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::file_name, char const [27]>, std::__1::integral_constant<bool, true> >, boost::parameter::aux::flat_like_arg_tuple<boost::log::v2_mt_posix::keywords::tag::auto_flush, boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::auto_flush, bool const>, std::__1::integral_constant<bool, true> >, boost::parameter::aux::flat_like_arg_tuple<boost::log::v2_mt_posix::keywords::tag::time_based_rotation, boost::parameter::aux::tagged_argument_rref<boost::log::v2_mt_posix::keywords::tag::time_based_rotation, boost::log::v2_mt_posix::sinks::file::rotation_at_time_point>, std::__1::integral_constant<bool, true> >, boost::parameter::aux::flat_like_arg_tuple<boost::log::v2_mt_posix::keywords::tag::format, boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::format, char const [25]>, std::__1::integral_constant<bool, true> >, boost::parameter::aux::flat_like_arg_tuple<boost::log::v2_mt_posix::keywords::tag::rotation_size, boost::parameter::aux::tagged_argument<boost::log::v2_mt_posix::keywords::tag::rotation_size, int const>, std::__1::integral_constant<bool, true> > > const&) in main.cc.o
  "void boost::log::v2_mt_posix::aux::put_integer<char>(boost::log::v2_mt_posix::aux::basic_ostringstreambuf<char, std::__1::char_traits<char>, std::__1::allocator<char> >&, unsigned int, unsigned int, char)", referenced from:
      boost::log::v2_mt_posix::aux::date_time_formatter<boost::log::v2_mt_posix::aux::decomposed_time_wrapper<boost::posix_time::ptime>, char>::format_short_year(boost::log::v2_mt_posix::aux::date_time_formatter<boost::log::v2_mt_posix::aux::decomposed_time_wrapper<boost::posix_time::ptime>, char>::context&) in main.cc.o
      boost::log::v2_mt_posix::aux::date_time_formatter<boost::log::v2_mt_posix::aux::decomposed_time_wrapper<boost::posix_time::ptime>, char>::format_full_year(boost::log::v2_mt_posix::aux::date_time_formatter<boost::log::v2_mt_posix::aux::decomposed_time_wrapper<boost::posix_time::ptime>, char>::context&) in main.cc.o
      boost::log::v2_mt_posix::aux::date_time_formatter<boost::log::v2_mt_posix::aux::decomposed_time_wrapper<boost::posix_time::ptime>, char>::format_numeric_month(boost::log::v2_mt_posix::aux::date_time_formatter<boost::log::v2_mt_posix::aux::decomposed_time_wrapper<boost::posix_ti [... cut ...]
  "boost::log::v2_mt_posix::sinks::file::rotation_at_time_point::operator()() const", referenced from:
      boost::log::v2_mt_posix::aux::light_function<bool ()>::impl<boost::log::v2_mt_posix::sinks::file::rotation_at_time_point>::invoke_impl(void*) in main.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [recorder] Error 1
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
make: *** [all] Error 2`

Thoughts?

NoahJaehnert, 2019-12-24:
> Ok - I have this fixed on OSX. Give this branch a try: https://github.com/robotastic/trunk-recorder/tree/remove-websocket
> I removed the feature that was breaking the build. Let me know if this doesn't work.
> If you are using the command line, do: `git checkout remove-websocket`

@robotastic Any additional thoughts/ideas? I tried compiling but am still getting errors (see post above). Thanks!

NoahJaehnert, 2020-02-06:
Update: Ended up switching to MacPorts, which resolved my issues above. Would still be good to figure out why same/similar packages in Brew don't allow trunk-recorder to compile correctly, but at least my issue is now solved...

PaulCapestany, 2020-04-10:
I'm also still running into the same issue seemingly related to boost as @NoahJaehnert while trying to build trunk-recorder via macports, specifically:

```
[100%] Linking CXX executable recorder
…
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [recorder] Error 1
make[1]: *** [CMakeFiles/recorder.dir/all] Error 2
make: *** [all] Error 2```

robotastic, 2020-05-01:
Have you had any luck? I am compiling the latest version of Trunk Recorder on OSX. I am on Catalina.

h00fty, 2020-05-01:
I was able to compile a few days ago. went smoothly without any issues.

closed 2020-08-04 by robotastic