TrunkRecorder/trunk-recorder #278: P25 does not properly record calls that are mixed-mode (clear & encryption)

bctrainers, 2019-11-15:
Loaded up Trunk Recorder today to mess about with things, and noticed an interesting issue with how trunk-recorder handles encrypted and clear-air radio transmissions on the same talk group.

Currently, trunk recorder does one of two things when it comes to a talk group in a mixed mode of transmission types (encrypted and clear-air)...

1) If the first transmission that comes in is clear-air on a mixed-mode talk group, trunk-recorder will record both clear and encrypted chatter for the entire duration until the timeout limit is reached. Which would appear to be normal behavior? 
2) If the first transmission that comes in is encrypted, and clear-air audio comes in after, trunk-recorder will persist in a 'monitoring' state, but never record anything as the initial transmission was encrypted.

rpdale, 2019-11-15:
All scanners that I am familiar with perform this way (unless you tell them to monitor encrypted.) Is there anything in the control channel output that tells the receiver that the encrypted comm went into the clear? I thought the status bit was just set at the onset.

bctrainers, 2019-11-15:
On my whistler 1098, it will pick up the clear-air and encrypted audio without delay or pause between the two modes. Worth noting that there's an option that can be set on the whistler application to play a busy tone while the big-E is active on a mixed-mode talk group that isn't on skip mode or lockout. So it is aware of what is and is not encrypted during an active transmission.

bctrainers, 2019-11-15:
Another issue (probably best as a new bug report) I've observed on the P25 system I have configured is with a resample-rate issue. The resample rate does not return to its original resample rate when a transmission ends  - it persists until the next transmission on the same recorder with TrunkRecorder begins. This appears to be causing some severe overlap of audio into other active TG's that key up - reproducible with HackRF and some RTLSDR's I had around.

Bad: 
`P25 Recorder - Adjust Channel Resampled Rate: 30487.8 System Rate: 24000 ARB Rate: 0.7872`
Good: 
`P25 Recorder Initial Rate: 487805 Resampled Rate: 48780.5 Initial Decimation: 41 Decimation: 10 System Rate: 48000 ARB Rate: 0.984`

The bad one above is from an automated-voice dispatch talk group for Fire & EMS. When TrunkRecorder starts up, the good is what persists on the indvidual recorders with TrunkRecorder... until the automated dispatch begins its broadcast. :/

closed 2024-12-15 by robotastic