TrunkRecorder/trunk-recorder #281: First sec of P25 transmissions sounds garbled

dbuttry, 2019-11-27:
I am monitoring 2 different P25 systems using 2 RTL-SDRs.  One of them sounds great https://openmhz.com/system/metcad but the other one doesn't.  My StarCom21-Champaign https://openmhz.com/system/metcad has a garbled sound at the beginning of most transmissions.  After the initial encryption type sound, the rest sounds okay.  Any ideas?
Thanks!!

bctrainers, 2019-11-30:
Hey, what does the RTLSDR's raw signal data via gqrx or SDR# look like? It sounds like classic drifting going on when the transmission has a frequency change and has to retune to another RTLSDR.

Could you post your config segment for the sc21chmp and metcad system?

dbuttry, 2019-12-01:
Here is my config file:

{
    "sources": [{
        "center": 851500000,
        "rate": 2500000.0,
        "squelch": 0,
        "error": 1000,
	"ppm": 0,
        "gain": 24,
        "digitalRecorders": 5,
        "driver": "osmosdr",
        "device": "rtl=00000001",
        "modulation": "qpsk"
 }, {
        "center": 854100000,
        "rate": 2500000,
        "squelch": 0,
        "error": 1000,
        "gain": 24,
        "digitalRecorders": 5,
        "driver": "osmosdr",
        "device": "rtl=00000002",
        "modulation": "qpsk"
    }],
    "systems": [{
        "control_channels": [853825000, 852562500, 853137500, 853525000],
        "type": "p25",
	"shortName": "METCAD",
	"talkgroupDisplayFormat": "id_tag",
	"hideEncrypted": "true",
        "talkgroupsFile": "METCAD-ChanList.csv",
	"apiKey": “xxxxxxxxxxxxxxxxxxxxx”
    },{
        "control_channels": [851937500, 852462500],
        "type": "p25",
        "shortName": "SC21CHMP",
        "talkgroupDisplayFormat": "id_tag",
        "hideEncrypted": "true",
        "talkgroupsFile": "STARCOM21-ChanList.csv",
	"apiKey": “xxxxxxxxxxxxxxxxxxxxx”
    }],
    "captureDir": "/home/dougb/Recordings/",
    "frequencyFormat": "mhz",
    "uploadServer": "https://api.openmhz.com"
}

bctrainers, 2019-12-01:
Some quick ramblings that come to mind:

A rate of 2500000 might be too much for your RTL-SDR's. I traditionally have run into the said issue once I've surpassed the 2400000 sample rate.  Try reducing the sample rate, but you'll need to account for the lost frequency range. Quick math says you'll need about 4.1MHz of frequency coverage, so 2400000 should be more than enough spread across the two SDR's, a total of 4.8 MHz of the spectrum covered rather the 5MHz currently configured. 

I assume you're trying to lock to Champaign, IL's Starcom and the single MetCAD Champaign, IL tower? Bold Italic: Primary CC, Bold: Alt CC.

Starcom: 851.4125, **851.9375**, **_852.4625_**, 852.9875, 854.9625, 855.1625
MetCAD: 851.0625, 851.550, 851.800, 852.100, **852.5625**, **853.1375**, **853.525**, **_853.825_**

Looking at the frequency space it uses, you're possibly overloading the RTL-SDR's with too much spectrum on the sampling rates. 

For sample rates with more than one SDR device, take the lowest frequency programmed (eg; 851000000) and add the sample rate (2400000) to it. That way, my next SDR's frequency center is not going to overlap with the next-lowest SDR configured (853400000). It's always best to keep in mind that you want to try and avoid "fringe" frequencies covered by any single RTL-SDR device. Have seen instances where this will cause really funky frequency decoding (robots, distortion, pops, bad-key-up on wrong TG/overlap).

Another option to test is to remove the functional system and see if the popping/pre-verbal audio of distortion shows up with the other system. If it is continuing, there's a very likely chance that it's the actual P25 system having wonky programming. I've never experienced such personally, however.

dbuttry, 2019-12-01:
Thank you very much!  I lowered the sample rate to 2.4 and changed my center freqs to 852.000 and 854.200.  That gives me a range of 850.800 - 853.200 & 853.000 - 855.400.  I know there is a small overlap (853.000 - 853.200) but I don't have anything in that range anyway.  So far this seems to be working better.  Time will tell.

dbuttry, 2019-12-01:
Well, maybe I spoke too soon....I will play with this some more.  Do you think I would be better off getting a 3rd RTL-SDR and lowering the sample rates more or do you recommend another SDR that isn't a ton of money?

bctrainers, 2019-12-01:
No need for a 3rd SDR currently, the spectrum you're covering is quite small compared to the other systems out there that use more than 10MHz of spectrum. :)

closed 2021-04-28 by robotastic

hayden-t, 2023-02-10:
any luck with this ?