TrunkRecorder/trunk-recorder #282: Failed parsing Config: std::bad_alloc

jacygamer, 2019-12-02:
New to trunk-recorder I am trying to monitor the "Kansas Statewide Interoperable Communication System (KSICS)"https://www.radioreference.com/apps/db/?sid=4390 "Harvey County system"
this is my config.json
 {
    "sources": [{
        "center": 85600000,
        "rate": 240000,
        "error": 54000,
        "gain": 39,
        "debugRecorders": 0,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=0"
    }],
    "systems": [{
        "control_channels": [857.987500],
        "type": "p25",
        "talkgroupsFile": "talkgroups.csv",
				"shortName": "ksstate_newton"
    }],
    "captureDir": "~/audio_files/"
}
output of running
[2019-12-02 10:50:34.109575] (info)   Using Config file: config-newton-rtl.json

[2019-12-02 10:50:34.110217] (info)
-------------------------------------
SYSTEMS
-------------------------------------
0
[2019-12-02 10:50:34.110819] (info)

System Number: 1
-------------------------------------
1
[2019-12-02 10:50:34.111616] (info)   Short Name: ksstate_newton
[2019-12-02 10:50:34.111872] (info)   System Type: p25
[2019-12-02 10:50:34.112114] (info)   Control Channels:
[2019-12-02 10:50:34.112350] (info)   857.987
[2019-12-02 10:50:34.112586] (info)   API Key:
[2019-12-02 10:50:34.112797] (info)   Upload Script:
[2019-12-02 10:50:34.113005] (info)   Call Log: true
[2019-12-02 10:50:34.113218] (info)   Audio Archive: true
[2019-12-02 10:50:34.113433] (info)   Loading Talkgroups...
[2019-12-02 10:50:34.114034] (info)   Read 135 talkgroups.
[2019-12-02 10:50:34.117851] (info)   Talkgroups File: talkgroups.csv
[2019-12-02 10:50:34.117937] (info)   Record Unknown Talkgroups: true
[2019-12-02 10:50:34.118076] (info)   Talkgroup Display Format: Id
[2019-12-02 10:50:34.118193] (info)   Hide Encrypted Talkgroups: false
[2019-12-02 10:50:34.118413] (info)   Hide Unknown Talkgroups: false
[2019-12-02 10:50:34.118636] (info)   Minimum Call Duration (in seconds): 0
[2019-12-02 10:50:34.118909] (info)
[2019-12-02 10:50:34.118984] (info)

-------------------------------------
SOURCES
-------------------------------------

[2019-12-02 10:50:34.119879] (info)   Driver: osmosdr
[2019-12-02 10:50:34.119895] (info)   Center: 8.560000e+07
[2019-12-02 10:50:34.120114] (info)   Rate: 240000
[2019-12-02 10:50:34.120141] (info)   Error: -43000
[2019-12-02 10:50:34.120157] (info)   PPM Error: 0
[2019-12-02 10:50:34.120172] (info)   Gain: 39
[2019-12-02 10:50:34.120264] (info)   IF Gain: 0
[2019-12-02 10:50:34.120283] (info)   BB Gain: 0
[2019-12-02 10:50:34.120297] (info)   LNA Gain: 0
[2019-12-02 10:50:34.120311] (info)   PGA Gain: 0
[2019-12-02 10:50:34.120325] (info)   TIA Gain: 0
[2019-12-02 10:50:34.120338] (info)   MIX Gain: 0
[2019-12-02 10:50:34.120417] (info)   VGA1 Gain: 0
[2019-12-02 10:50:34.120513] (info)   VGA2 Gain: 0
[2019-12-02 10:50:34.120535] (info)   Squelch: 0
[2019-12-02 10:50:34.120555] (info)   Idle Silence: false
[2019-12-02 10:50:34.120646] (info)   Digital Recorders: 4
[2019-12-02 10:50:34.120671] (info)   Debug Recorders: 0
[2019-12-02 10:50:34.120691] (info)   SigMF Recorders: 0
[2019-12-02 10:50:34.120709] (info)   Analog Recorders: 0
[2019-12-02 10:50:34.120728] (info)   Modulation: qpsk
[2019-12-02 10:50:34.120838] (info)   Source Device: rtl=0
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
Using device #0 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[2019-12-02 10:50:34.772341] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2019-12-02 10:50:34.772373] (info)   Setting sample rate to: 240000
[R82XX] PLL not locked!
[2019-12-02 10:50:34.899027] (info)   Actual sample rate: 240000
[2019-12-02 10:50:34.899085] (info)   Tunning to 8.555700e+07
[2019-12-02 10:50:34.953592] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6
[2019-12-02 10:50:34.953627] (info)   Max Freqency: 8.572000e+07
[2019-12-02 10:50:34.953644] (info)   Min Freqency: 8.548000e+07
[2019-12-02 10:50:34.981642] (info)   LNA Gain set to: 0
[2019-12-02 10:50:35.009068] (info)   TIA Gain set to: 0
[2019-12-02 10:50:35.036864] (info)   PGA Gain set to: 0
[2019-12-02 10:50:35.065500] (info)   Gain set to: 38.6
[2019-12-02 10:50:35.065552] (info)   Setting antenna to [RX]
[2019-12-02 10:50:35.069965] (error)   wav error closing file

[2019-12-02 10:50:35.070121] (error)   Failed parsing Config: std::bad_alloc
[2019-12-02 10:50:35.070286] (info)   [ksstate_newton]  Started with Control Channel: 8.579875e+02
[2019-12-02 10:50:35.070738] (error)   Unable to setup a System to record, exiting...
Trying to figure out what the errors  " wav error closing file" and " Failed parsing Config: std::bad_alloc"

Thanks

gofaster, 2019-12-02:
"control_channels": [857.987500],  <-- This value should be in Hz, not MHz.

bctrainers, 2019-12-02:
Hey, so there's a couple of things with your configuration that are suspect at first glance. Listing them below...

1. KSICS Harvey County's P25 spans from 854.4 to 858.0 (rounding off values).
2. Your sample rate on the single RTLSDR is not sufficient enough to cover that spectrum properly. You will need a second RTLSDR configured due to it being more than 2.4MHz of spectrum. 
3. Your sample rate will need to be 2400000, not 240000.
4. Your control channel should be `857987500` in Hz - not the MHz form.

jacygamer, 2019-12-02:
Thanks for the reply's I was using the wrong config file had the control channel fixed and the rate but I did add a second dongle this is the new config file
{
    "sources": [{
        "center": 856000000,
        "rate":  2400000,
        "error": 0,
        "gain": 39,
		"ppm": 54,
        "debugRecorders": 0,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=0"
    }, {
        "center": 854000000,
        "rate":  2400000,
        "error": 0,
        "gain": 39,
		"ppm": 54,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=1"
    }],
    "systems": [{
        "control_channels": [857987500],
        "type": "p25",
        "talkgroupsFile": "talkgroups.csv",
				"shortName": "newton"
    }],
    "captureDir": "~/audio_files"
and the output 
} [2019-12-02 13:05:21.522506] (info)   Using Config file: config-newton-rtl.json

[2019-12-02 13:05:21.539170] (info)
-------------------------------------
SYSTEMS
-------------------------------------
0
[2019-12-02 13:05:21.539834] (info)

System Number: 1
-------------------------------------
1
[2019-12-02 13:05:21.540710] (info)   Short Name: newton
[2019-12-02 13:05:21.542244] (info)   System Type: p25
[2019-12-02 13:05:21.542493] (info)   Control Channels:
[2019-12-02 13:05:21.542737] (info)   8.57988e+08
[2019-12-02 13:05:21.543904] (info)   API Key:
[2019-12-02 13:05:21.544057] (info)   Upload Script:
[2019-12-02 13:05:21.544204] (info)   Call Log: true
[2019-12-02 13:05:21.544300] (info)   Audio Archive: true
[2019-12-02 13:05:21.544323] (info)   Loading Talkgroups...
[2019-12-02 13:05:21.559630] (info)   Read 1669 talkgroups.
[2019-12-02 13:05:21.559790] (info)   Talkgroups File: talkgroups.csv
[2019-12-02 13:05:21.559921] (info)   Record Unknown Talkgroups: true
[2019-12-02 13:05:21.560080] (info)   Talkgroup Display Format: Id
[2019-12-02 13:05:21.560325] (info)   Hide Encrypted Talkgroups: false
[2019-12-02 13:05:21.560537] (info)   Hide Unknown Talkgroups: false
[2019-12-02 13:05:21.560748] (info)   Minimum Call Duration (in seconds): 0
[2019-12-02 13:05:21.560967] (info)
[2019-12-02 13:05:21.561144] (info)

-------------------------------------
SOURCES
-------------------------------------

[2019-12-02 13:05:21.562010] (info)   Driver: osmosdr
[2019-12-02 13:05:21.562192] (info)   Center: 8.560000e+08
[2019-12-02 13:05:21.562395] (info)   Rate: 2400000
[... 100 lines cut ...]
[2019-12-02 13:05:23.963540] (info)      P25 Recorder Initial Rate: 480000 Resampled Rate: 30000 Initial Decimation: 5 Decimation: 16 System Rate: 30000 ARB Rate: 1
[2019-12-02 13:05:23.963726] (info)   P25 Recorder Taps - initial: 363 channel: 1455 ARB: 727 Total: 2545
[2019-12-02 13:05:23.967805] (info)
-------------------------------------

[2019-12-02 13:05:23.968358] (info)

-------------------------------------
INSTANCE
-------------------------------------

[2019-12-02 13:05:23.969386] (info)   Capture Directory: ~/audio_files
[2019-12-02 13:05:23.969604] (info)   Upload Server:
[2019-12-02 13:05:23.969826] (info)   Status Server:
[2019-12-02 13:05:23.970035] (info)   Instance Key:
[2019-12-02 13:05:23.970244] (info)   Instance Id:
[2019-12-02 13:05:23.970451] (info)   Default Mode: digital
[2019-12-02 13:05:23.970663] (info)   Call Timeout (seconds): 3
[2019-12-02 13:05:23.970874] (info)   Log to File: false
[2019-12-02 13:05:23.971081] (info)   Control channel warning rate: 10
[2019-12-02 13:05:23.971290] (info)   Control channel retune limit: 0
[2019-12-02 13:05:23.971501] (info)   Frequency format: 0
[2019-12-02 13:05:23.971709] (info)   Status as String: false
[2019-12-02 13:05:23.971925] (info)   [newton]  Started with Control Channel: 8.579875e+08
[2019-12-02 13:05:23.972236] (error)   Unable to setup a System to record, exiting...

bctrainers, 2019-12-03:
So, this seems to be a common issue - and it's a gotchya that had me going for a loop a while back.

RTL-SDR's maximum bandwidth at 2.4Msps (2.4MHz), can only be split 50/50 per each side of the center that you've programmed. That means that the left of the center tuned-frequency can only go 1.2MHz, and the right of the center can only go 1.2MHz for a total of 2.4MHz. You'll need to program both SDR's to be in the middle of KSICS frequency ranges assigned for Harvey County. As per https://www.radioreference.com/apps/db/?siteId=26062 - You will need to configure one of the SDR's to the bouts of 855650000 and the second to 858000000

Also, RTL-SDR's PPM isn't as graceful on channel drifting, unlike the "error" value for a RTL-SDR. PPM is more useful on devices with a larger frequency/spectrum capture. You will likely need to do some checking with a 0 PPM setting, and adjusting the RTL-SDR's error value. Ref this page: https://github.com/robotastic/trunk-recorder/wiki/Configuring-a-System

jacygamer, 2019-12-03:
Thanks For the help changed the error and reset the spacing per dongle and is working now thanks

closed 2019-12-03 by jacygamer