TrunkRecorder/trunk-recorder #283: Length of recording in JSON file does not match WAV file

aaknitt, 2019-12-06:
It appears that the length of the recording indicated in the JSON file is typically much longer than the length of the actual audio recording.  Noticed this locally and also checked on OpenMHz and noticed it with other systems.

aaknitt, 2019-12-08:
It appears that the reason for this discrepancy is because the time indicated in the JSON file is the entire time that the talkgroup is active, while the WAV recording "removes silences" between transmissions on that talkgroup - when no IMBE data is present, the WAV file is not being padded with zeros.  This is probably desirable for most uses cases.  

I had been thinking about putting together a script to create Audacity multi-track project files from a subset of talkgroups (each talkgroup a separate audio track) for "synchronized" re-creation of transmissions during an incident.  With the current behavior, this type of re-creation would be an approximation at best since silences between transmissions on the same talkgroup would not be captured.  However, I see now that the JSON file contains timestamp information on when each transmission within the recording started, and this could be used to split the WAV files for more accurate re-creation.  One request would be to include milliseconds in the srcsList->time field.

closed 2019-12-08 by aaknitt

bctrainers, 2019-12-08:
Hi, where exactly are you seeing `srcsList->time` called at in the source files?

I am assuming you're talking about Line 145 in call.cc?

https://github.com/robotastic/trunk-recorder/blob/2e2efb6ffa1f1f101e0fe1b21a92a57104dba07f/trunk-recorder/call.cc#L145

aaknitt, 2019-12-08:
I was just looking at the .json file.  Looking at the source, it looks like it's in Line 151 of call.cc:

https://github.com/robotastic/trunk-recorder/blob/2e2efb6ffa1f1f101e0fe1b21a92a57104dba07f/trunk-recorder/call.cc#L151