TrunkRecorder/trunk-recorder #284: Recordings cut before end of transmission, audio from wrong talkgroup in recording - Pi 4

aaknitt, 2019-12-06:
I'm running trunk-recorder with two RTL-SDR dongles on a Pi 4 and have noticed that the recordings of many transmissions (usually longer transmissions) are cut short and in many cases audio is being recorded into WAV files for the wrong talkgroup.  

I'm not sure if this is simply because I'm overtaxing the Pi 4 or if it's an issue with trunk-recorder and audio buffering.  

Received signal strength/decoding seem very good.  If I can collect more data to troubleshoot please advise.

bctrainers, 2019-12-06:
This is an issue with the USB "host" machine. The USB on the device (not the RTLSDR) cannot handle the amount of bandwidth required for the slices needed with the recorders active. You will need a more powerful machine. Virtual Machines also suffer from this issue. This means you'll need a machine that has a decent CPU (something with 8 cores ideally) and sufficient USB ports on the motherboard.

rpdale, 2019-12-06:
My pi4 does MPSCS P25 just fine with 2 RTLSDRs.

aaknitt, 2019-12-06:
> This is an issue with the USB "host" machine. The USB on the device (not the RTLSDR) cannot handle the amount of bandwidth required for the slices needed with the recorders active. You will need a [...]

As an update on this, it seems like when transmissions get cut short, the end of the transmission frequently ends up in a subsequent recording...sometimes a recording for the same talkgroup, other times in a recording for a different (incorrect) talkgroup.  So it seems like (I can't say this for sure) that transmissions are all being received and decoded, just not put in the correct file.  I'm trying to determine if some of this may be related to talkgroup patching, but I don't think that explains all of it.   With that in mind, do you still think this could still be related to USB bandwidth?

aaknitt, 2019-12-06:
> My pi4 does MPSCS P25 just fine with 2 RTLSDRs.

Out of curiosity, what bandwidth are you running each of them at, and how many digital recorders do you have assigned to each?

bctrainers, 2019-12-06:
I had the exact same issue while I was messing about with TrunkRecorder with five RTLSDR's, and another configuration with one HackRF One. 

This was running on various hardware configurations via VMWare ESXi, VMWare workstation, Raspberry Pi3, Raspberry Pi4, and an underpowered machine (2 cores, 4GB RAM). All did the same thing: severe audio overlapping, but the JSON reported the timings and unit ID's correctly.

It all came down to the USB bandwidth available as a whole. I was barely scratching the surface of the VMWare workstation and ESXi machines vCPU's (unlike the pi's). It was all being bottlenecked at the USB.  Raspberry Pi's shared-bus on the 3 was a killer. VMWare appears to default to the lower-end of speeds to utilize on its USB pass-thru's. 

Slapped the HackRF into a much more powerful machine, and the overlapping of audio vanished, unless somehow a certain talk group keys up and causes the sample rate to go awry.

aaknitt, 2019-12-06:
> I had the exact same issue while I was messing about with TrunkRecorder with five RTLSDR's, and another configuration with one HackRF One.
>
> This was running on various hardware configurations via [...]

Got it, thanks.  Your comment made me remember that I added a USB flash drive to save the audio to and that there may have been a correlation between adding that and the problem.  I ran a quick test where I removed the USB flash drive and save the audio back to the Pi SD card and while that didn't completely eliminate the issue, it did seem to improve things a reasonable amount, so I think your explanation makes sense.  That's unfortunate, as the Pi 4 seems to have enough processing HP to keep up...too bad that USB bandwidth is the bottleneck. 

I'll close the issue as it seems to be related to the machine running trunk-recorder, not the program itself.

aaknitt, 2019-12-06:
> > I had the exact same issue while I was messing about with TrunkRecorder with five RTLSDR's, and another configuration with one HackRF One.
> > This was running on various hardware configurations via VMWare ESXi, VMWare workstation, Raspberry Pi3, Raspberry Pi4, and an underpowered machine (2 cores, 4GB RAM). All did the same thing: severe audio overlapping, but the JSON reported the timings and unit ID's correctly.
> > It all came down to the USB bandwidth available as a whole. I was barely scratching the surface of the VMWare workstation and ESXi machines vCPU's (unlike the pi's). It was all being bottlenecked at the USB. Raspberry Pi's shared-bus on the 3 was a killer. VMWare appears to default to the lower-end of speeds to utilize on its USB pass-thru's.
> > Slapped the HackRF into a much more powerful machine, and the overlapping of audio vanished, unless somehow a certain talk group keys up and causes the sample rate to go awry.
> 
> Got it, thanks. Your comment made me remember that I added a USB flash drive to save the audio to and that there may have been a correlation between adding that and the problem. I ran a quick test where I removed the USB flash drive and saved the audio back to the Pi SD card and while that didn't completely eliminate the issue, it did seem to improve things a reasonable amount, so I think your explanation makes sense. That's unfortunate, as the Pi 4 seems to have enough processing HP to keep up...too bad that USB bandwidth is the bottleneck.
> 
> I'll close the issue as it seems to be related to the machine running trunk-recorder, not the program itself.

However, the more I'm thinking about this...

The sampling rate of the RTL dongle is the same all the time regardless of how many voice channels/recorders are currently active.  It seems like this problem is most prevalent when multiple digital recorders (talkgroups) are active at the same time.  The USB data pipe is simply for the I/Q samples, and those are being passed regardless of how many digital recorders are active.  So perhaps this is something to do with the processor load and buffering more than USB bandwidth?

bctrainers, 2019-12-06:
From my findings, it was down to the bandwidth that the USB "hub" or controller was negotiating the USB port speed at. VM's configured with USB 2.0 mode was using the Full-Speed data rate - which is 12Mbps. Directly connected to a machine without the usage of a VM, was getting the full negotiated USB 2.0 speed without issue. That's one of the issues that's been notable over the years.

Your mention of "most prevalent when multiple digital recorders (talkgroups) are active" is among another issue. Sometimes this is a CPU issue, but it's not always true to that being the sole issue. It's an issue I've been having for quite some time, regardless of any hardware I've used. No matter how well-adjusted that the PPM/Error is.

It's mostly related to prolonged transmissions, generally 30 seconds or longer for me.

rpdale, 2019-12-06:
"sources": [{
        "center": 852362500.0,
        "rate": 2400000.0,
        "error": 1000,
        "gain": 47,
        "driver": "osmosdr",
        "device": "rtl=1",
        "modulation": "fsk4",
        "digitalRecorders": 5
    },
{
        "center": 770543750.0,
        "rate": 1800000.0,
        "error": -32000,
        "gain": 47,
        "driver": "osmosdr",
        "device": "rtl=2",
        "modulation": "fsk4",
        "digitalRecorders": 3
    }],

From: aaknitt
Sent: Thursday, December 5, 2019 10:38 PM
To: robotastic/trunk-recorder
Cc: rpdale; Comment
Subject: Re: [robotastic/trunk-recorder] Recordings cut before end oftransmission, audio from wrong talkgroup in recording - Pi 4 (#284)

My pi4 does MPSCS P25 just fine with 2 RTLSDRs.
Out of curiosity, what bandwidth are you running each of them at, and how many digital recorders do you have assigned to each?

aaknitt, 2019-12-21:
Well I upgraded to a much more powerful machine (Core i5 quad core 8GB RAM running Ubuntu) and I'm still seeing similar issues with audio ending up in the wrong talkgroup, usually related to long transmissions.  As before, seems to be worse when several talkgroups are active at the same time.  Per comments [here](https://forums.radioreference.com/threads/trunk-recorder-on-pi-3.396903/) (which I tend to agree with) I think it may be an audio buffering issue of some sort, or perhaps time delay between control channel decoding and audio decoding?  Not really sure.

robotastic, 2019-12-21:
The could be some sort of issue around buffering. There are buffers in the pipeline for processing samples. I remember looking into try to reduce the buffer size, the risk is that making the buffers too small might mean losing some samples. It has been a tough problem to work with because it doesn't happen too often.

aaknitt, 2019-12-21:
> The could be some sort of issue around buffering. There are buffers in the pipeline for processing samples. I remember looking into try to reduce the buffer size, the risk is that making the buffers [...]

It seems to happen pretty regularly for me so if you have any thoughts on what to look for, log, or modify for troubleshooting purposes let me know.  I'm hoping to try to dig into and understand the source a bit better over the holidays to try to track this down, but I think it'll be a pretty steep learning curve so if you have any hints to point me in the right direction it would be appreciated.  I'm a newbie when it comes to GNUradio and P25.  I did have a few messages with the author of the boatbod fork of OP25 (seems to be a very popular one) and he did mention he had do make some changes to address audio buffering issues for that package.

robotastic, 2019-12-21:
Hmm... I am using the Boatbod fork in Trunk Recorder. 

There is a little bit about buffers here: https://www.gnuradio.org/blog/2017-01-05-buffers/

I think used to have some code in with set_max_output_buffer... but I can't find that code now. I don't think it is in there anymore. It should be in some of the older commits. 

https://www.gnuradio.org/doc/doxygen/classgr_1_1block.html#a711fb69e35fb871e9a5e41d6a1f2afb1

patrickfish, 2019-12-24:
I also experience this on a HackRF (rate is tuned down to 5000000.0) on an i7-870 machine. 9 digital recorders with 2 analog recorders.

I found some buffer code in 95f33fb...I just built with this added back and will see if that has any improvements.

bctrainers, 2019-12-25:
From the looks of it, the `set_max_output_buffer` commit was also modified back on Jul 24 2016 here: https://github.com/robotastic/trunk-recorder/commit/9a4fe7529e81a9b8947d3fe17c838f453cee6dfc - but this appears to be only related to output/stdout logging.

Original buffers were commended out here: https://github.com/robotastic/trunk-recorder/commit/bde3fc27c2d328007af5e3f015cdd9af825d9e6a

robotastic, 2019-12-25:
Good find! Those look like the buffers. I think I was playing around with the buffer size there.

I also tried to setup something to clear out queues... maybe. 
Try uncommenting this: https://github.com/robotastic/trunk-recorder/blob/668fe62224ef7fb772d4263ed6f8c773683c0626/trunk-recorder/recorders/p25_recorder.cc#L297

dougdever, 2019-12-25:
I suspect bctrainers found it... I deliberately rolled back to older code on a couple of very busy systems because it was so bad.

patrickfish, 2019-12-27:
Seems like I've had better results with adding the buffers back in at 4096. I also uncommented this:

https://github.com/robotastic/trunk-recorder/blob/668fe62224ef7fb772d4263ed6f8c773683c0626/trunk-recorder/recorders/p25_recorder.cc#L297

with no adverse results. I was able to increase the rate on my HackRF since it turned out to be a buffer issue.

bctrainers, 2019-12-28:
> Seems like I've had better results with adding the buffers back in at 4096. I also uncommented this: [...]

Out of curiosity, where did you place in the code at on the files? Have a diff file by chance?

patrickfish, 2019-12-28:
> Out of curiosity, where did you place in the code at on the files? Have a diff file by chance?

Here's the diff. The audio cutting off and wrong talkgroup audio has reduced greatly but I still notice it from time to time. I also bumped my HackRF rate back up when I changed these values so that may be contributing -- system load has increased but I still have plenty of CPU leftover. 

```
$ git diff
diff --git a/trunk-recorder/recorders/p25_recorder.cc b/trunk-recorder/recorders/p25_recorder.cc
index a64f795..3e0ea01 100644
--- a/trunk-recorder/recorders/p25_recorder.cc
+++ b/trunk-recorder/recorders/p25_recorder.cc
@@ -294,7 +294,7 @@ void p25_recorder::set_tdma(bool phase2) {
 }

 void p25_recorder::clear() {
-  //op25_frame_assembler->clear();
+  op25_frame_assembler->clear();
 }

 void p25_recorder::autotune() {
diff --git a/trunk-recorder/systems/p25_trunking.cc b/trunk-recorder/systems/p25_trunking.cc
index 00985bb..4a7f4f3 100644
--- a/trunk-recorder/systems/p25_trunking.cc
+++ b/trunk-recorder/systems/p25_trunking.cc
@@ -137,6 +137,20 @@ p25_trunking::p25_trunking(double f, double c, long s, gr::msg_queue::sptr queue
   bool do_crypt              = 0;
   op25_frame_assembler = gr::op25_repeater::p25_frame_assembler::make(sys_num, idle_silence, wireshark_host, udp_port, verbosity, do_imbe, do_output, do_msgq, rx_queue, do_audio_output, do_tdma, do_crypt);

+       to_float->set_max_output_buffer(4096);
+       rescale->set_max_output_buffer(4096);
+       slicer->set_max_output_buffer(4096);
+       op25_frame_assembler->set_max_output_buffer(4096);
+       arb_resampler->set_max_output_buffer(4096);
+       fm_demod->set_max_output_buffer(4096);
+       baseband_amp->set_max_output_buffer(4096);
+       sym_filter->set_max_output_buffer(4096);
+       fsk4_demod->set_max_output_buffer(4096);
+       agc->set_max_output_buffer(4096);
+       costas_clock->set_max_output_buffer(4096);
+       diffdec->set_max_output_buffer(4096);
+
+
   if (!qpsk_mod) {
     connect(self(),        0, prefilter,            0);
     connect(prefilter,     0, channel_lpf,          0);
```

aaknitt, 2020-01-04:
I've been able to do some more detailed analysis of audio and logs and I think I have a theory as to what's going on.  I made the same changes as @patrickfish describes above and like him, found that adding in the set_max_output_buffer(4096) did help but did not completely eliminate the problem.  

The main discovery from reviewing audio and log files is that whenever a transmission gets cut off early, the audio from the end of that transmission ends up in whatever talkgroup gets recorded by the same recorder later.  So, for example, if recorder 3 is in use to record talkgroup 87 and that transmission gets cut short, the end of the audio from the talkgroup 87 transmission will end up at the beginning of the transmission in whatever talkgroup ends up using recorder 3 next.  This is how audio ends up in the wrong talkgroup.  

I suspect this is being caused by the control channel and voice channels being decoded at different points in time asynchronously as they are being decoded by different processes (or maybe threads, I'm not sure) by the host PC, and not in exactly real time.  I think the control channel is getting decoded more quickly and the voice recorder is getting stopped (based on decoded control channel data) before the entire IMBE stream from the voice channel is fully decoded and sent to the WAV file.  The remaining IMBE data/audio is still stuck in the buffer for that recorder and then gets dumped into the next talkgroup that happens to use the same recorder. 

I'm still too green to be able to figure out a solution.  One answer is to use a higher power PC so that the real-time latency is lower.  However, this is not an optimal solution...if this truly is just a time-synch issue among processes/threads, there should be a way to get them synced back up and allow good operation on a lower-end PC.  One possible solution(at least for P25) could be to have the voice recorder monitor the Link Control Word in LDU1 in a voice superframe, which contains the TGID and continue putting out data as long as the TGID in the voice frame matches the TGID from the control channel.

I wonder if for P25 systems you could even set up a trunk-recorder system to not even monitor the control channel if you have enough recorders to cover each voice channel...just decode all voice channels and get TGID info directly from the Link Control Word.

robotastic, 2020-01-04:
@aaknitt my assumption has been that there are samples left in all of the different buffers on the recorder flowgraph. When one recording ends, there may still be some received transmission samples left in the buffers. When then next one starts, those old ones are pushed through and found at the beginning of the next recording. Reducing the buffer size between different GR blocks, should reduce left over cruft. 

I am not sure of a good way to really solve it. ideally you could shove some noise into the graph to push out any left over good signals. This would be tough to do though because it would mean stopping the graph and reconfiguring it.

One other thing to try is making **callTimeout** longer than 3 seconds. This would keep the Recorder going for longer on a channel, even after there have been messages on the control channel saying this channel is being used. This should be fine as long as that channel doesn't immediately start being used for another transmission. Maybe try it at 5 seconds.

dandrzejewski, 2020-01-04:
> One other thing to try is making **callTimeout** longer than 3 seconds. This would keep the Recorder going for longer on a channel, even after there have been messages on the control channel saying [...]

I am trying this with 8 seconds on a Raspberry Pi 4 on a moderately busy system that has this trouble, and I will report back.

patrickfish, 2020-01-04:
Thanks Luke -- do the buffers have a default value? I've added them back at 4096 which seemed to have helped but didn't fully solve the issue. I know they were at 512 at one point before they were removed and I haven't tried that yet.

I will play around with callTimeout first and see what happens.

aaknitt, 2020-02-12:
Fixed by upgrading to 3.1.2

closed 2020-02-12 by aaknitt