TrunkRecorder/trunk-recorder #285: Unable to setup a system

rpdale, 2019-12-22:
I had a SD card go bad, and of all my Pi's that are running, somehow this is the one I never backed up! So hardware is all the same, I installed a new system using the same settings in config.json and everything looks good - until the end?

config.json
{
    "sources": [{
        "center": 852362500.0,
        "rate": 2040000.0,
        "error": 1000,
        "gain": 48,
        "digitalRecorders": 5,
        "modulation": "fsk4",
        "driver": "osmosdr",
        "device": "rtl=1"
    }, {
        "center": 770543750.0,
        "rate": 1800000,
        "error": -32000,
        "gain": 48,
        "digitalRecorders": 3,
        "modulation": "fsk4",
        "driver": "osmosdr",
        "device": "rtl=2"
    }],
    "systems": [{
        "control_channels": [853475000],
	"talkgroupsFile": "ChanList.csv",
        "type": "p25",
	"shortName": "mpscs"
    }],
    "captureDir": "/home/pi/mpscs/"
}

[2019-12-22 09:27:20.448843] (info)   Using Config file: ./config.json

[2019-12-22 09:27:20.449949] (info)   
-------------------------------------
SYSTEMS
-------------------------------------
0
[2019-12-22 09:27:20.450171] (info)   

System Number: 1
-------------------------------------
1
[2019-12-22 09:27:20.450290] (info)   Short Name: mpscs
[2019-12-22 09:27:20.450323] (info)   System Type: p25
[2019-12-22 09:27:20.450355] (info)   Control Channels: 
[2019-12-22 09:27:20.450468] (info)   8.53475e+08 
[2019-12-22 09:27:20.450530] (info)   API Key: 
[2019-12-22 09:27:20.450556] (info)   Upload Script: 
[2019-12-22 09:27:20.450581] (info)   Call Log: true
[2019-12-22 09:27:20.450615] (info)   Audio Archive: true
[2019-12-22 09:27:20.450641] (info)   Loading Talkgroups...
[2019-12-22 09:27:20.451157] (info)   Read 43 talkgroups.
[2019-12-22 09:27:20.451197] (info)   Talkgroups File: ChanList.csv
[2019-12-22 09:27:20.451225] (info)   Record Unknown Talkgroups: true
[2019-12-22 09:27:20.451261] (info)   Talkgroup Display Format: Id
[2019-12-22 09:27:20.451292] (info)   Hide Encrypted Talkgroups: false
[2019-12-22 09:27:20.451318] (info)   Hide Unknown Talkgroups: false
[2019-12-22 09:27:20.451343] (info)   Minimum Call Duration (in seconds): 0
[2019-12-22 09:27:20.451373] (info)   
[2019-12-22 09:27:20.451402] (info)   

-------------------------------------
SOURCES
-------------------------------------

[2019-12-22 09:27:20.451487] (info)   Driver: osmosdr
[2019-12-22 09:27:20.451513] (info)   Center: 8.523625e+08
[2019-12-22 09:27:20.451568] (info)   Rate: 2040000
[... 103 lines cut ...]
[2019-12-22 09:27:22.715217] (info)   	 P25 Recorder Initial Rate: 600000 Resampled Rate: 30000 Initial Decimation: 3 Decimation: 20 System Rate: 30000 ARB Rate: 1
[2019-12-22 09:27:22.715435] (info)   P25 Recorder Taps - initial: 273 channel: 1819 ARB: 727 Total: 2819
[2019-12-22 09:27:22.724933] (info)   
-------------------------------------

[2019-12-22 09:27:22.725114] (info)   

-------------------------------------
INSTANCE
-------------------------------------

[2019-12-22 09:27:22.725314] (info)   Capture Directory: /home/pi/mpscs
[2019-12-22 09:27:22.725375] (info)   Upload Server: 
[2019-12-22 09:27:22.725421] (info)   Status Server: 
[2019-12-22 09:27:22.725448] (info)   Instance Key: 
[2019-12-22 09:27:22.725474] (info)   Instance Id: 
[2019-12-22 09:27:22.725500] (info)   Default Mode: digital
[2019-12-22 09:27:22.725528] (info)   Call Timeout (seconds): 3
[2019-12-22 09:27:22.725556] (info)   Log to File: false
[2019-12-22 09:27:22.725584] (info)   Control channel warning rate: 10
[2019-12-22 09:27:22.725613] (info)   Control channel retune limit: 0
[2019-12-22 09:27:22.725642] (info)   Frequency format: 0
[2019-12-22 09:27:22.725669] (info)   Status as String: false
[2019-12-22 09:27:22.725712] (info)   [mpscs]	Started with Control Channel: 8.534750e+08
[2019-12-22 09:27:22.725784] (error)   Unable to setup a System to record, exiting...

gofaster, 2019-12-22:
The control channel freq is above the max frequency of your source.  Either increase it's bandwidth/sample rate or change the center freq.

Using device #0 Realtek RTL2838UHIDIR SN: 1
...
[2019-12-22 09:27:21.217735] (info) Setting sample rate to: 2040000
Exact sample rate is: 2040000.121593 Hz
[2019-12-22 09:27:21.333608] (info) Tunning to 8.523635e+08
...
[2019-12-22 09:27:21.384554] (info) Max Freqency: 8.533825e+08  
[2019-12-22 09:27:21.384618] (info) Min Freqency: 8.513425e+08
...
[2019-12-22 09:27:22.725712] (info) [mpscs] Started with Control Channel: 8.534750e+08
[2019-12-22 09:27:22.725784] (error) Unable to setup a System to record, exiting...

bctrainers, 2019-12-22:
@gofaster  is correct. Your RTL-SDR's frequency bandwidth isn't large enough to capture the control channel and the rest of the desired frequencies. 

There are multiple 853.475 MHz control channels on the MPSC P25 system, so I am unsure which one you're trying to connect and lock to.

rpdale, 2019-12-22:
Ugh - awkward :) Thanks. That must have been an old config file I backed up before I was all set. I've expanded the rate just a bit (using site 1 - 001) and all is good. Now I just have to recreate all my priorities in the talkgroup file.

closed 2019-12-22 by rpdale