TrunkRecorder/trunk-recorder #288: Compiling with GNUradio 3.8 fails

vabiro, 2019-12-29:
Hi,

I have been having a recurring problem compiling trunk-recorder since the repository has upgraded gnuradio to 3.8.

It continues to die at 69% while looking for gnuradio/blocks/multiply_const_ff.h

I have checked the  gnuradio repository and that file no longer exists after 3.7. I have tried rolling back the version to 3.7, but it continues to give errors. 

Here is the error that I am getting:
`[ 67%] Building CXX object lib/op25_repeater/lib/CMakeFiles/gnuradio-op25_repeater.dir/CCITTChecksumReverse.cpp.o
[ 68%] Linking CXX shared library libgnuradio-op25_repeater.so
[ 68%] Built target gnuradio-op25_repeater
Scanning dependencies of target recorder
[ 69%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/main.cc.o
In file included from /root/trunk-recorder/trunk-recorder/source.h:11,
                 from /root/trunk-recorder/trunk-recorder/main.cc:42:
/root/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:23:10: fatal error: gnuradio/blocks/multiply_const_ff.h: No such file or directory
   23 | #include <gnuradio/blocks/multiply_const_ff.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/recorder.dir/build.make:63: CMakeFiles/recorder.dir/trunk-recorder/main.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:156: CMakeFiles/recorder.dir/all] Error 2
make: *** [Makefile:130: all] Error 2`

What am I missing here? Has anyone else managed to compile with 3.8?
I have been wrestling with this for weeks, and can't get past it. 

HELP

bctrainers, 2019-12-29:
What flavor of *nix by chance?

May need to add in some code to analog_recorder.h and the digital/p25 side of things.

```cpp
#if GNURADIO_VERSION < 0x030800
#include <gnuradio/analog/sig_source_c.h>
#include <gnuradio/blocks/multiply_const_ff.h>
#include <gnuradio/blocks/multiply_cc.h>
#else
#include <gnuradio/analog/sig_source.h>
#include <gnuradio/blocks/multiply_const.h>
#include <gnuradio/blocks/multiply.h>
#endif
```

vabiro, 2019-12-29:
Thanks

I am using ArchLinux 5.4.6-arch3-1.

Where would I add the above to the files?

bctrainers, 2019-12-29:
Try this code in `./trunk-recorder/recorders/analog_recorder.h`

```cpp
#if GNURADIO_VERSION < 0x030800
#include <gnuradio/blocks/multiply_const_ff.h>
#else
#include <gnuradio/blocks/multiply_const.h>
#endif 
```

Replacing the [code at Line 23](https://github.com/robotastic/trunk-recorder/blob/master/trunk-recorder/recorders/analog_recorder.h#L23)

You may need to repeat the same on recorder.h, sigmf_recorder.h and in p25_recorder.h.

Oddly enough, p25_recorder.h has two include instances of `gnuradio/blocks/multiply_const_ff.` - am unsure why. Should be save removing the second instance of `#include <gnuradio/blocks/multiply_const_ff.h>` within p25_recorder.h.

Do note the above is very much so untested - so am not sure if the above corrections will resolve your problem currently. I don't have a virtual machine built and running with Arch currently.

vabiro, 2019-12-29:
Progress!
I was not not working with the If statement, so I just replaced the include with include <gnuradio/blocks/multiply_const.h>

It is now complaining about fir_filter_fff.h. Again, there are changes, and that file doesn't exist.

Here's the new error:
`[ 68%] Built target gnuradio-op25_repeater
Scanning dependencies of target recorder
[ 69%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/main.cc.o
In file included from /root/trunk-recorder/trunk-recorder/source.h:11,
                 from /root/trunk-recorder/trunk-recorder/main.cc:42:
/root/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:28:10: fatal error: gnuradio/filter/fir_filter_fff.h: No such file or directory
   28 | #include <gnuradio/filter/fir_filter_fff.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/recorder.dir/build.make:63: CMakeFiles/recorder.dir/trunk-recorder/main.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:156: CMakeFiles/recorder.dir/all] Error 2
make: *** [Makefile:130: all] Error 2`

vabiro, 2019-12-29:
I just took a look at the changes that GQRX made to accomodate 3.8, and it looks like there were many. They are [here](https://github.com/csete/gqrx/pull/705/files)
I suspect that the GNURADIO_VERSION is unique to the GQRX code, and populated from:
`# Detect GNU Radio version and link against log4cpp for 3.8
GNURADIO_VERSION = $$system(pkg-config --modversion gnuradio-runtime)
` which I don't know if it is done in trunk-recorder. I think that's why the IF wasn't working

Any thoughts on how we can futureproof trunk-recorder while we are on this odyssey. I suspect I am not going to be the last, if the changes to GQRX are any indication.

robotastic, 2019-12-30:
You guys are a step ahead of me. I do development on a Mac and Gnuradio v3.8 support isn't great. I am just getting a docker environment setup for 3.8 development now. I may use this as an opportunity to restructure the codebase a little.

vabiro, 2019-12-30:
@robotastic 
Thank you very much. This has confounded me for the past month, and I would appreciate your , and @bctrainers help. I think we could all benefit.

Cheers
Victorr

vabiro, 2020-01-14:
Hi guys,

I am without a working version of trunk-recorder, and I wonder - since others seem to have it working - if there is a *nix version that works. I am not wedded to Arch, and would happily build something that is working.

Any suggestions?

Thanks

daryljones, 2020-01-14:
I had the same problem on Ubuntu. Had to go back to the previous version.

robotastic, 2020-01-16:
Ok - I think I have a branch that atleast builds for GR 3.8. I still have to actually test it out, but I need to move to a linux box to do that. Try the `gr3.8` branch and see if it atleast compiles.

daryljones, 2020-01-16:
@robotastic   I'm using Ubuntu 18.04 and will try to build the gr3.8 branch tomorrow.

vabiro, 2020-01-16:
> Ok - I think I have a branch that atleast builds for GR 3.8. I still have to actually test it out, but I need to move to a linux box to do that. Try the `gr3.8` branch and see if it atleast compiles.

Hi,

I just tried the gr3.8 branch, and received a bunch of new errors at the same spot. There was a significant number of line, so I'll just quote the first few dozen:

> [ 68%] Built target gnuradio-op25_repeater
[ 69%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/main.cc.o
In file included from /root/trunk-recorder/lib/../trunk-recorder/uploaders/../systems/p25_trunking.h:46,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/../systems/system.h:8,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/uploader.h:19,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/call_uploader.h:4,
                 from /root/trunk-recorder/lib/../trunk-recorder/call.h:31,
                 from /root/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink.h:26,
                 from /root/trunk-recorder/trunk-recorder/recorders/recorder.h:56,
                 from /root/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:47,
                 from /root/trunk-recorder/trunk-recorder/source.h:11,
                 from /root/trunk-recorder/trunk-recorder/main.cc:42:
/root/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h: In member function ‘virtual float gr::op25_repeater::gardner_costas_cc::get_freq_error()’:
/root/trunk-recorder/lib/op25_repeater/include/op25_repeater/gardner_costas_cc.h:51:43: warning: no return statement in function returning non-void [-Wreturn-type]
   51 |       virtual float get_freq_error(void) {}
      |                                           ^
In file included from /usr/include/boost/asio/executor.hpp:338,
                 from /usr/include/boost/asio/basic_socket.hpp:27,
                 from /usr/include/boost/asio/basic_datagram_socket.hpp:20,
                 from /usr/include/boost/asio.hpp:24,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/uploader.h:8,
                 from /root/trunk-recorder/lib/../trunk-recorder/uploaders/call_uploader.h:4,
                 from /root/trunk-recorder/lib/../trunk-recorder/call.h:31,
                 from /root/trunk-recorder/lib/gr_blocks/nonstop_wavfile_sink.h:26,
                 from /root/trunk-recorder/trunk-recorder/recorders/recorder.h:56,
                 from /root/trunk-recorder/trunk-recorder/recorders/analog_recorder.h:47,
                 from /root/trunk-recorder/trunk-recorder/source.h:11,
                 from /root/trunk-recorder/trunk-recorder/main.cc:42:
/usr/include/boost/asio/impl/executor.hpp: In instantiation of ‘boost::asio::execution_context& boost::asio::executor::impl< <template-parameter-1-1>, <template-parameter-1-2> >::context() [with Executor = std::reference_wrapper<boost::asio::io_context>; Allocator = std::allocator<void>]’:
/usr/include/boost/asio/impl/executor.hpp:177:22:   required from here
/usr/include/boost/asio/impl/executor.hpp:179:22: error: ‘class std::reference_wrapper<boost::asio::io_context>’ has no member named ‘context’
  179 |     return executor_.context();
      |            ~~~~~~~~~~^~~~~~~
/usr/include/boost/asio/impl/executor.hpp: In instantiation of ‘void boost::asio::executor::impl< <template-parameter-1-1>, <template-parameter-1-2> >::on_work_started() [with Executor = std::reference_wrapper<boost::asio::io_context>; Allocator = std::allocator<void>]’:
/usr/include/boost/asio/impl/executor.hpp:167:8:   required from here
/usr/include/boost/asio/impl/executor.hpp:169:15: error: ‘class std::reference_wrapper<boost::asio::io_context>’ has no member named ‘on_work_started’
  169 |     executor_.on_work_started();
      |     ~~~~~~~~~~^~~~~~~~~~~~~~~
/usr/include/boost/asio/impl/executor.hpp: In instantiation of ‘void boost::asio::executor::impl< <template-parameter-1-1>, <template-parameter-1-2> >::on_work_finished() [with Executor = std::reference_wrapper<boost::asio::io_context>; Allocator = std::allocator<void>]’:
/usr/include/boost/asio/impl/executor.hpp:172:8:   required from here
/usr/include/boost/asio/impl/executor.hpp:174:15: error: ‘class std::reference_wrapper<boost::asio::io_context>’ has no member named ‘on_work_finished’
  174 |     executor_.on_work_finished();
      |     ~~~~~~~~~~^~~~~~~~~~~~~~~~
/usr/include/boost/asio/impl/executor.hpp: In instantiation of ‘void boost::asio::executor::impl< <template-parameter-1-1>, <template-parameter-1-2> >::dispatch(boost::asio::executor::function&&) [with Executor = std::reference_wrapper<boost::asio::io_context>; Allocator = std::allocator<void>]’:
/usr/include/boost/asio/impl/executor.hpp:182:8:   required from here
/usr/include/boost/asio/impl/executor.hpp:184:15: error: ‘class std::reference_wrapper<boost::asio::io_context>’ has no member named ‘dispatch’
  184 |     executor_.dispatch(BOOST_ASIO_MOVE_CAST(function)(f), allocator_);
      |     ~~~~~~~~~~^~~~~~~~
/usr/include/boost/asio/impl/executor.hpp: In instantiation of ‘void boost::asio::executor::impl< <template-parameter-1-1>, <template-parameter-1-2> >::post(boost::asio::executor::function&&) [with Executor = std::reference_wrapper<boost::asio::io_context>; Allocator = std::allocator<void>]’:
/usr/include/boost/asio/impl/executor.hpp:187:8:   required from here
[... cut ...]
/usr/include/boost/optional/detail/optional_relops.hpp:165:6: note: candidate: ‘template<class T> bool boost::operator==(boost::none_t, const boost::optional<T>&)’
  165 | bool operator == ( none_t , optional<T> const& y ) BOOST_NOEXCEPT
      |      ^~~~~~~~
/usr/include/boost/optional/detail/optional_relops.hpp:165:6: note:   template argument deduction/substitution failed:
In file included from /usr/include/boost/asio/executor.hpp:338,
                 from /usr/include/boost/asio/basic_socket.hpp:27,

robotastic, 2020-01-16:
> boost::asio::executor::impl< , >::on_work_started() [with Executor =
> boost::asio::executor::impl< , >::on_work_finished() [with Executor =
> boost::asio::executor::impl< , >::post(boost::asio::executor::function&&)
> boost::asio::executor::impl< , >::defer(boost::asio::executor::function&&)
> boost::asio::executor::impl< , >::equals(const
> candidate: ‘template bool boost::operator==(const boost::optional&, const
> 28 | bool operator == ( optional const& x, optional const& y )
> 62 | bool operator == ( optional const& x, T const& y )
> candidate: ‘template bool boost::operator==(const T&, const
> 96 | bool operator == ( T const& x, optional const& y )
> candidate: ‘template bool boost::operator==(const boost::optional&,
> 131 | bool operator == ( optional const& x, none_t ) BOOST_NOEXCEPT
> candidate: ‘template bool boost::operator==(boost::none_t, const
> 165 | bool operator == ( none_t , optional const& y ) BOOST_NOEXCEPT

Which linux distro and Boost version are you using? I will get the Docker
container I am using for development up online tonight.

vabiro, 2020-01-16:
I'm using ArchLinux Boost 1.71.0 and gnuradio 3.8.0.0-6

closed 2020-08-04 by robotastic