TrunkRecorder/trunk-recorder #289: P25 on Analog channels

JoeGilkey, 2019-12-30:
A few years ago Motorola introduced something called TPS (Tactical Public Safety), which is a P25 header (Unit ID, etc) over an analog channel. This is the signaling that FDNY uses. I’m wondering how hard it would be to implement this on the Analog decoder channel. I’m thinking that the analog channel would need to pass the audio into the P25 frame slicer to start the decoder chain. Has anyone tried this before, or does anyone have any ideas?

Thanks,
Joe

robotastic, 2020-01-29:
If anyone has some GnuRadio code that does this, I would be happy to try and integrate it.

JoeGilkey, 2020-02-03:
@robotastic I figured out how to do this. I just need to connect the FM demodulator output to the FSK4 slicer and op25_frame_assembler, then parse the messages. I’ll clean up the code and add a pull request after #299 is in since it relies on that code.

JoeGilkey, 2020-02-03:
FWIW, this is the signaling system that FDNY uses.

closed 2020-07-31 by robotastic