TrunkRecorder/trunk-recorder #290: Airspy Gain Settings

wx-jc, 2020-01-04:
Trunk Recorder doesn't appear to support the 3 individual gain settings of the Airspy.  Maybe I'm just doing something wrong?  Instead, only "gain" works.

**Working config:**
{
        "sources": [{
                "center": 854543750,
                "rate": 10000000,
                "gain": 8,
                "digitalRecorders": 11,
                "driver": "osmosdr",
                "modulation": "qpsk",
                "device": "airspy",
                "error": 0.2

**Bad config:**

{
        "sources": [{
                "center": 854543750,
                "rate": 10000000,
                "ifGain": 8,
                "lnaGain": 8,
                "mixGain": 8,
                "digitalRecorders": 11,
                "driver": "osmosdr",
                "modulation": "qpsk",
                "device": "airspy",
                "error": 0.2

The documentation mentions support for mixGain and lnaGain.  Those don't work on their own either.  GQRX shows mix, LNA, and IF gains supported.

The regular gain setting works fine, but I think people (including me when I first set up TR) get tripped up on this with the Airspy.

n71fs, 2020-01-04:
This tripped me up as well, also I had issues using "ppm" adjustment and just ended up using "error" to do the same thing.

robotastic, 2020-01-08:
I will look through it tonight. I don't have an AirSpy, so I was just using code that someone submitted. It maybe that the way the driver handles gain has changed. Do the gain settings work in GQRX? If so, I can try to copy that code.

wx-jc, 2020-01-08:
Hi Luke,

Yes - the gain settings in GQRX work correctly.

Thanks!

Jarrod

robotastic, 2020-01-15:
Hmm... I am making the same calls as GQRX. If you check here, the values should be configurable: https://github.com/robotastic/trunk-recorder/blob/3e4ed87eac7534e4584569af3320576f02c79537/trunk-recorder/main.cc#L302
Maybe try including a `gain` setting too. If one is not included, it will default to zero which might cancel out the other settings.

wx-jc, 2020-01-15:
Weird.  

Using the regular "gain" is the only way to get it to work.  I assume it makes the other 3 settings the same value.  Luckily setting just a standard gain has been working fine for me.  It just threw me off when I first set things up a year ago.

robotastic, 2020-01-15:
I could be wrong, but I believe Gain is still a valid setting on the AirSpy. The OsmoSDR driver maybe mapping it to one of other gain settings. The other gain settings should work though and a probably worth setting on the AirSpy.

JoeGilkey, 2020-01-16:
It looks like the gain setting is being applied after the other AirSpy settings. This is probably causing the other settings to be overwritten. The gain should probably be moved to before the detailed gain settings.

robotastic, 2020-01-24:
I reverted this change - the detailed gain was clobbering the general gain and setting it to zero. I think I have to add tests around the detailed gains before setting them. I will test this out.

robotastic, 2020-01-29:
Ok - give the latest release on the master branch a try. All of the gain settings only get applied if they are set. Try leaving Gain unset, it looks like the AirSpy driver might be doing something extra when it is set.

wx-jc, 2020-01-29:
No improvement for me...did a 'git pull' and used the following for the config.  Had to add "gain" back in to decode (with a value of '8').

{
        "sources": [{
                "center": 854543750,
                "rate": 10000000,
                "mixGain": 8,
                "lnaGain": 8,
                "ifGain": 8, 
                "digitalRecorders": 11,
                "driver": "osmosdr",
                "modulation": "qpsk",
                "device": "airspy",
                "error": 0.2
        }],

Here is a pic of using the same settings in GQRX:
[image]

robotastic, 2020-01-29:
Ah! So the problem is actually around the Error / PPM. Error is specified as the amount of Hz the tuning is off by. This is the more reliable way of doing it. There is some more info here: https://github.com/robotastic/trunk-recorder/wiki/Configuring-a-System <https://github.com/robotastic/trunk-recorder/wiki/Configuring-a-System>

You can also try using ppm (parts per million) of error. You can do that by removing error from config and replacing it with:

“ppm”: 0.2,

wx-jc, 2020-02-01:
I agree I wasn't using error correctly...I changed it to to PPM but this doesn't seem to fix the issue with the gain settings.  I still have to use regular gain instead of the individual ones.

closed 2024-12-15 by robotastic