TrunkRecorder/trunk-recorder #291: OpenMHz Length Field Inaccurate

wx-jc, 2020-01-05:
For a while now, the "Len" column on the OpenMHz site has been showing much longer lengths than the actual call duration.  However, the counter near the bottom of the screen is still good.

robotastic, 2020-01-08:
yea - I had to change the way OpenMHz calculated recording time. Previously, it has been calling FFMPEG on the server to analyze how long each M4A file was. There was some bug in the node module and there was a memory leak. Now I am just calculating recording length based on when a recorder started and stopped. If I move this calculation to Trunk Recorder I might be able to add this back.

closed 2020-01-08 by robotastic

wx-jc, 2020-02-15:
Analog recorders are WAY worse, so is this something you can look into?  Check out this example...The "Washington PD" recording was only a few seconds long.  

[image]