TrunkRecorder/trunk-recorder #292: alphatags

dh47, 2020-01-15:
How would I use alphatags to label the below freqs, for 141412500 North and 141487500 for South.
```
    "systems": [
        {
            "type": "conventional",
            "channels": [141412500, 141487500],
            "shortName": "Ambo",
			"minDuration": 4,
            "callLog": false,
			"defaultMode": "analog",
            "audioArchive": true
```

dh47, 2020-01-23:
Surely someone could help with this?  I can't find an example being used in any of the configs, and the readme while it does show the option to use it, doesn't specify how to use it.

gofaster, 2020-01-23:
```
"systems": [
    {
        "type": "conventional",
        "channels": [141412500, 141487500],
         "alphatags: ["North", "South"],
         "shortName": "Ambo",
         "minDuration": 4,
         "callLog": false,
	  "defaultMode": "analog",
          "audioArchive": true
```

closed 2020-01-26 by dh47