TrunkRecorder/trunk-recorder #302: repeated src/unit entries in json

gofaster, 2020-02-04:
Thanks for the continual improvements.  Just dropped in the latest build and am seeing repeated src (P25 unit id) entries in the output json files.

```
{
  "freq": 771356000,
  "start_time": 1580816198,
  "stop_time": 1580816209,
  "emergency": 0,
  "talkgroup": 401,
  "srcList": [
    {
      "src": 61006,
      "time": 1580816198,
      "pos": 0,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": 61006,
      "time": 1580816198,
      "pos": 0,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": 65036,
      "time": 1580816203,
      "pos": 2.7,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": 65036,
      "time": 1580816203,
      "pos": 2.7,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    }
  ],
  "play_length": 3.96,
  "source": 0,
  "freqList": [
    {
      "freq": 771356250,
      "time": 1580816198,
      "pos": 0,
      "len": 35616,
      "error_count": 70,
      "spike_count": 17
    }
  ]
}

```

bctrainers, 2020-02-04:
Didn't catch this on my local git daily-sync build, is this happening on all talkgroup activations?

gofaster, 2020-02-04:
Yes so far, every P25 recording.
```
{
  "freq": 769331000,
  "start_time": 1580816491,
  "stop_time": 1580816517,
  "emergency": 0,
  "talkgroup": 3,
  "srcList": [
    {
      "src": 65039,
      "time": 1580816491,
      "pos": 0,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": 65039,
      "time": 1580816491,
      "pos": 0,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": 1802698,
      "time": 1580816495,
      "pos": 1.62,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": 1802698,
      "time": 1580816495,
      "pos": 1.62,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": 65039,
      "time": 1580816500,
      "pos": 3.42,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": 65039,
      "time": 1580816500,
      "pos": 3.42,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": 1802698,
      "time": 1580816509,
      "pos": 9.9,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    },
    {
      "src": 1802698,
      "time": 1580816509,
      "pos": 9.9,
      "emergency": 0,
      "signal_system": "",
      "tag": ""
    }
  ],
  "play_length": 12.42,
  "source": 0,
  "freqList": [
    {
      "freq": 770956250,
      "time": 1580816491,
      "pos": 0,
      "len": 0,
      "error_count": 0,
      "spike_count": 0
    },
    {
      "freq": 770331250,
      "time": 1580816495,
      "pos": 1.62,
      "len": 0,
      "error_count": 0,
      "spike_count": 0
    },
    {
      "freq": 769656250,
      "time": 1580816500,
      "pos": 3.42,
      "len": 0,
      "error_count": 0,
      "spike_count": 0
    },
    {
      "freq": 769331250,
      "time": 1580816509,
      "pos": 9.9,
      "len": 21984,
      "error_count": 0,
      "spike_count": 0
    }
  ]
}
```

```
$ git show
commit 1770f7783d241b30cbf2d019d74742a226f806b2
Merge: 228e71f 551a8f8
Author: Luke Berndt <lukekb@gmail.com>
Date:   Mon Feb 3 22:08:06 2020 -0500

    Merge pull request #300 from robotastic/pr/299

    Pr/299

```

robotastic, 2020-02-04:
Yep - looks like the check to see if the Source is already on the list got dropped. I added it back in, and made the logic a little clearer. should be good now.

closed 2020-02-04 by robotastic

JoeGilkey, 2020-02-04:
@gofaster @robotastic That was probably an issue from my check-in (wish they had a facepalm emoji).