TrunkRecorder/trunk-recorder #305: Moved from Raspberry Pi to Debian desktop and now getting Control Channel Message Decode Rate errors

CharlesCage, 2020-02-04:
Apologies in advance if I'm missing something weird. I had this working great on a Raspberry PI, but wanted to move to a desktop for increased stability. Config file should be fine as it was working great on the Pi setup. 

I am seeing this, which is different from what I see on the pi build, and worries me:

```
Decim: 16 Decim2:  4
[2020-02-04 13:10:03.523545] (info)      P25 Trunking two-stage decimator - Initial decimated rate: 128000 Second decimated rate: 32000 FA: 6250 FB: 16000 System Rate: 2048000
[2020-02-04 13:10:03.527284] (info)      P25 Trunking ARB - Initial Rate: 2048000 Resampled Rate: 32000 Initial Decimation: 16 System Rate: 24000 ARB Rate: 0.75

```

Things I've tried so far:
- rtl_eeprom shows all radios present and serials correct
- I've run rtl_test on all four radios simultaneously and am not seeing errors
- I've run rtl_tcp on each radio individually to listen to them with SDR# and can see/hear the control channel/have confirmed the error for each radio (visually aligning signal and doing the freq math)

I just get this sinking feeling that I have some kind of mis-match in a utility or something that's b0rking up the digital signal...

Full output from running recorder:

```
[2020-02-04 13:10:01.266610] (info)   Using Config file: ./config.json

[2020-02-04 13:10:01.267004] (info)
-------------------------------------
SYSTEMS
-------------------------------------
0
[2020-02-04 13:10:01.267036] (info)

System Number: 1
-------------------------------------
1
[2020-02-04 13:10:01.267050] (info)   Short Name: frisco
[2020-02-04 13:10:01.267069] (info)   System Type: p25
[2020-02-04 13:10:01.267081] (info)   Control Channels:
[2020-02-04 13:10:01.267114] (info)   8.59612e+08
[2020-02-04 13:10:01.267174] (info)   API Key:
[2020-02-04 13:10:01.267181] (info)   Upload Script:
[2020-02-04 13:10:01.267188] (info)   Call Log: true
[2020-02-04 13:10:01.267218] (info)   Audio Archive: true
[2020-02-04 13:10:01.267225] (info)   Loading Talkgroups...
[2020-02-04 13:10:01.267371] (info)   Read 44 talkgroups.
[2020-02-04 13:10:01.267398] (info)   Talkgroups File: frisco-talkgroups.csv
[2020-02-04 13:10:01.267406] (info)   Record Unknown Talkgroups: true
[2020-02-04 13:10:01.267417] (info)   Talkgroup Display Format: Id
[2020-02-04 13:10:01.267436] (info)   Hide Encrypted Talkgroups: false
[2020-02-04 13:10:01.267456] (info)   Hide Unknown Talkgroups: false
[2020-02-04 13:10:01.267463] (info)   Minimum Call Duration (in seconds): 0
[2020-02-04 13:10:01.267473] (info)
[2020-02-04 13:10:01.267486] (info)

-------------------------------------
SOURCES
-------------------------------------

[2020-02-04 13:10:01.267511] (info)   Driver: osmosdr
[2020-02-04 13:10:01.267533] (info)   Center: 8.585937e+08
[2020-02-04 13:10:01.267588] (info)   Rate: 2048000
[2020-02-04 13:10:01.267611] (info)   Error: -19500
[2020-02-04 13:10:01.267643] (info)   PPM Error: 0
[2020-02-04 13:10:01.267655] (info)   Gain: 47
[2020-02-04 13:10:01.267667] (info)   IF Gain: 0
[2020-02-04 13:10:01.267678] (info)   BB Gain: 0
[2020-02-04 13:10:01.267688] (info)   LNA Gain: 0
[2020-02-04 13:10:01.267699] (info)   PGA Gain: 0
[2020-02-04 13:10:01.267710] (info)   TIA Gain: 0
[2020-02-04 13:10:01.267720] (info)   MIX Gain: 0
[2020-02-04 13:10:01.267730] (info)   VGA1 Gain: 0
[2020-02-04 13:10:01.267741] (info)   VGA2 Gain: 0
[... 238 lines cut ...]
            "device": "rtl=33,buflen=65536"
        },
        {
            "center": 856406200.0,
            "rate": 2048000.0,
            "error": -13500,
            "gain": 47,
            "digitalRecorders": 2,
            "modulation": "qpsk",
            "driver": "osmosdr",
            "device": "rtl=44,buflen=65536"
        }
        ],
    "systems": [{
        "control_channels": [859612500],
        "type": "p25",
        "talkgroupsFile": "frisco-talkgroups.csv",
        "shortName": "frisco",
        "uploadScript": "encode-local-sys-0.sh"
    }],
    "captureDir": "/home/chuckcage/p25files/"

}

```

CharlesCage, 2020-02-04:
Confirmed that the required installed packages are all the same versions as they were on the pi. In case it's useful:

```
autoconf/stable,now 2.69-11 all [installed]
automake/stable,now 1:1.16.1-4 all [installed]
build-essential/stable,now 12.6 amd64 [installed]
cmake/stable,now 3.13.4-1 amd64 [installed]
git/stable,now 1:2.20.1-2+deb10u1 amd64 [installed]
gnuradio-dev/stable,now 3.7.13.4-4+b1 amd64 [installed]
gnuradio/stable,now 3.7.13.4-4+b1 amd64 [installed]
gr-osmosdr/stable,now 0.1.4-14+b10 amd64 [installed]
libass-dev/stable,now 1:0.14.0-2 amd64 [installed]
libboost-all-dev/stable,now 1.67.0.1 amd64 [installed]
libcppunit-dev/stable,now 1.14.0-3 amd64 [installed]
libfreetype6-dev/stable,now 2.9.1-3+deb10u1 amd64 [installed]
libhackrf-dev/stable,now 2018.01.1-2 amd64 [installed]
liblog4cpp5-dev/stable,now 1.1.3-1 amd64 [installed]
libsdl2-dev/stable,now 2.0.9+dfsg1-1 amd64 [installed]
libssl-dev/stable,stable,now 1.1.1d-0+deb10u2 amd64 [installed]
libtheora-dev/stable,now 1.1.1+dfsg.1-15 amd64 [installed]
libtool/stable,now 2.4.6-9 all [installed]
libuhd-dev/stable,now 3.13.1.0-3 amd64 [installed]
libva-dev/stable,now 2.4.0-1 amd64 [installed]
libvdpau-dev/stable,now 1.1.1-10 amd64 [installed]
libvorbis-dev/stable,now 1.3.6-2 amd64 [installed]
libxcb-shm0-dev/stable,now 1.13.1-2 amd64 [installed]
libxcb-xfixes0-dev/stable,now 1.13.1-2 amd64 [installed]
libxcb1-dev/stable,now 1.13.1-2 amd64 [installed]
pkg-config/stable,now 0.29-6 amd64 [installed]
texinfo/stable,now 6.5.0.dfsg.1-4+b1 amd64 [installed]
zlib1g-dev/stable,now 1:1.2.11.dfsg-1 amd64 [installed]

```

gofaster, 2020-02-04:
Those decimation messages are informational, from improvements that robotastic recently made to the signal flow.

You mentioned that the config file was working great on the Pi; but..
Your decode rate is low, and the error values you've defined for your rtl-sdr's seem large.  Perhaps you can double check with op25; I've have had good results with this method vs eyeballing in SDR# or GQRX.  Dial in the ppm and fine-tune error for each rtl-sdr and transfer the results to the 'PPM' and/or  'Error' definitions for each source in trunk-recorder.  PPM is a multiplier while Error is absolute.
https://forums.radioreference.com/threads/rpi-3-rtl-sdr-op25.354875/page-26#post-3029817

Also, your gain value is very high.  Are you sure you need all that gain?  It could be raising your noise floor, clipping your signal and decreasing overall s/n ratio.

CharlesCage, 2020-02-04:
Ok -- installed op25 and am seeing control channel decoding with the following:

Radio 0:
` ./rx.py --args 'rtl=11' -N 'LNA:47' -S 2400000 -f 859.6125e6 -o 25000 -q 22`

Radio 1:
` ./rx.py --args 'rtl=22' -N 'LNA:47' -S 2400000 -f 859.6125e6 -o 25000 -q 24`

Radio 2:
` ./rx.py --args 'rtl=33' -N 'LNA:47' -S 2400000 -f 859.6125e6 -o 25000 -q 27`

Radio 3:
` ./rx.py --args 'rtl=44' -N 'LNA:47' -S 2400000 -f 859.6125e6 -o 25000 -q 1`

gofaster, 2020-02-04:
If you have it centered on the plots with a decent s/n then I am running out of ideas.
The only other thing i noticed from your log is that your rtl-sdr with the FC0012 tuner reports supporting a max gain of 19.2, and that's what it's set to - despite the requested 47 in your config.
```
[2020-02-04 13:10:02.385260] (info)   Source Device: rtl=33,buflen=65536
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
Using device #2 Realtek RTL2838UHIDIR SN: 33
Using 15 buffers of size 65536.
Found Fitipower FC0012 tuner
[2020-02-04 13:10:02.839868] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2020-02-04 13:10:02.839908] (info)   Setting sample rate to: 2048000
[2020-02-04 13:10:02.883387] (info)   Actual sample rate: 2048000
[2020-02-04 13:10:02.883460] (info)   Tunning to 8.542452e+08
[2020-02-04 13:10:02.922585] (info)   Gain Stage: LNA supported values: -9.9 -4 7.1 17.9 19.2
[2020-02-04 13:10:02.922656] (info)   Max Freqency: 8.552927e+08
[2020-02-04 13:10:02.922673] (info)   Min Freqency: 8.532447e+08
[2020-02-04 13:10:02.940576] (info)   Gain set to: 19.2
[2020-02-04 13:10:02.940672] (info)   Setting antenna to [RX]
```

My log:
```
[2020-02-02 20:11:56.655170] (info)   Source Device: rtl=00000003
gr-osmosdr 0.1.5 (0.1.5) gnuradio 3.7.11
built-in source types: file fcd rtl rtl_tcp uhd hackrf rfspace airspy soapy redpitaya
Using device #0 Realtek RTL2838UHIDIR SN: 00000003
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[2020-02-02 20:11:57.279264] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2020-02-02 20:11:57.279344] (info)   Setting sample rate to: 2560000
[R82XX] PLL not locked!
[2020-02-02 20:11:57.403288] (info)   Actual sample rate: 2560000
[2020-02-02 20:11:57.403389] (info)   Tunning to 8.524301e+08
[2020-02-02 20:11:57.457754] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6
[2020-02-02 20:11:57.457835] (info)   Max Freqency: 8.537112e+08
[2020-02-02 20:11:57.457880] (info)   Min Freqency: 8.511512e+08
[2020-02-02 20:11:57.484808] (info)   Gain set to: 12.5
[2020-02-02 20:11:57.484908] (info)   Setting antenna to [RX]
```

CharlesCage, 2020-02-05:
@gofaster all thoughts greatly appreciated and explored! I'm stumped, but I'm not giving up.

So some more grist for the mill:

I used some really cheap SDRs for this setup. Shouldn't be a problem as they were working fine on the pi, and I'm used to temp drift, etc. But to eliminate that, I swapped a good unit in for device 0. (NESDR, 0.5ppm, confirmed that it's dead on). 

Also checked no issues in OP25:
`./rx.py --args 'rtl=00000001' -N 'LNA:47' -S 2400000 -f 859.6125e6 -o 25000 -q -0
`
I think you may be on to something wtih gain. Is it possible that OP25 is setting the gain differently than Trunk Recorder? 

To try and see what's going on, I popped the first radio up on rtl_tcp, then pointed OP25 and Trunk Recorder at it, since we can see the rtl_tcp output and (maybe?) see what the driver is actually telling the radio to do gain-wise.

`rtl_tcp -a 0.0.0.0 -p 1337 -d0`
`
./rx.py --args 'rtl_tcp=127.0.0.1:1337' -N 'LNA:47' -S 2400000 -f 859.6125e6 -o 25000 -q -0`

Result (seen in rtl_tcp window):

```
client accepted! localhost 52534
Allocating 15 zero-copy buffers
set gain mode 1
set agc mode 0
set direct sampling 0
Disabled direct sampling mode
set offset tuning 0
set bias tee 0
set gain 480
set sample rate 2400000
set freq 859637500

```

Now swapped rtl_tcp in for first radio in Trunk Recorder:
```

        {
            "center": 858593700.0,
            "rate": 2048000.0,
            "ppm": 0,
            "error": 0,
            "gain": 47,
            "digitalRecorders": 2,
            "modulation": "qpsk",
            "driver": "osmosdr",
            "device": "rtl_tcp=127.0.0.1:1337"
        },

```

And the result (from rtl_tcp window):
```

client accepted! localhost 52536
Allocating 15 zero-copy buffers
set gain mode 1
set agc mode 0
set direct sampling 0
Disabled direct sampling mode
set offset tuning 0
set bias tee 0
set sample rate 2048000
set freq 858593700
set gain 480
ll+, now 1
ll+, now 2
ll+, now 3
ll+, now 4
ll+, now 5
ll+, now 6
ll+, now 7
ll+, now 8
ll+, now 9
ll+, now 10
ll+, now 11
ll+, now 12
ll+, now 13
ll+, now 14
ll+, now 15
ll+, now 16
ll+, now 17

```

Hm. 

(Thanks, btw, for sharing your thoughts.)

CharlesCage, 2020-02-05:
Update on working those differences:

I tried setting sampling rate to 2400000 and removing the other radios to prevent any USB/latency issues:
```

client accepted! localhost 52544
Allocating 15 zero-copy buffers
set gain mode 1
set agc mode 0
set direct sampling 0
Disabled direct sampling mode
set offset tuning 0
set bias tee 0
set sample rate 2400000
set freq 858593700
set gain 480

```

Still same issues.

Could it be the frequency offset applied by OP25 and not Trunk Recorder? So far I'm not able to see how to tell Trunk Recorder to use the same offset.

gofaster, 2020-02-05:
I think the OP25 offset tuning is to avoid the center DC spike.  In which case it doesn't apply to TR unless your control/voice channel happens to be at TR's configured"center:" freq.
Just to get a clear picture in my mind, what is the issue at this point?

CharlesCage, 2020-02-05:
Hey, I think I figured this out! For whatever reason it works if I place one of the radios DIRECTLY CENTERED on the control freq. Even moving it off a little causes the error. Not sure what's going on and I have more troubleshooting to do, but I think I've found at least the general area of the cause. Maybe something has slightly changed the bandwidth on the PC vs the pi?

CharlesCage, 2020-02-10:
Ok, closing this as I seem to have resolved it with a number of common fixes. Specifically, I dialed everything back in frequency-wise and replaced all my radios with very accurate ones. 

A hardy thank you to @gofaster for sticking it out with me.

closed 2020-02-10 by CharlesCage