TrunkRecorder/trunk-recorder #306: P25 system configs ignore the CSV priority setting

bctrainers, 2020-02-05:
From the gitter chat room.

On Project25 systems that are configured with a CSV file, the talkgroups configured with a record priority value of 99 or more continue to be recorded - said files are not discarded either. 

The code is located at https://github.com/robotastic/trunk-recorder/blob/5305e8f5320315094bcb646e571f79e5e4f9acb2/trunk-recorder/source.cc#L454 which appears to be getting referenced by https://github.com/robotastic/trunk-recorder/blob/ed127e214899d0d1760d269a9399a81e78640535/trunk-recorder/main.cc#L580 

From looking at the code above, only if the talkgroup mode from the CSV is parsed as 'A' (for analog) and not both A and D options are being discarded/ignored properly.

bctrainers, 2020-02-05:
Current config.json file...

```json
{
        "sources": [{
                        "center": "855000000",
                        "rate": "9500000",
                        "squelch": "0",
                        "error": "0",
                        "ppm": "17",
                        "gain": "0",
                        "ifGain": "8",
                        "bbGain": "40",
                        "digitalRecorders": 15,
                        "debugRecorders": 0,
                        "analogRecorders": 0,
                        "digitalLevels": 8,
                        "driver": "osmosdr",
                        "device": "hackrf=56b35f",
                        "modulation": "qpsk"
        }],
        "systems": [{
                        "control_channels": [853775000,853962500],
                        "type": "p25",
                        "talkgroupsFile": "/root/trunk-build/joco.csv",
                        "recordUnknown": false,
                        "shortName": "JoCo",
                        "callLog": true,
                        "hideEncrypted": false,
                        "hideUnknownTalkgroups": false
        }],
        "defaultMode": "digital",
        "captureDir": "/root/trunk-build/audio_files",
        "callTimeout": "3",
        "logFile": "1",
        "frequencyFormat": "mhz"
}
```

gofaster, 2020-02-05:
Per gitter follow-on; Not recording priority>99 is and has been working for me since v3.0 .
Perhaps temporarily add some some debug output around the fragments above to see what trunk-recorder thinks the TG's priority is?

bctrainers, 2020-02-05:
After digging around, folks on gitter chat concluded that the HEX value in the CSV has to be filled out properly for the priority to function.

closed 2020-02-05 by bctrainers

rpdale, 2020-02-05:
It doesn’t have to be “proper” (i.e. you don’t have to convert the actual talkgroup to hex) but does need something in there. From: BrettSent: Wednesday, February 5, 2020 1:36 PMTo: robotastic/trunk-recorderCc: SubscribedSubject: Re: [robotastic/trunk-recorder] P25 system configs ignore the CSV priority setting (#306) After digging around, folks on gitter chat concluded that the HEX value in the CSV has to be filled out properly for the priority to function.