TrunkRecorder/trunk-recorder #311: Issue with recording P25 Phase 2 traffic

CqDx914, 2020-02-08:
Hello, 

I have been using a Raspberry Pi 3 B+ with trunk-recorder installed and it’s doing a great job on decoding and recording P25 Phase 1 traffic, but when the network recently started carrying Phase 2 traffic, all it records were garble and unintelligible traffic. I see the logs reflect it recognizes TDMA traffic but it just would not record Phase 2 intelligibly.

Anyone has experienced this as well?

My raspberry pi is currently running on the following
Linux version 4.14.79-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1159 SMP Sun Nov 4 17:50:20 GMT 2018

robotastic, 2020-02-09:
It is definitely possible that the math gets mixed up on ARM chips. The only real difference between OP25 and Trunk Recorder for Phase 2, are these files: https://github.com/robotastic/trunk-recorder/tree/master/lib/lfsr
I will try to look through and see if I spot any obvious things where it is assuming

GTR8000, 2020-02-11:
I also think that Phase II decoding is hosed up on the ARM platform.

We tried running on a Pi 3 B+ and could not get any intelligible audio from a TDMA system. Switched over to an ACEPC AK2 using the exact same RTL dongle and config.json, and decoding was excellent.

CqDx914, 2020-02-11:
GTR8000 what distro did you use on the ACEPC AK2 may I ask? 
I had some luck setting up Ubuntu 18.04 using parallel desktop on my mac and got trunk-recorder working, but currently having problem with it using airspy dongle, so more experimentation on my side is needed..

GTR8000, 2020-02-11:
Ubuntu 18.04.3 LTS, Trunk Recorder v3.1.2, with fdk-aac and sox also installed to support the OpenMHz change from ffmpeg encoding.

edinburger, 2020-02-21:
Just started a new setup with RasPi4 (4GB) and am experiencing the same issues with TDMA recordings.  Garbled audio sounds like it is encrypted.  No issues with Phase-I TGs though.

CqDx914, 2020-03-16:
I can confirm that when I use my Macbook Pro (much more horse power than Rp3 B+ obviously), I have no issue decoding Phase 2 traffic.

mikewren, 2020-06-17:
Not sure if it's related, but I experienced the garbled audio issues on a P25 phase II system on talkgroups encoded as TDMA using a TR version from March.  Recompiled on 6/13 and it's not working perfectly - thanks!

Cellphone70, 2020-06-17:
I was using the docker version experiencing the issue.  I abandoned that, and compiled from the ditros about 2 weeks ago, and Phase 2 has been working for me since.

jonk6, 2020-06-17:
> Recompiled on 6/13 and it's not working perfectly - thanks!

not? Or now?

mikewren, 2020-06-17:
> > Recompiled on 6/13 and it's not working perfectly - thanks!
> 
> @mikewren - not? Or now?

Sorry, typo, it's NOW working perfect - https://openmhz.com/system/monroecony

closed 2020-08-04 by robotastic