TrunkRecorder/trunk-recorder #315: "pos" timestamps in JSON srcList don't align with WAV audio

aaknitt, 2020-02-10:
Working on a [script to rebuild incident audio in an Audacity project](https://github.com/aaknitt/incident_builder).  When splitting WAV files using the timing information in the associated JSON files I'm running into many instances where the "pos" timestamps cut the WAV file in the middle of a transmission.  This seems to occur mostly towards the end of longer WAV files that have multiple transmissions in them.  Could be related to #284

### Example:
Example showing instances where "pos" timestamps end up in the middle of transmissions.  This example also happens to show missing audio at the end (8 entries in srcList but only 7 actual transmissions captured in WAV file), which is issue #284.
[image]

### Example when WAV is split:
The example below shows a instance where the script split a wav file in the middle of a radio transmission (highlighted in red), where it should have been split near the yellow line instead.
[image]

robotastic, 2020-02-10:
Ok - I did some digging. I didn't find anything that was obviously wrong. It could be that the there is significant delay between when a Radio is granted a channel and when the audio actually makes it over the air and is decoded. Right now position is generated by the messages from the control channel. When the Src of a transmission for a TG doesn't match the one currently assigned for a call, TR will look at the current length of the .wav file and use that as the `pos` in the JSON. Still, it doesn't seem like there should be a 2 second gap. I don't think processing and buffers are that long.

I did used to do it differently. I instead looked for a change in the Src ID when the audio is decoded. The wav recorder would get a tag associated with the audio with the Src identified. I dug up some of that code. It just spits it out to the screen. Can you see if this is more accurate: https://github.com/robotastic/trunk-recorder/tree/experiment

aaknitt, 2020-02-10:
> I did used to do it differently. I instead looked for a change in the Src ID when the audio is decoded. The wav recorder would get a tag associated with the audio with the Src identified. I dug up [...]

Ok, I'll give that a try.  May be some days or weeks before I'm able to get to it.

aaknitt, 2020-02-11:
It's looking like upgrading to 3.1.2 may have resolved both this and #284.  I'll keep checking every few days but if all continues to work well I'll close out both issues.

aaknitt, 2020-02-12:
Fixed by upgrading to 3.1.2

closed 2020-02-12 by aaknitt