TrunkRecorder/trunk-recorder #316: Tune Offset: Freq exceeds limit error

CharlesCage, 2020-02-10:
Hi, all!

Getting this weird error pretty regularly, and trying to figure out what I have wrong in my config to make it happen.

```
[2020-02-09 19:56:59.836424] (info)   [frisco]  TG:        420  Freq: 8.566125e+08      TG not in Talkgroup File 
[2020-02-09 19:56:59.836481] (info)     - Starting P25 Recorder Num [6] TG:        420  Freq: 8.566125e+08      TDMA: false     Slot: 0
[2020-02-09 19:56:59.836586] (info)   [frisco]  TG:        420  Freq: 8.566125e+08      Starting Recorder on Src: rtl=44,buflen=65536
[2020-02-09 19:57:09.140198] (info)     - Starting P25 Recorder Num [7] TG:          2  Freq: 8.576125e+08      TDMA: false     Slot: 0
[2020-02-09 19:57:09.140345] (info)   [frisco]  TG:          2  Freq: 8.576125e+08      Starting Recorder on Src: rtl=44,buflen=65536
[2020-02-09 19:57:13.031959] (info)     - Starting P25 Recorder Num [4] TG:        103  Freq: 8.542125e+08      TDMA: false     Slot: 0
[2020-02-09 19:57:13.032003] (info)   Tune Offset: Freq exceeds limit: 1.0125e+06 compared to: 960000
terminate called after throwing an instance of 'std::out_of_range'
  what():  firdes check failed: 0 < fb <= sampling_freq / 2
/usr/bin/start-trunk-recorder.sh: line 2: 13931 Aborted                 ./recorder

```
My config, for reference:

[{   
    "sources": [

        {   
            "center": 859612500.0,
            "rate": 2048000.0,
            "ppm": 0,
            "gain": 47,
            "digitalRecorders": 2,
            "modulation": "qpsk",
            "driver": "osmosdr",
            "device": "rtl=11,buflen=65536"
        },
        {   
            "center": 851062500.0,
            "rate": 2048000.0,
            "ppm": 0,
            "gain": 47,
            "digitalRecorders": 2,
            "modulation": "qpsk",
            "driver": "osmosdr",
            "device": "rtl=22,buflen=65536"
        },
        {   
            "center": 853200000.0,
            "rate": 2048000.0,
            "ppm": 0,
            "gain": 47,
            "digitalRecorders": 2,
            "modulation": "qpsk",
            "driver": "osmosdr",
            "device": "rtl=33,buflen=65536"
        },
        {   
            "center": 857475000.0,
            "rate": 2048000.0,
            "ppm": 0,
            "gain": 47,
            "digitalRecorders": 2,
            "modulation": "qpsk",
            "driver": "osmosdr",
            "device": "rtl=44,buflen=65536"
        }
        ],
    "systems": [{
        "control_channels": [859612500],
        "type": "p25",
        "talkgroupsFile": "frisco-talkgroups.csv",
        "shortName": "frisco",
        "uploadScript": "encode-local-sys-0.sh"
    }],
    "captureDir": "/home/chuckcage/p25files/"

}
](url)

robotastic, 2020-02-10:
Yep - I think I created this problem when I changed how channels are created. Basically the problem is that TR does a very simple calculation to see if a Source can cover a channels freq. It should take into account the size of the channel. A quick fix would be to set the sample rate on the RTLSDR to 2400000 to make sure you can have enough room to cover everything. I will work on a better fix.

robotastic, 2020-09-27:
Looks like I fixed this with this commit. Re-open if it didn't fix it.
https://github.com/robotastic/trunk-recorder/commit/e987fceb3c081fa834dbfcc3b44a2c3c48709560#diff-43e932befd82e6fc6d5ed13cab1f35df

closed 2020-09-27 by robotastic