TrunkRecorder/trunk-recorder #318: P25Conventional stops recording

gofaster, 2020-02-15:
Monitoring a conventional P25 system with delayCreateOutput enabled.
A P25 recorder listens for a call, then starts a recorder and stops it afterwards.  This works fine for a while.
This system transmits it's call sign as Morse code as plain FM every 30 mins.  
All P25 calls after this station ID transmission will not trigger a P25 recorder to start.  It stays in the "Listening P25 recorder state" and regularly increments the elapsed counter at the usual 200 sec interval.
This situation is immediately corrected by restarting the trunk-recorder instance - and works normally until after the next station ID transmission.

gofaster, 2020-02-16:
Tested with delayCreateOutput set to false.  Same issue, all P25 calls after the station ID transmission are not recorded.
[image]

gofaster, 2020-02-16:
With P25 verbosity set at 10 - before a station ID transmission, P25 calls emit lines like the following.  All calls are recorded and voice decodes sound great.
There are no lines emitted when the station ID is transmitted.
All subsequent P25 calls do not emit any lines.
Is it possible to add debug logging statements to the blocks in this part of the flowgraph to see where it dies?

```
...
1581888801.760599 NAC 0x114 TDU3:
1581888801.760609 NAC 0x114 TDU3:
1581888801.760669 NAC 0x114 TDU3:
1581888801.818591 NAC 0x114 TDU3:
1581888801.818635 NAC 0x114 TDU3:
1581888801.818645 NAC 0x114 TDU3:
1581888801.818654 NAC 0x114 TDU3:
1581888801.818664 NAC 0x114 TDU3:
1581888801.915752 NAC 0x114 TDU3:
1581888801.915795 NAC 0x114 TDU3:
1581888801.915806 NAC 0x114 TDU3:
1581888801.915815 NAC 0x114 TDU3:
1581888801.915824 NAC 0x114 TDU3:
1581888801.973320 NAC 0x114 TDU3:
1581888801.973367 NAC 0x114 TDU3:
1581888801.973377 NAC 0x114 TDU3:
1581888801.973386 NAC 0x114 TDU3:
1581888801.973414 NAC 0x114 TDU3:
1581888802.072238 NAC 0x114 TDU3:
1581888802.072278 NAC 0x114 TDU3:
1581888802.072289 NAC 0x114 TDU3:
1581888802.072299 NAC 0x114 TDU3:
1581888802.656260 NAC 0x114 TDU3:
1581888802.755658 NAC 0x114 HDU:  ESS: tgid=1, mfid=0, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, gly_errs=0, rs_errs=0
1581888802.966156 NAC 0x114 LDU1: LCW: ec=0, pb=0, sf=0, lco=0, srcaddr=102, grpaddr=1 : 00 00 04 00 00 01 00 00 66
1581888802.966238 IMBE 18 3e 92 32 1e 40 00 24 0b 00 52 errs 0
1581888802.966377 IMBE 18 50 ce ae c9 a2 00 96 9b 7a 63 errs 0
1581888802.966457 IMBE 18 73 88 08 35 95 04 61 09 61 f2 errs 0
1581888802.966552 IMBE 94 a3 cb 69 c3 51 00 01 3f be 55 errs 0
1581888802.966679 IMBE 18 1f d0 f0 4f 82 06 48 29 ab 9a errs 0
1581888802.966778 IMBE 18 42 6b db b9 fc 06 ad 4e a0 e3 errs 0
1581888802.966873 IMBE 18 38 9e 3e 1b 40 07 f9 84 1d 62 errs 0
1581888802.966967 IMBE 18 47 e0 c0 a4 1f 04 20 7a 1f cb errs 0
1581888802.967056 IMBE 18 0b f8 f8 dc d0 04 76 88 b6 aa errs 0
1581888803.121508 NAC 0x114 LDU2: ESS: algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00rs_errs=0
1581888803.121567 IMBE 18 0c 77 d7 66 67 05 2b f3 61 8b errs 0
1581888803.121678 IMBE 18 45 65 e5 8c ba 04 65 28 bd 4a errs 0
1581888803.121774 IMBE 18 3a 1b 1b 67 73 04 70 78 7a 2b errs 0
1581888803.121867 IMBE 18 77 80 20 10 01 05 a0 96 24 aa errs 0
1581888803.121957 IMBE 18 28 be 7e 57 d9 06 69 9b ce e3 errs 0
1581888803.122052 IMBE 18 0e 73 f3 0a 18 05 83 a5 b3 b2 errs 0
1581888803.122145 IMBE 18 39 9c 1c 7f fc 05 1e 27 fe 0b errs 0
1581888803.122236 IMBE 18 74 07 27 0c b2 04 60 68 31 52 errs 0
1581888803.122325 IMBE 18 3b 98 18 37 ed 05 10 53 a1 2b errs 0
1581888803.278144 NAC 0x114 LDU1: LCW: ec=0, pb=0, sf=0, lco=0, srcaddr=102, grpaddr=1 : 00 00 04 00 00 01 00 00 66
1581888803.278212 IMBE 18 57 41 a1 c8 a2 04 d8 bd 94 4a errs 0
1581888803.278319 IMBE 18 28 be 7e 57 cd 05 75 91 40 83 errs 0
1581888803.278406 IMBE 18 3f 11 31 4e b2 04 32 8b 91 aa errs 0
1581888803.278504 IMBE 18 09 fc fc 9d 90 07 75 21 cc 03 errs 0
1581888803.278600 IMBE 18 19 dc 9c b5 c5 04 6a b9 6a ca errs 0
1581888803.278693 IMBE 18 41 ec ec 95 b9 06 54 98 db eb errs 0
1581888803.278790 IMBE 18 1a 5b 93 e7 67 07 77 90 bb 92 errs 0
1581888803.278886 IMBE a1 07 e1 7d d1 7d 00 05 c2 f9 59 errs 0
1581888803.279021 IMBE 90 5f 6d 29 c0 18 00 0e b6 30 28 errs 0
1581888803.491602 NAC 0x114 LDU2: ESS: algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00rs_errs=0
1581888803.491656 IMBE 24 e5 ea 06 eb 5d 8a be 3a 34 ef errs 0
1581888803.491799 IMBE 2a e2 dd 8a 06 4f e9 57 96 15 b4 errs 0
1581888803.492021 IMBE 2f 4a 8d 4e 3a 39 fb c4 ca c3 e3 errs 0
1581888803.492274 IMBE 2f 2f 80 4d 7d cb fd ea bc 86 de errs 0
1581888803.492552 IMBE 33 4e 81 2d 3a a6 ff cc 36 82 b1 errs 0
1581888803.492831 IMBE 33 4a 99 fb 1e 2f fe 82 e1 00 ea errs 0
1581888803.493097 IMBE 31 ce a1 7f 38 d4 f3 85 ed 65 9d errs 0
1581888803.493350 IMBE 31 bc 26 86 cb 96 90 6e 19 90 62 errs 0
[... 12 lines cut ...]
1581888803.859494 IMBE 18 0e f2 d2 f6 4d 04 42 99 98 c3 errs 0
1581888803.859602 IMBE 18 72 8a 2a 11 51 04 c5 1d 45 8a errs 0
1581888803.859697 IMBE 18 3a 1b 3b 8f 62 05 30 26 a2 7b errs 0
1581888803.859819 IMBE 18 55 c4 a4 ca 82 05 88 27 cd 32 errs 0
1581888803.859912 IMBE 18 60 2f 4f 7d ac 04 4d 48 ac 03 errs 0
1581888803.860062 IMBE 18 0b f8 f8 4d 52 04 7e 29 42 ca errs 0
1581888803.860165 IMBE 18 66 a2 62 5e 68 06 6a ac c8 43 errs 0
1581888803.860263 IMBE 18 53 49 a8 d9 b8 06 a3 46 12 92 errs 0
1581888803.860362 IMBE 18 0a 7b bb cf 7a 06 07 eb 74 5b errs 0
1581888804.016710 NAC 0x114 LDU1: LCW: ec=0, pb=0, sf=0, lco=0, srcaddr=102, grpaddr=1 : 00 00 04 00 00 01 00 00 66
1581888804.016780 IMBE 18 0c f6 e6 56 f9 07 53 95 c7 ca errs 0
1581888804.016894 IMBE 18 2c 37 57 22 ff 06 9f 9e c1 a3 errs 0
1581888804.016996 IMBE 18 3b 98 38 1f 80 07 20 83 c6 32 errs 0
1581888804.017095 IMBE 80 24 30 66 f5 af 00 0c 76 86 c1 errs 0
1581888804.017214 IMBE 18 1f 51 b1 c2 33 06 e9 3d 8c 42 errs 0
1581888804.017313 IMBE 8c 45 68 f7 50 40 00 0e 29 de 33 errs 0
1581888804.017435 IMBE 18 53 c8 28 dd 30 07 41 65 3c 92 errs 0
1581888804.017534 IMBE 18 57 c0 80 b4 01 06 09 db de eb errs 0
1581888804.017634 IMBE d8 00 00 00 00 00 00 00 00 00 00 errs 0
1581888804.017738 NAC 0x114 TDU3:
[2020-02-16 16:33:30.006519] (info)   [HDD]-[1]       TG:          1 (                  HDD)        Freq: 4.603625e+08      Ending Recorded Call - Last Update: 84s Call Elapsed: 84
[2020-02-16 16:33:30.006732] (info)     - Stopping P25 Recorder Num [5] TG:          1 (                  HDD)        Freq: 4.603625e+08      TDMA: false     Slot: 0
[2020-02-16 16:33:30.006782] (info)   Running upload script: 
...
```

gofaster, 2020-02-17:
Turned up log verbosity and discovered that after the ID transmission (red arrow), the gr pwr_squelch block will not open for subsequent P25 transmissions (up to 12 hrs beyond).  I'm trying different values for "alpha" with varying results.

[image]
```

[2020-02-17 11:01:15.809642] (info)     - Started P25 Recorder Num [0]  TG:          1 (                  HDD)        Freq: 4.603625e+08     TDMA: false      Slot: 0
[2020-02-17 11:01:15.812457] (info)   [HDD]   TG:          1 (                  HDD)        Freq: 4.603625e+08

[2020-02-17 11:01:35.013070] (trace)   Recorder: 0.18 Idle: false Count: 0
[2020-02-17 11:01:36.010019] (trace)   Recorder: 1.08 Idle: false Count: 0
[2020-02-17 11:01:37.007026] (trace)   Recorder: 2.268 Idle: false Count: 0
[2020-02-17 11:01:38.013951] (trace)   Recorder: 2.34 Idle: true Count: 0
[2020-02-17 11:01:39.010735] (trace)   Recorder: 3.06 Idle: false Count: 1
[2020-02-17 11:01:40.007685] (trace)   Recorder: 3.42 Idle: true Count: 0
[2020-02-17 11:01:41.004421] (trace)   Recorder: 3.6 Idle: false Count: 1
[2020-02-17 11:01:42.011314] (trace)   Recorder: 4.5 Idle: false Count: 0
[2020-02-17 11:01:43.008119] (trace)   Recorder: 5.688 Idle: false Count: 0
[2020-02-17 11:01:44.005528] (trace)   Recorder: 6.588 Idle: false Count: 0
[2020-02-17 11:01:45.013055] (trace)   Recorder: 7.56 Idle: false Count: 0
[2020-02-17 11:01:46.010139] (trace)   Recorder: 8.64 Idle: false Count: 0
[2020-02-17 11:01:47.006905] (trace)   Recorder: 9.54 Idle: false Count: 0
[2020-02-17 11:01:48.013931] (trace)   Recorder: 10.548 Idle: false Count: 0
[2020-02-17 11:01:49.011142] (trace)   Recorder: 11.628 Idle: false Count: 0
[2020-02-17 11:01:50.007953] (trace)   Recorder: 12.6 Idle: false Count: 0
[2020-02-17 11:01:51.005028] (trace)   Recorder: 13.608 Idle: false Count: 0
[2020-02-17 11:01:52.011795] (trace)   Recorder: 14.688 Idle: false Count: 0
[2020-02-17 11:01:53.008691] (trace)   Recorder: 15.66 Idle: false Count: 0
[2020-02-17 11:01:54.005487] (trace)   Recorder: 16.2 Idle: false Count: 0
[2020-02-17 11:01:55.012484] (trace)   Recorder: 16.2 Idle: false Count: 0
[2020-02-17 11:01:56.009235] (trace)   Recorder: 16.2 Idle: true Count: 0
[2020-02-17 11:01:57.006094] (trace)   Recorder: 16.2 Idle: true Count: 1
[2020-02-17 11:01:58.012825] (trace)   Recorder: 16.2 Idle: true Count: 2
[2020-02-17 11:01:59.009804] (trace)   Recorder: 16.2 Idle: true Count: 3
[2020-02-17 11:02:00.006648] (trace)   Recorder: 16.2 Idle: true Count: 4
[2020-02-17 11:02:01.013616] (trace)   Recorder: 16.2 Idle: true Count: 5
[2020-02-17 11:02:01.013761] (info)   [HDD]-[0]       TG:          1 (                  HDD)        Freq: 4.603625e+08      Ending Recorded Call - Last Update: 46s Call Elapsed: 46
[2020-02-17 11:02:01.013958] (info)     - Stopping P25 Recorder Num [0] TG:          1 (                  HDD)        Freq: 4.603625e+08     TDMA: false      Slot: 0
[2020-02-17 11:02:01.014034] (info)   Running upload script: ./encode-local-sys-5.sh /home/radio/recordings/HDD/2020/2/17/1-1581955275_460362500.wav &
[2020-02-17 11:02:01.016784] (info)     - Started P25 Recorder Num [0]  TG:          1 (                  HDD)        Freq: 4.603625e+08     TDMA: false      Slot: 0
Encoding: /home/radio/recordings/HDD/2020/2/17/1-1581955275_460362500.wav
...
[2020-02-17 11:02:02.004261] (trace)   Recorder: 0.36 Idle: false Count: 0
[2020-02-17 11:02:03.011339] (trace)   Recorder: 1.26 Idle: false Count: 0
[2020-02-17 11:02:04.008251] (trace)   Recorder: 2.268 Idle: false Count: 0
[2020-02-17 11:02:05.005044] (trace)   Recorder: 2.52 Idle: true Count: 0
[2020-02-17 11:02:06.012055] (trace)   Recorder: 2.52 Idle: true Count: 1
[2020-02-17 11:02:07.008835] (trace)   Recorder: 2.52 Idle: true Count: 2
[2020-02-17 11:02:08.005660] (trace)   Recorder: 2.988 Idle: false Count: 3
[2020-02-17 11:02:09.012538] (trace)   Recorder: 3.888 Idle: false Count: 0
[2020-02-17 11:02:10.009254] (trace)   Recorder: 4.14 Idle: false Count: 0
[2020-02-17 11:02:11.005956] (trace)   Recorder: 4.14 Idle: false Count: 0
[2020-02-17 11:02:12.012853] (trace)   Recorder: 4.14 Idle: true Count: 0
[2020-02-17 11:02:13.009697] (trace)   Recorder: 4.14 Idle: true Count: 1
[2020-02-17 11:02:14.006583] (trace)   Recorder: 4.14 Idle: true Count: 2
[2020-02-17 11:02:15.013575] (trace)   Recorder: 4.14 Idle: true Count: 3
[2020-02-17 11:02:16.010526] (trace)   Recorder: 4.14 Idle: true Count: 4
[2020-02-17 11:02:17.007641] (trace)   Recorder: 4.14 Idle: true Count: 5
[2020-02-17 11:02:17.007818] (info)   [HDD]-[0]       TG:          1 (                  HDD)        Freq: 4.603625e+08      Ending Recorded Call - Last Update: 16s Call Elapsed: 16
[2020-02-17 11:02:17.007959] (info)     - Stopping P25 Recorder Num [0] TG:          1 (                  HDD)        Freq: 4.603625e+08     TDMA: false      Slot: 0
[2020-02-17 11:02:17.008018] (info)   Running upload script: ./encode-local-sys-5.sh /home/radio/recordings/HDD/2020/2/17/1-1581955321_460362500.wav &
[2020-02-17 11:02:17.010526] (info)     - Started P25 Recorder Num [0]  TG:          1 (                  HDD)        Freq: 4.603625e+08     TDMA: false      Slot: 0
Encoding: /home/radio/recordings/HDD/2020/2/17/1-1581955321_460362500.wav
...
[2020-02-17 11:04:36.010112] (info)   Currently Active Calls: 1
[2020-02-17 11:04:36.010269] (info)   TG:          1 (                  HDD) Freq: 4.603625e+08 Elapsed: 139 State: recording
[2020-02-17 11:04:36.010316] (info)     [ 0 ] State: active
[... cut ...]
[2020-02-17 11:11:18.012359] (info)   [ rtl=00000007 ]
[2020-02-17 11:11:18.012386] (info)   [ D0 ] State: active
[2020-02-17 11:14:39.008113] (info)   Currently Active Calls: 1
[2020-02-17 11:14:39.008250] (info)   TG:          1 (                  HDD) Freq: 4.603625e+08 Elapsed: 742 State: recording
[2020-02-17 11:14:39.008312] (info)     [ 0 ] State: active
[2020-02-17 11:14:39.008335] (info)   Recorders:
[2020-02-17 11:14:39.008356] (info)   [ rtl=00000007 ]
[2020-02-17 11:14:39.008381] (info)   [ D0 ] State: active

```

robotastic, 2020-02-18:
Well that is just weird. There is a bit of documentation here: https://wiki.gnuradio.org/index.php/Power_Squelch
It looks like it is doing everything correctly. Maybe try adjusting the squelch threshold?
Is this a QPSK system?  If so, try replace line 64 of p25conventional_recorder.cc with this:
````
  if (!qpsk_mod) {
    reset();
  } else {
    costas_clock->reset();
  }
````
Maybe that strong signal is messing up the tuning?
You could also try uncommenting the stuff in the reset() function in p25_recorder.cc

gofaster, 2020-02-18:
Digging more into squelch.  Found that increasing alpha would appear to keep things going through 1 or 3 cycles of the ID transmission, then quit.  Tried simple_squelch with slightly improved longevity.  
I ended up splicing a udp_sink to the output of squelch and watching that in GRC to directly verify  when/if squelch is open.  When the issue happens, I see data from the udp_sink in GRC (squelch has opened) but no recordings (or idle timer messages).  Will keep poking further down the flow to hopefully find the cause.

I don't know how to properly resolve the following...
```

[ 63%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/recorders/p25conventional_recorder.cc.o
In file included from /home/radio/src/trunk-recorder/trunk-recorder/trunk-recorder/recorders/p25conventional_recorder.cc:1:0:
/home/radio/src/trunk-recorder/trunk-recorder/trunk-recorder/recorders/p25_recorder.h: In member function ‘virtual void p25conventional_recorder::start(Call*)’:
/home/radio/src/trunk-recorder/trunk-recorder/trunk-recorder/recorders/p25_recorder.h:196:46: error: ‘gr::op25_repeater::gardner_costas_cc::sptr p25_recorder::costas_clock’ is private
   gr::op25_repeater::gardner_costas_cc::sptr costas_clock;
                                              ^
/home/radio/src/trunk-recorder/trunk-recorder/trunk-recorder/recorders/p25conventional_recorder.cc:67:5: error: within this context
     costas_clock->reset();
     ^
CMakeFiles/recorder.dir/build.make:542: recipe for target 'CMakeFiles/recorder.dir/trunk-recorder/recorders/p25conventional_recorder.cc.o' failed
```

gofaster, 2020-02-19:
Adding   " costas_clock->reset()" didn't make any difference.

Looking at the outputs of slicer and p25_frame_assembler blocks: 
- When working, there is data from both during a P25 call.
- When not working after an ID transmission, there is data from the slicer (0x00-0x03) but no output from the p25_assembler_block during a P25 call. The slicer does output some data during the ID transmission. Perhaps they are throwing the p25_frame_assembler block into a spin..?

I'll try poking into the p25_recorder.cc reset() function next.

gofaster, 2020-02-20:
Set up some IQ recordings to make test cycles faster rather than waiting for transmissions in real time.
I can reliably cause it to stop recognizing/recording P25 calls. 
Uncommenting combinations of the functions in reset() did not change the behaviour.

One clue I discovered is if the ID transmission happens DURING the idle timeout from a previous P25 call, the subsequent P25 call is recognized/recorded.    If it is transmitted AFTER the timeout, subsequent calls are ignored - even though the slicer block still sends non 0x00 symbols to the p25_frame_assembler block.
```
...
1582207121.753934 NAC 0x114 TDU3:
1582207121.753944 NAC 0x114 TDU3:
1582207121.753956 NAC 0x114 TDU3:
1582207121.753969 NAC 0x114 TDU3:
1582207121.830077 NAC 0x114 TDU3:
1582207121.830121 NAC 0x114 TDU3:
[2020-02-20 08:58:42.007315] (trace)   Recorder: 2.16 Idle: true Count: 0
[2020-02-20 08:58:43.013987] (trace)   Recorder: 2.16 Idle: true Count: 1
[2020-02-20 08:58:44.011834] (trace)   Recorder: 2.16 Idle: true Count: 2
[2020-02-20 08:58:45.008807] (trace)   Recorder: 2.16 Idle: true Count: 3
[2020-02-20 08:58:46.005679] (trace)   Recorder: 2.16 Idle: true Count: 4
>>> Start of ID transmission <<<<
[2020-02-20 08:58:47.012502] (trace)   Recorder: 2.16 Idle: false Count: 5
[2020-02-20 08:58:48.009658] (trace)   Recorder: 2.16 Idle: false Count: 0
[2020-02-20 08:58:49.006366] (trace)   Recorder: 2.16 Idle: false Count: 0
[2020-02-20 08:58:50.013163] (trace)   Recorder: 2.16 Idle: false Count: 0
[2020-02-20 08:58:51.009680] (trace)   Recorder: 2.16 Idle: false Count: 0
>>> End of ID transmission <<<<
[2020-02-20 08:58:52.006391] (trace)   Recorder: 2.16 Idle: true Count: 0
[2020-02-20 08:58:53.013211] (trace)   Recorder: 2.16 Idle: true Count: 1
[2020-02-20 08:58:54.009969] (trace)   Recorder: 2.16 Idle: true Count: 2
[2020-02-20 08:58:55.006822] (trace)   Recorder: 2.16 Idle: true Count: 3
[2020-02-20 08:58:56.013724] (trace)   Recorder: 2.16 Idle: true Count: 4
[2020-02-20 08:58:57.010374] (trace)   Recorder: 2.16 Idle: true Count: 5
[2020-02-20 08:58:57.010499] (info)   [TESTSYS]-[0]     TG:          1 (                   test)        Freq: 4.603625e+08      Ending Recorded Call - Last Update: 26s        Call Elapsed: 26
[2020-02-20 08:58:57.010634] (info)     - Stopping P25 Recorder Num [0] TG:          1 (                   test)        Freq: 4.603625e+08    TDMA: false      Slot: 0
[2020-02-20 08:58:57.010695] (info)   Running upload script: ./testscript.sh /tmp/TESTSYS/2020/2/20/1-1582207111_460362500.wav &
[2020-02-20 08:58:57.012956] (info)     - Started P25 Recorder Num [0]  TG:          1 (                   test)        Freq: 4.603625e+08    TDMA: false      Slot: 0
>>> Recording the subsequent call <<<<
1582207143.335242 NAC 0x114 TDU3:
1582207143.407275 NAC 0x114 HDU:  ESS: tgid=1, mfid=0, algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00, gly_errs=17, rs_errs=0
1582207143.559239 NAC 0x114 LDU1: LCW: ec=0, pb=0, sf=0, lco=0, srcaddr=1014027, grpaddr=1 : 00 00 04 00 00 01 0f 79 0b
1582207143.559318 IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 44 errs 0
1582207143.559483 IMBE 04 0c fd 7b fb 7d f2 7b 3d 9e 44 errs 1
1582207143.559674 IMBE 18 3e 92 32 1e 40 00 24 0b 00 52 errs 0
1582207143.559823 IMBE a8 64 80 55 6c 0f 00 03 b5 80 b9 errs 1
1582207143.559997 IMBE 18 63 29 79 18 12 05 ea 81 b2 c2 errs 0
1582207143.560108 IMBE 18 3c 17 37 12 f9 02 94 8f 92 13 errs 0
1582207143.560212 IMBE 18 60 ae 7e 10 d5 02 d4 9d 50 92 errs 0
1582207143.560320 IMBE 18 72 0b 0a eb a6 03 27 e6 ab 33 errs 0
1582207143.560417 IMBE 18 61 ac 2c 4c ce 03 24 07 69 42 errs 0
1582207143.783247 NAC 0x114 LDU2: ESS: algid=80, keyid=0, mi=00 00 00 00 00 00 00 00 00rs_errs=0
1582207143.783292 IMBE 18 51 cc 8c b2 b5 00 7e 5d 17 db errs 0
1582207143.783393 IMBE 18 63 29 69 0a 32 03 90 22 83 f2 errs 1
1582207143.783483 IMBE 18 60 2f 67 5b 48 01 9c 86 46 2b errs 0
1582207143.783573 IMBE 18 63 a8 28 49 1a 01 26 56 80 5a errs 0
1582207143.783661 IMBE 18 0b f8 d8 fa 4c 02 a4 3e 38 6b errs 0
1582207143.783753 IMBE 18 60 ae ee 0b 0e 02 a6 ea 3a 52 errs 0
1582207143.783841 IMBE 18 47 e0 e0 c6 4b 00 65 b1 d6 4b errs 0
1582207143.783965 IMBE 18 62 2b 6b 8b ca 01 ae a7 d7 02 errs 0
1582207143.784051 IMBE 18 50 4f 9f a1 b7 00 a3 7e 61 f3 errs 0
...
```

gofaster, 2020-02-26:
Working upstream, I think I found the culprit.  The ID transmission causes the costas clock recovery block to lock into a rotated state or something.  The reported phase error value never comes back to zero.  Resulting symbols to the p25_frame_assembler are nonsensical and it never find frame sync.
If I reset the costas block after an ID transmission,  the subsequent P25 frames are decoded successfully.
Likely trunked P25 recorder instantiations can also be knocked out if it receives this type of transmission while it's listening(?)

robotastic, 2020-02-27:
Nice find! That is some good sleuthing. Could you insert a costas_clock->reset() at the end of the stop() function? That way it would get called after a recording is over. There could still be an issues if the tone happens while a recording is not happening. For regular P25 recorders the costas_clock gets reset with every retune, so that maybe why it hasn't cropped up before.

gofaster, 2020-02-27:
Looks like that's it, the test instance has been recording for the past 36 hrs.  Yes that recorder will still go deaf if the ID is sent while it's just listening for a call. 
I had stuck that reset within the print_status function (every 200 sec)  in main.cc but but that's a bit heavy handed.  Getting some understanding of C++ and T-R along the way  but I'd like to periodically reset the clock only for an idle P25Conventional recorder.  I'd appreciate any pointers (no pun intended :) )

blue1sparky, 2020-04-24:
Did this commit fix the problem? I'm running into the same issue with my local PD on P25 Conventional.

gofaster, 2020-04-24:
The workaround has been working for me on a relatively low call volume agency.  There is a chance that a call can be missed if it comes in before the periodic reset.

JoeGilkey, 2020-04-24:
I think the op25 state variables need to be cleared after a period of silence. That prevents any hiccups due to the repeated announcement.

robotastic, 2020-05-01:
@JoeGilkey Which state values are these? I think I setup a reset function that can be called that resets the vocoder in OP25.

mjmikolay, 2020-12-08:
Just adding that I too am having this issue and would be interested in workarounds. Not sure what exactly up above helped solve the issue or if this is still open. Thank you.

taclane, 2021-08-23:
> Just adding that I too am having this issue and would be interested in workarounds. Not sure what exactly up above helped solve the issue or if this is still open. Thank you.

Seems like the issue is still open.  Before I found this thread, I spent quite a bit of time troubleshooting gain settings, antennas, and devices to see conventional p25 feeds go dark after about 90 minutes.  Since I'm running the docker version, I set up a cron job to restart the trunk-recorder container twice an hour.  Restarts are fairly fast on a raspi 4 (~20 sec), but it's certainly not an elegant solution.

I'll to try building it from source to see if gofaster's patches work any better, but a periodic restart seems adequate for the low call volume frequencies I was monitoring.

mjmikolay, 2021-10-20:
Thank you for confirming that. I will give it a try just in case but I'll continue to be patient.

taclane, 2021-11-20:
So https://github.com/robotastic/trunk-recorder/commit/d9fb8791b71e5582e24f2a11022677a10270445e helps a little, but maybe not for the right reasons.

After dropping some debug messages in main.cc, it seems like there are two problems.
1. When a Conventional P25 receiver sees a sufficiently strong analog signal (like a CW station ID) it makes a mess of the Gardner Costas block and nothing will decode correctly until it is reset.
2. When the demodulator is in that state, the P25 blocks aren't producing audio, so `get_current_length()` never goes above 0.

Trunk-recorder ends up swamped with junk data, but since the call length is 0 there is no way for `manage_conventional_call()` to escape the loop and terminate the call or reset the demodulator.  In [the above report](https://github.com/robotastic/trunk-recorder/issues/318#issuecomment-589084846) where "the ID transmission happens DURING the idle timeout from a previous P25 call", the call has already accrued some length before it started receiving garbage and will eventually negotiate the if/thens to terminate the call normally.

The reason why https://github.com/robotastic/trunk-recorder/commit/d9fb8791b71e5582e24f2a11022677a10270445e _sorta_ works, is that even after the costas block wanders off, it sill sometimes manages to decode a fraction of a second of garbled comms.  As soon as `get_current_length()` is > 0, the call can time out and reset the recorder/costas for the next transmission.

taclane, 2021-11-21:
For Conventional P25 recorders, when there is **an open squelch** and **a call length of 0** it seems to be a reliable indicator that either:
- An analog broadcast is occurring on that frequency 
- The demodulator is having issues and probably needs a reset

I'm testing the above theory with a couple basic fixes, but it's already at the point where it ran overnight just fine.  Still working after 17 hours is about 15 hours longer than it's ever gone without the need for a restart.

Looking back at how boatbod handled similar issues, it wasn't addressed in the op25 libraries because they weren't the issue.  The demodulator sometimes needs periodic resets, and due to the way `manage_conventional_call()` was, it wasn't setup to catch when the recorder was being fed a stream of non-P25 data.

If it still seems to be working tomorrow I can create a pull request.

taclane, 2021-12-10:
The [`auto-reset`](https://github.com/robotastic/trunk-recorder/tree/auto-reset) branch is currently available to test if there's anyone else still having problems with analog interference in conventional p25.

Using open/close squelch to reset the costas timer seems to be an effective fix for this issue.

taclane, 2021-12-15:
The P25 auto-reset patch is now part of the master branch.

As long as you've got the system squelch set correctly, this shouldn't be an issue anymore

robotastic, 2021-12-15:
I will close this for now, feel free to reopen if other issues occur

closed 2021-12-15 by robotastic