TrunkRecorder/trunk-recorder #319: Buffer Constraint Error

chrkyl, 2020-02-20:
Working with a fresh install, when I run recorder for the first time I get an error "max output buffer set to 807424 instead of requested 4096 terminate called after throwing instance of 'std::runtime_error'.

I think it is a problem with GNU Radio but I have no idea where to start looking.  Anyone have any ideas?

jcummingsg2, 2020-05-13:
I am having the same issue currently.  Fresh install of Ubuntu 18.04 server and trying to run Trunk Recorder.  Have 2 RTL_SDR's configured and configuring a smartnet trunk system.  Running ./recorder after build and config gives me the following error after setting up the system.

[2020-05-13 21:25:17.980490] (info)   Control channel: 852.887488 MHz
Nothing found
[2020-05-13 21:25:17.980662] (info)      smartnet Trunking single-stage decimator - Initial decimated rate: 0 Second decimated rate: 0 Initial Decimation: 0 System Rate: 2048000
[2020-05-13 21:25:17.984604] (info)      smartnet Trunking ARB - Initial Rate: 2048000 Resampled Rate: 18123 Initial Decimation: 113 System Rate: 18000 ARB Rate: 0.993213
gr::log :WARN: flat_flowgraph - Block (multiply_cc0) max output buffer set to 1082880 instead of requested 4096
terminate called after throwing an instance of 'std::runtime_error'
  what():  problems allocating a buffer with the given max output buffer constraint!
Aborted (core dumped)

SDLyle, 2020-06-02:
I'm having the same issue.  My first attempt was with rtlsdr and raspberry pi 4 with a fresh ubuntu 18.04 install as per the wiki.  I was able to successfully compile the software and received the buffer error when starting.   I then attempted to use the Docker method on a laptop running ubuntu 18.04.  I successfully ran Docker-build and when I executed Docker-up I again got the same error.  Both laptop and pi4 have gnuradio version 3.7.11-10  installed.  This software looks great and I really don't want to use unitrunker! l Many thanks!

gr::log :WARN: flat_flowgraph - Block (multiply_cc2) max output buffer set to 1082880 instead of requested 4096
recorder_1  | terminate called after throwing an instance of 'std::runtime_error'
recorder_1  |   what():  problems allocating a buffer with the given max output buffer constraint!

SDLyle, 2020-06-11:
Quick follow up, I've had a little progress getting past this issue by lowering the "Rate" value in the Sources section of the JSON config for my rtlsdr devices.  It makes covering the band a bit more difficult losing some of that bandwidth.  I'm working through some other issues so I'm not sure if this is actually solution but I no longer abort with that error.

robotastic, 2020-08-04:
I removed the code related to set_max_output_buffer(4096); Try pulling the latest version and reopen this thread if it doesn't work.

closed 2020-08-04 by robotastic