TrunkRecorder/trunk-recorder #320: What's wrong?

Nokoa, 2020-02-21:
I just can't seem to get it to record.  What's wrong?

config.json
`{
    "sources": [{
        "center": 772581250,
        "rate": 2400000,
        "ppm": 54.5,
        "gain": 46,
        "digitalRecorders": 20,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=1"
    },{
        "center": 770281250,
        "rate": 2400000,
        "ppm": -0.8,
        "gain": 46,
        "debugRecorders": 0,
        "digitalRecorders": 20,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=0"
    }, {
        "center": 774081250,
        "rate": 2400000,
        "ppm": -1.1,
        "gain": 46,
        "digitalRecorders": 20,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=2"
    }],
    "systems": [{
        "control_channels": [772306250],
        "type": "p25",
	"shortName": "svrcs",
        "talkgroupsFile": "svrcs.csv"
    }, {
        "control_channels": [772843750],
        "type": "p25",
	"shortName": "ebrcs",
        "talkgroupsFile": "ebrcs.csv"
    }],

	"captureDir" : "/home/mdt/Desktop/recordings"
}`

`mdt@mdt-BK285AA-ABA-600-1152:~/Desktop/trunk-recorder/trunk-build$ ./recorder
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[2020-02-21 14:36:01.508608] (info)   Using Config file: ./config.json

[2020-02-21 14:36:01.509063] (info)   
-------------------------------------
     Trunk Recorder
-------------------------------------
0
[2020-02-21 14:36:01.509095] (info)   
-------------------------------------
SYSTEMS
-------------------------------------
0
[2020-02-21 14:36:01.509224] (info)   

System Number: 1
-------------------------------------
1
[2020-02-21 14:36:01.509248] (info)   Short Name: svrcs
[2020-02-21 14:36:01.509264] (info)   System Type: p25
[... 400 lines cut ...]
[2020-02-21 14:36:14.009818] (error)   [ebrcs]	There is only one control channel defined
[2020-02-21 14:36:14.009856] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-02-21 14:36:17.006469] (error)   [svrcs]	 Control Channel Message Decode Rate: 2/sec, count:  6
[2020-02-21 14:36:17.006544] (error)   [ebrcs]	There is only one control channel defined
[2020-02-21 14:36:17.006567] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-02-21 14:36:20.008335] (error)   [svrcs]	 Control Channel Message Decode Rate: 2/sec, count:  6
[2020-02-21 14:36:20.008452] (error)   [ebrcs]	There is only one control channel defined
[2020-02-21 14:36:20.008502] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-02-21 14:36:23.006222] (error)   [svrcs]	 Control Channel Message Decode Rate: 2/sec, count:  6
[2020-02-21 14:36:23.006586] (error)   [ebrcs]	There is only one control channel defined
[2020-02-21 14:36:23.006641] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-02-21 14:36:26.005489] (error)   [svrcs]	 Control Channel Message Decode Rate: 2/sec, count:  6
[2020-02-21 14:36:26.005585] (error)   [ebrcs]	There is only one control channel defined
[2020-02-21 14:36:26.005628] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-02-21 14:36:29.013873] (error)   [svrcs]	There is only one control channel defined
[2020-02-21 14:36:29.013954] (error)   [svrcs]	 Control Channel Message Decode Rate: 1.33333/sec, count:  4
[2020-02-21 14:36:29.014013] (error)   [ebrcs]	There is only one control channel defined
[2020-02-21 14:36:29.014052] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-02-21 14:36:32.012116] (error)   [svrcs]	 Control Channel Message Decode Rate: 2/sec, count:  6
[2020-02-21 14:36:32.012322] (error)   [ebrcs]	There is only one control channel defined
[2020-02-21 14:36:32.012365] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
^C
 Signal caught!
[2020-02-21 14:36:33.477221] (info)   stopping flow graph
`

vabiro, 2020-02-22:
Hi,
What do you get when you run `rtl_test`? Have you tried removing and replacing the RTL-SDRs (I.e. power cycle)?

Try dropping the rate to 2040000. 

Try adding " ,buflen=65536". After the serial number of the device

Hope that helps

fractal74, 2020-02-22:
In your config file, try putting the serial numbers of the devices.

Instead of:
"device": "rtl=X"

try:
"device": "rtl=0000000X"

Nokoa, 2020-02-23:
I tried all those with no change. I played around with PPM also, although it should be properly tuned. I did successfully run SDRTrunk, and OP25 with those RTLs.

fractal74, 2020-02-23:
Maybe try using error instead of ppm in the config. 

Are those the only control channels for the systems you are trying? No alternate control channels currently active?

Nokoa, 2020-02-23:
I will try to do error. No, there are more control channels, but these I know work as when I am not trying to run Trunk Recorder I am running SDRTrunk and I am receiving through those 2 and they work.

mbt25, 2020-02-25:
EBRCS
12 freqs for site 1 Alco Southwest Simulcast

SVRCS
12 freqs for site 1 West

First off, define more than one control channel.   Put in all control channels for both EBRCS and SVRCS.

The, keep in mind that you are in total trying to capture activity from 24 frequencies:

769.29375
769.58125
769.86875
770.08125
770.33125
770.76875
771.04375
771.34375
771.40625
771.64375
771.85625
771.94375
772.15625
772.24375
772.30625
772.45625
772.54375
772.60625
772.84375
772.90625
773.18125
773.93125
774.40625
774.68125

So take those frequencies above and paste them into http://garvas.org/trunk-recorder/ .   Then set your sample rate

( 2.4 = 2400000 rate; 2.154 = 2154000 rate ; 2.040 = 2040000 rate)

Pay attention to how many digitalRecorders you should set up for each dongle.   You don't want 20 digitalRecorders for each dongle.

Here are the calculations based upon the frequencies you want to monitor -- dependent upon which rate you choose.

rate:  2400000
center freq 1:  770.46875 ; number of recorders:  10
center freq 2:  772.89375 ; number of recorders:  12
center freq 3:  774.54375 ; number of recorders:  2

rate:  2154000
center freq 1:  770.35    ; number of recorders:  9
center freq 2:  772.4125  ; number of recorders:  12
center freq 3:  774.30625 ; number of recorders:  3

rate:  2040000
center freq 1:  770.35    ; number of recorders:  9
center freq 2:  772.4125  ; number of recorders:  12
center freq 3:  774.30625 ; number of recorders:  3

Again, please add all control channels

For EBRCS:
"control_channels": [771943750, 772243750, 772543750, 772843750],

For SVRCS:
"control_channels": [771856250, 772156250, 772306250, 772456250],

Nokoa, 2020-02-26:
I made those changes and then it worked. It started displaying channels that it was motioning, and saving files. Then I stopped it. Restarted it and it's not working again.

```
{
    "sources": [{
        "center": 770468750,
        "rate": 2400000,
        "ppm": 54.5,
        "gain": 49.6,
        "digitalRecorders": 10,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=00000002"
    },{
        "center": 772893750,
        "rate": 2400000,
        "ppm": -0.8,
        "gain": 49.6,
        "debugRecorders": 0,
        "digitalRecorders": 12,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=00000001"
    }, {
        "center": 774543750,
        "rate": 2400000,
        "ppm": -1.1,
        "gain": 49.6,
        "digitalRecorders": 2,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=00000003"
    }],
    "systems": [{
	"control_channels": [771856250, 772156250, 772306250, 772456250],
        "type": "p25",
	"shortName": "svrcs",
        "talkgroupsFile": "svrcs.csv"
    }, {
	"control_channels": [771943750, 772243750, 772543750, 772843750],
        "type": "p25",
	"shortName": "ebrcs",
        "talkgroupsFile": "ebrcs.csv"
    }],

	"captureDir" : "/home/mdt/Desktop/recordings"
}
```

```
mdt@mdt-BK285AA-ABA-600-1152:~/Desktop/trunk-recorder/trunk-build$ ./recorder 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[2020-02-25 17:41:10.564453] (info)   Using Config file: ./config.json

[2020-02-25 17:41:10.565061] (info)   
-------------------------------------
     Trunk Recorder
-------------------------------------
0
[2020-02-25 17:41:10.565089] (info)   
-------------------------------------
SYSTEMS
-------------------------------------
0
[2020-02-25 17:41:10.565130] (info)   

System Number: 1
-------------------------------------
1
[... 395 lines cut ...]
[2020-02-25 17:42:25.016276] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-02-25 17:42:28.010159] (error)   [svrcs]	 Control Channel Message Decode Rate: 2/sec, count:  6
[2020-02-25 17:42:28.010243] (error)   [ebrcs] Retuning to Control Channel: 7.722438e+08
[2020-02-25 17:42:28.010333] (info)   	 - System Source - Min Freq: 7.717437e+08 Max Freq: 7.740438e+08
[2020-02-25 17:42:28.010512] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-02-25 17:42:31.003732] (error)   [svrcs]	 Control Channel Message Decode Rate: 2/sec, count:  6
[2020-02-25 17:42:31.003825] (error)   [ebrcs] Retuning to Control Channel: 7.725437e+08
[2020-02-25 17:42:31.003897] (info)   	 - System Source - Min Freq: 7.717437e+08 Max Freq: 7.740438e+08
[2020-02-25 17:42:31.004065] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-02-25 17:42:34.006143] (error)   [svrcs]	 Control Channel Message Decode Rate: 2/sec, count:  6
[2020-02-25 17:42:34.006245] (error)   [ebrcs] Retuning to Control Channel: 7.728438e+08
[2020-02-25 17:42:34.006306] (info)   	 - System Source - Min Freq: 7.717437e+08 Max Freq: 7.740438e+08
[2020-02-25 17:42:34.006497] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-02-25 17:42:37.009975] (error)   [svrcs]	 Control Channel Message Decode Rate: 2/sec, count:  6
[2020-02-25 17:42:37.010100] (error)   [ebrcs] Retuning to Control Channel: 7.719437e+08
[2020-02-25 17:42:37.010182] (info)   	 - System Source - Min Freq: 7.717437e+08 Max Freq: 7.740438e+08
[2020-02-25 17:42:37.010334] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-02-25 17:42:40.003536] (error)   [svrcs]	 Control Channel Message Decode Rate: 2/sec, count:  6
[2020-02-25 17:42:40.003633] (error)   [ebrcs] Retuning to Control Channel: 7.722438e+08
[2020-02-25 17:42:40.003697] (info)   	 - System Source - Min Freq: 7.717437e+08 Max Freq: 7.740438e+08
[2020-02-25 17:42:40.003861] (error)   [ebrcs]	 Control Channel Message Decode Rate: 0/sec, count:  0
[2020-02-25 17:42:43.007400] (error)   [svrcs]	 Control Channel Message Decode Rate: 2/sec, count:  6
[2020-02-25 17:42:43.007499] (error)   [ebrcs] Retuning to Control Channel: 7.725437e+08
[2020-02-25 17:42:43.007562] (info)   	 - System Source - Min Freq: 7.717437e+08 Max Freq: 7.740438e+08
```

mbt25, 2020-02-26:
Well if it worked, then you stopped it, then it would not work again when you started it -- as long as you didn't edit anything, just keep trying it.   Sometimes I'll fire it up and it won't work (rarely), and then I'll stop it and wait about 30 seconds and start it and voila it works again.

It's either going to be a PPM correction issue or a signal issue (signal overload or weak signal).   It can't manage to find a good signal on the EBRCS control channel freq.

Good spot for support:

https://gitter.im/trunk-recorder/Lobby#

gofaster, 2020-02-26:
mbt25 is right as to the likely causes.
A 100% decode rate for a P25 system is ~40/sec.  Your logs are indicating 1-2/sec for svrcs and 0 for ebrcs.   
Also for the rtl-sdr I believe the underlying osmosdr device driver actually truncates a defined ppm floating point value to an integer...

mbt25, 2020-02-26:
A gain of 49.6 (max gain) is pretty concerning to me .   I live in a rural area with the nearest 700/800 freq almost 10 miles away.   I really couldn't get away with a 49.6 gain setting.   I could, but my SNR would be less at 49.6 than it is at 32/33 -- and being on frequency and having a good SNR are what you are looking for.   If the systems are local systems to you, you should not need a gain setting more than about 32/33 I would think.   I haven't had to mess with ppm adjustments, thankfully.   Figuring out those adjustments would really throw me for a loop.

Nokoa, 2020-02-26:
I am monitoring two systems with shared SDRs. One system is 2 miles away, and the other is 20 miles away. It is semi receiving now I think. I don't think I have very good rates. I went back and tried to get a better error correction. I measured it to the middle of the control channel signal.

I was having issues using the 2.4 mhz frequencies because it kept saying that it cannot tune to 771.64385. So I changed it to the 2.04mhz frequencies. But when I changed the rate in the config it gave an error.

I think it's working now, I will be monitoring it and comparing to see how it goes.

Thanks for your help!

```
{
    "sources": [{
        "center": 772412500,
        "rate": 2400000,
        "error": -41650,
        "gain": 21,
        "digitalRecorders": 12,
        "modulation": "qpsk",
        "driver": "osmosdr",
	"digitalLevels" : 12,
        "device": "rtl=00000002"
    },{
        "center": 770350000,
        "rate": 2400000,
        "error": 1050,
        "gain": 35,
        "debugRecorders": 0,
        "digitalRecorders": 9,
        "modulation": "qpsk",
        "driver": "osmosdr",
	"digitalLevels" : 12,
        "device": "rtl=00000001"
    }, {
        "center": 774306250,
        "rate": 2400000,
        "error": 1250,
        "gain": 27,
        "digitalRecorders": 3,
        "modulation": "qpsk",
        "driver": "osmosdr",
	"digitalLevels" : 12,
        "device": "rtl=00000003"
    }],
    "systems": [{
	"control_channels": [772306250, 772156250, 771856250, 772456250],
        "type": "p25",
	"shortName": "svrcs",
        "talkgroupsFile": "svrcs.csv",
        "uploadScript" : "scripts/upload-call.sh 1"
    }, {
	"control_channels": [771943750, 772243750, 772543750, 772843750],
        "type": "p25",
	"shortName": "ebrcs",
        "talkgroupsFile": "ebrcs.csv",
        "uploadScript" : "scripts/upload-call.sh 31"
    }],

	"captureDir" : "/home/mdt/Desktop/recordings"
}

```

```
[2020-02-25 22:45:11.361521] (info)   Using Config file: ./config.json

[2020-02-25 22:45:11.361964] (info)   
-------------------------------------
     Trunk Recorder
-------------------------------------
0
[2020-02-25 22:45:11.361987] (info)   
[... 351 lines cut ...]
[2020-02-25 22:45:42.651147] (info)   [svrcs]	TG:       4777	Freq: 7.744063e+08	Starting Recorder on Src: rtl=00000003
[2020-02-25 22:45:45.432731] (info)   [svrcs]	TG:      35713	Freq: 7.726063e+08	Not Recording: ENCRYPTED 
[2020-02-25 22:45:46.010114] (info)   [ebrcs]	TG:       3407	Freq: 7.710438e+08	Ending Recorded Call - Last Update: 4s	Call Elapsed: 15
[2020-02-25 22:45:46.010463] (info)   	- Stopping P25 Recorder Num [12]	TG:       3407	Freq: 7.710438e+08 	TDMA: false	Slot: 0
[2020-02-25 22:45:46.010605] (info)   Running upload script: ./scripts/upload-call.sh 31 /home/mdt/Desktop/recordings/ebrcs/2020/2/25/3407-1582699531_770331250.wav &
[2020-02-25 22:45:46.264558] (info)   	- Starting P25 Recorder Num [12]	TG:       3405	Freq: 7.698687e+08 	TDMA: false	Slot: 0
[2020-02-25 22:45:46.264836] (info)   [ebrcs]	TG:       3405	Freq: 7.698687e+08	Starting Recorder on Src: rtl=00000001
Call imported successfully.
[2020-02-25 22:45:48.877472] (info)   	 - Retune failed, starting a new recording using a new source
[2020-02-25 22:45:48.877654] (info)   	- Starting P25 Recorder Num [0]	TG:       1651	Freq: 7.726063e+08 	TDMA: true	Slot: 1
[2020-02-25 22:45:48.877909] (info)   [svrcs]	TG:       1651	Freq: 7.726063e+08	Starting Recorder on Src: rtl=00000002
[2020-02-25 22:45:49.314237] (info)   	 - Message - 	TMDA: true 	Slot: 1	Call - 	TMDA: true	Slot: 0
[2020-02-25 22:45:49.314428] (info)   [svrcs]	TG:       4777	Freq: 7.746813e+08	Update Retuning - New Freq: 7.746813e+08	Elapsed: 7s 	Since update: 3s
[2020-02-25 22:45:51.008920] (info)   [svrcs]	TG:       1651	Freq: 7.744063e+08	Ending Recorded Call - Last Update: 4s	Call Elapsed: 11
[2020-02-25 22:45:51.009178] (info)   	- Stopping P25 Recorder Num [21]	TG:       1651	Freq: 7.744063e+08 	TDMA: true	Slot: 1
[2020-02-25 22:45:51.009281] (info)   Running upload script: ./scripts/upload-call.sh 1 /home/mdt/Desktop/recordings/svrcs/2020/2/25/1651-1582699540_774406250.wav &
[2020-02-25 22:45:51.138607] (info)   	- Starting P25 Recorder Num [13]	TG:       3407	Freq: 7.707688e+08 	TDMA: false	Slot: 0
[2020-02-25 22:45:51.138853] (info)   [ebrcs]	TG:       3407	Freq: 7.707688e+08	Starting Recorder on Src: rtl=00000001
[2020-02-25 22:45:51.887742] (info)   [svrcs]	TG:       4777	Freq: 7.731812e+08	Update Retuning - New Freq: 7.731812e+08	Elapsed: 9s 	Since update: 0s
[2020-02-25 22:45:52.009570] (info)   [svrcs]	TG:       1651	Freq: 7.726063e+08	Ending Recorded Call - Last Update: 4s	Call Elapsed: 4
[2020-02-25 22:45:52.009793] (info)   	- Stopping P25 Recorder Num [0]	TG:       1651	Freq: 7.726063e+08 	TDMA: true	Slot: 1
[2020-02-25 22:45:52.009879] (info)   Running upload script: ./scripts/upload-call.sh 1 /home/mdt/Desktop/recordings/svrcs/2020/2/25/1651-1582699548_772606250.wav &
Call imported successfully.
Call imported successfully.
```

robbiet480, 2020-06-02:
@Nokoa Any chance you could share ebrcs.csv?

closed 2020-06-07 by Nokoa