TrunkRecorder/trunk-recorder #321: Some recordings leak into the wrong files

Nokoa, 2020-02-28:
It appears that sometimes I have recording files for one thing that also contain parts from other talkgroups.

Attached, one is a fire dispatch tg and the other is a utilities tg. Note that the fire recording also contains apart of the utilities recording from the other file. 

Looking at the file name it seems as they were both transmitted very close to eachother, on the same frequencies. TDMA?
[2recordings.zip](https://github.com/robotastic/trunk-recorder/files/4265012/2recordings.zip)

bctrainers, 2020-02-28:
Assuming you're on the latest version (commit e987fceb3c081fa834dbfcc3b44a2c3c48709560 / e987fce ) of trunk-recorder, this is indeed still possible to happen and only occurs when there are rapid-frequency changes. This normally will occur when there's a high CPU load or high-load on the SDR device. The result is what you hear.

This issue was much worse in 2019, and was resolved (mostly) with the double decimation update. 

I've not dug too deep into the code of how TR/OP25 deals/handles the Control Channel to Voice channels assigned to each talk group. But more or less, it-appears-to, or at least seems to be an issue where the SDR/Trunk Recorder code is sluggish to change the freq slice fast enough from what the CC says.

Nokoa, 2020-07-07:
Thank you for providing the above information. This has still been occurring semi frequently using latest version. Just wanted to let you know in case you been working on it further...

Nokoa, 2021-03-13:
Just an update, this is still occurring frequently when audio from a different transmission is added at the end of a different audio file. I am recording  about 10 analog channels, and 2 sites of a P25 Phase 2  system.

gonzoblue, 2021-07-16:
I am having the same issue starting a few months back. I have updated to the latest several times since then and it is still happening. I have tried a nooelec and airspy SDR, same issue. This is on a Lenovo desktop that runs about 50% cpu most the time and is only used for trunk-recorder, so I thought resources would be OK. This is on a P25 Phase 1 system.

robotastic, 2021-09-02:
Could you give the `4.0-beta` branch of the code a try? https://github.com/robotastic/trunk-recorder/tree/4.0-beta 
I re-wrote the call handling so it should strongly separate everything!

Nokoa, 2021-09-02:
> Could you give the `4.0-beta` branch of the code a try? https://github.com/robotastic/trunk-recorder/tree/4.0-beta
> I re-wrote the call handling so it should strongly separate everything!

I will give it a try in the next few days. Thank you for pushing this change

ab0tj, 2021-09-02:
I also switched to rtl_tcp on the Pi and Trunk Recorder is now running on a VM on a much more powerful machine in case it was a resource issue. That didn't seem to help with the 3.x branch.

Nokoa, 2021-10-16:
Thanks for providing this update. I have a version from a couple days ago and for the most part it works great. However I noticed a couple very small instances where audio from another transmission is heard at end of another talkgroup transmissions. I will look into it some more to confirm that it’s actually occurring.

robotastic, 2021-10-16:
Thanks - Any chance you were also able to capture the log info the Trunk Recorder prints out when this overlap happened?  That would help narrow down what is causing it.

Nokoa, 2021-10-16:
Unfortunately I don't think I have log saving on, I couldn't locate a log file. But I just turned it on, so hopefully I can get capture it next time. 

On another note, I noticed that trunk-recorder is saving 2 audio files 1 wav and 1 m4a, not sure if I have something misconfigured or what not. I tried setting compressWav to false, and it didn't seem to change anything. This is an issue for me as a directory watcher from rdio-scanner was going through these folders and it would only delete the json and wav files, but I am left with all the m4a files.

carissavixen, 2021-10-16:
This looks identical to the issue I'm experiencing all the time on my install (dupsc21 on openmhz) - I'll check the logs for a definitive overlapping call period and post details in this thread.

carissavixen, 2021-10-17:
Okay, found a call this afternoon that contains the glitch. Here's the logs of that call. Specifically, Unit ID 1106852 should not be in the TG 2607 file - that's a unit on a completely different talkgroup (TG 9011), which was recorded correctly in its own file. I also included that call's log. It looks like, at least for this one that I grepped out, these have sequential call log numbers
[log.txt](https://github.com/robotastic/trunk-recorder/files/7358485/log.txt)
.

robotastic, 2021-10-17:
@carissavixen great job capturing this! It is really interesting - 
```
[2021-10-16 19:52:02.368859] (info)   [dupsc21]	32682C	TG:       2607	Freq: 7.71106e+08	Starting new Transmission 	Src ID:  279923
[2021-10-16 19:52:14.392620] (info)   [dupsc21]	32682C	TG:       2607	Freq: 7.71106e+08	Starting new Transmission 	Src ID:  279923
```
That second New Transmission, looks like it is actually from 1106852 and should have not been started. I will go double check the logic and see if some more checks are needed... and add some more debug messages to see if I can catch anything.

robotastic, 2021-10-18:
Ah! yep - I had it set to check against SmartNet talkgroups, and you have to shift left 4 times to get that number. I just changed it so it should check both SmartNet and P25 type talkgroup ids....

tdduarte, 2022-02-03:
Is there any updates on this... s'Seeing this a lot on a smart zone smartnet system here... gets really bad when the system is busy.. let me know if there is any info i can provide to help

robotastic, 2022-02-04:
@tdduarte If you are able to catch the log messages around when this is happening, that would be great. I thought I had mostly fixed this, but I could be missing something.

robotastic, 2024-12-15:
I think this has been addressed in later versions

closed 2024-12-15 by robotastic