TrunkRecorder/trunk-recorder #323: RPi4 + RTL-SDR can't tune P25 control channel

irishismyname, 2020-03-01:
**Observation**: I reinstalled/rebuilt trunk-recorder, copied my configs, talkgroups, and scripts from a working RPi3 w/buster and moved them to a fresh RPi4 w/buster. Trunk-recorder can't tune in the control channel unless _the center frequency of one of the SDR dongles is set to the control channel frequency_.

**Precident**: I noticed that @CharlesCage made a very similar observation in https://github.com/robotastic/trunk-recorder/issues/305#issuecomment-582580162.

**Sanity check**: Grabbed the `recorder` binary from the RPi3 build from a8f477a5fe7ac52c17180cc5b1f5f97c052ea233 and called it `recorder-old` and to my surprise it works as expected.

**config.json**: 
```
{
    "captureDir": "audio_files",
    "frequencyFormat": "mhz",
    "statusAsString": true,
    "controlWarnRate": -1,
    "logLevel": "debug",
    "sources": [
        {
            "center": 857000000,
            "rate": 2400000.0,
            "squelch": 0,
            "error": 440,
            "gain": 49.6,
            "antenna": "RX",
            "digitalRecorders": 4,
            "driver": "osmosdr",
            "device": "rtl=00000001,buflen=65536",
            "modulation": "qpsk"
        },
        {
            "center": 859000000,
            "rate": 2400000.0,
            "squelch": 0,
            "error": 950,
            "gain": 49.6,
            "antenna": "RX",
            "digitalRecorders": 4,
            "driver": "osmosdr",
            "device": "rtl=00000002,buflen=65536",
            "modulation": "qpsk"
        }
    ],
    "systems": [{
        "shortName": "NHC",
        "audioArchive": false,
        "control_channels": [856512500],
        "type": "p25",
        "talkgroupsFile": "talkgroups/nhc.csv",
	"uploadScript": "scripts/upload-call.sh 12",
        "hideEncrypted": true,
        "talkgroupDisplayFormat": "tag_id",
        "recordUnknown": true,
        "hideUnknownTalkgroups": false
    }]
}
```

gofaster, 2020-03-01:
Try it with the rate set to 2048000 and 2560000 and see if it works.

irishismyname, 2020-03-01:
No change at 2048000 and 2560000.

porcej, 2020-03-01:
I am having this same issue on several x86 based systems (running Ubuntu 18.04 and Ubuntu 19.10)

robotastic, 2020-03-02:
Well that is weird - I didn't have an issue with the Pi and I have it working with Ubuntu on an x86 box. 

Does rolling back to an earlier version help? Could you try compile one of the earlier releases? 

 It is weird that the earlier version worked. The latest version of Trunk Recorder does more condition of the channel. It maybe more picky on the tuning.

@irishismyname Your gain levels seem really high. Are you using an RTLSDR with a TXCO? I wouldn't expect to see that much error either. I would also make sure you have a really good power supply.

robotastic, 2020-03-02:
okay... I am an idiot. It looks like I never actually intially tuned the control channels. They only get tuned once the message count is low, but if you only have one control channel specified, it will never get retuned.
I always have a couple control channels in there, so I didn't catch this.

Try the latest update out.

porcej, 2020-03-02:
@robotastic it works! That seems to have fixed the issue!

robotastic, 2020-03-02:
Sweet! @irishismyname let me know if this fixes things for you too. Also, try adjusting the gain a bit.

irishismyname, 2020-03-02:
Good to go! I had been running it at 40.2 for no particular reason, but I maxed it out while trying to get running again. Here's what 49.6 looks like:
[image]

I figured since the signal wasn't clipping and the SNR looks to be around 40dB it was ok. How much is too much? Here's 24 dB gain:
[image]

closed 2020-03-02 by irishismyname