TrunkRecorder/trunk-recorder #325: Feature request: different uploads from 1 CC decoder.

buhalo1, 2020-03-15:
In some cases multiple agences presented in one system and much likely to be able to upload their traffic to different sections of web frontend (POLICE & EMS & FD for example). Now i have to create multiple decoders with different TG lists for single CC freqency (within the same site), which increases CPU consumption. 
Would be nice to be able to have multiple TG lists for 1 CC to assign different TG's to different system uploads. 

Thank you.

blantonl, 2020-03-15:
Why don't you just have your back-end uploader handle the routing to different front ends?

buhalo1, 2020-03-15:
This is my site config.
```
 [{
        "control_channels": [433000000],
	"type": "p25",
	"recordUnknown": "false",      
	"audioArchive": "false",        
	"callLog": "true",   
	"talkgroupDisplayFormat": "id_tag",    
	"talkgroupsFile": "talkgroups/ABC.csv",
	"shortName": "ABC",
	"uploadScript": "upload.sh 1"
     }]
```
"talkgroupsFile": "talkgroups/ABC.csv",
Contains whole list of TGs to be recorded.	
`"uploadScript": "upload.sh 1"`
Specifies to which frontend subsystem audio file will be uploaded.
I need to change "upload.sh #"` for different TG#s which are used by different agencies.
I see no way to make uploader handle the routing to different front ends untill
there is no support for more than one  
`"talkgroupsFile": "talkgroups/ABC.csv",`
string with its own 
`"uploadScript": "upload.sh 1"`
string.

Each recorded file contains TG# so it may be possible to implement secondary lookup routine within agency specific TG lists that were uploaded to frontend to decide which "upload.sh #" to use but anyway it's just too complex to me.

kcwebby, 2020-03-16:
I agree with Lindsey.

Create a new script, which looks at the talkgroup of the information being
passed to them, and routes it to the proper upload script based on what
talkgroup, or modify the upload script to do the same.

robotastic, 2020-08-04:
This seems like a sensible approach. Closing for now

closed 2020-08-04 by robotastic