TrunkRecorder/trunk-recorder #327: Feature request: Upload to Broadcastify Calls

blantonl, 2020-03-26:
See: https://wiki.radioreference.com/index.php/Broadcastify-Calls-API

An integrated option to upload to the Broadcastify Calls Platform.  

https://www.broadcastify.com/calls

robotastic, 2020-03-27:
Cool - I think this is pretty similar to what is being done with OpenMHz. I think what is in place in TR should work with this API. Do you know of things that would need to be changed?

blantonl, 2020-03-27:
I could adapt our API to use your current upload process for OpenMHZ with the following changes

1) Add a configuration option, per system, to specify the Broadcastify Calls System ID.  Keep the API key option and upload server options.

2)  Instead of prepending the short name to the upload URL path, simply call a static URL to do the multi-part form post.   Like:
 https://api.broadcastify.com/upload-call
 https://api.broadcasitfy.com/upload-call-dev

3)  Capture a return URL from the API call.  The return URL is a one-time use signed URL for upload

4)  Instead of sending the audio call as multi-part form data, instead upload the audio clip using an HTTP PUT to the URL that was returned from the API call.  Set the content type header to audio/aac

Everything else in the OpenMHZ upload implementation would stay the same.

blantonl, 2020-04-17:
Checking in on this... any updates?

robotastic, 2020-04-19:
This definitely seems doable, and I would be supportive of enabling an easy upload to broadcastify. These changes will take a bit of work. It probably makes the most sense to mod call-uploader.cc and have another one for broadcastify. Would you be able to submit a PR that has this functionality working? I can clean it up and make sure things are well integrated.

Also - any thoughts on integrating with the Radio Reference? It would be cool to have a Config & TG building tool. Maybe it could be a simple webapp that asks a series of questions and build a strawman config.json to get you started that has all of the freq for the system you want to scan.

blantonl, 2020-04-19:
I'm not a C++ developer so I would definitely be out of my realm here in developing for Trunk-Recorder.  

I might instead just go the route of using your call-uploader.cc - I've already figured out the back end components to take in the call data directly by just having someone specify our server instead of openMhz, however I was reluctant to do that since I didn't want to force people to have to make a choice between the two. 

I've documented the API pretty well... 

https://wiki.radioreference.com/index.php/Broadcastify-Calls-API

We're providing incentives to developers to get their apps aboard the BCFY Calls platform.  Basically stated, there is $2,500 waiting for someone to grab.  Since I can't do it, I'll allocate that $2,500 to whoever does the PR for Broadcastify Calls based on the terms outlined above in the developer incentives section of the API docs.

blantonl, 2020-04-19:
I can get a tool in place as well to help generate config.json files from the database.  Let me rotate that into the development pipeline on RR

blantonl, 2020-04-19:
I think we have a developer that is going to tackle this and submit a PR to get this functionality going... stay tuned.

PR #329 (2020-04-26): Adding built-in support for uploading to Broadcastify Calls
This PR introduces a way to upload calls to Broadcastify Calls without needing extra scripts. A user would update their `config.json` with new parameters mentioned in the README file for uploading to work. This has been developed to work alongside of OpenMHz.com uploads, instead of as a replacement to it.

I have this running on a system as a demonstration: https://www.broadcastify.com/calls-dev/systems/7

To handle the uploads, I added a dependency on `libcurl` since that seems to be much safer than using a custom HTTP implementation [... cut]
closed 2020-04-26 by robotastic