TrunkRecorder/trunk-recorder #328: Unable to retune System control channel, freq not covered by the Source used for the inital control channel freq

Pacmannion, 2020-04-05:
So, I was trying to get a trunk-recorder instance (Long story short, I've switched from using the Docker container) going and was having some issues. I've kept editing the config.json and to try resolve these issues, but now I keep getting the above error.

Here's the log:
`[2020-04-04 15:17:26.861970] (info)   [bosops]	Started with Control Channel: 8.570125e+08
Decim: 20 Decim2:  4
[2020-04-04 15:17:26.862206] (info)   	 P25 Trunking two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 2000000
[2020-04-04 15:17:26.868164] (info)   	 P25 Trunking ARB - Initial Rate: 2000000 Resampled Rate: 25000 Initial Decimation: 20 System Rate: 24000 ARB Rate: 0.96
[2020-04-04 15:17:29.003520] (error)   [bosops] Retuning to Control Channel: 8.574375e+08
[2020-04-04 15:17:29.003684] (info)   	 - System Source - Min Freq: 8.564877e+08 Max Freq: 8.583877e+08
[2020-04-04 15:17:29.003825] (error)   [bosops]	 Control Channel Message Decode Rate: 0.333333/sec, count:  1
[2020-04-04 15:17:32.004695] (error)   [bosops] Retuning to Control Channel: 8.584375e+08
[2020-04-04 15:17:32.004789] (info)   	 - System Source - Min Freq: 8.564877e+08 Max Freq: 8.583877e+08
[2020-04-04 15:17:32.004864] (error)   	 - Unable to retune System control channel, freq not covered by the Source used for the inital control channel freq.
[2020-04-04 15:17:32.004902] (error)   [bosops]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2020-04-04 15:17:35.006032] (error)   [bosops] Retuning to Control Channel: 8.594375e+08
[2020-04-04 15:17:35.006180] (info)   	 - System Source - Min Freq: 8.564877e+08 Max Freq: 8.583877e+08
[2020-04-04 15:17:35.006221] (error)   	 - Unable to retune System control channel, freq not covered by the Source used for the inital control channel freq.
[2020-04-04 15:17:35.006240] (error)   [bosops]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2020-04-04 15:17:38.006033] (error)   [bosops] Retuning to Control Channel: 8.570125e+08
[2020-04-04 15:17:38.006111] (info)   	 - System Source - Min Freq: 8.564877e+08 Max Freq: 8.583877e+08
[2020-04-04 15:17:38.006182] (error)   [bosops]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-04-04 15:17:41.007831] (error)   [bosops] Retuning to Control Channel: 8.574375e+08
[2020-04-04 15:17:41.007872] (info)   	 - System Source - Min Freq: 8.564877e+08 Max Freq: 8.583877e+08
[2020-04-04 15:17:41.007916] (error)   [bosops]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2020-04-04 15:17:44.007845] (error)   [bosops] Retuning to Control Channel: 8.584375e+08
[2020-04-04 15:17:44.008456] (info)   	 - System Source - Min Freq: 8.564877e+08 Max Freq: 8.583877e+08
[2020-04-04 15:17:44.008605] (error)   	 - Unable to retune System control channel, freq not covered by the Source used for the inital control channel freq.
[2020-04-04 15:17:44.008702] (error)   [bosops]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2020-04-04 15:17:47.010023] (error)   [bosops] Retuning to Control Channel: 8.594375e+08
[2020-04-04 15:17:47.010100] (info)   	 - System Source - Min Freq: 8.564877e+08 Max Freq: 8.583877e+08
[2020-04-04 15:17:47.010146] (error)   	 - Unable to retune System control channel, freq not covered by the Source used for the inital control channel freq.`

What perplexes me is that I didn't change the control channels in the config.json file, I was only editing the `talkgroupsFile `and `captureDir` entries to try and resolve my issues.

Here's my config.json file:
`{
	"sources": [{
		"center": 854912500.0,
		"rate": 2000000.0,
		"error": 70,
		"gain": 80,
		"digitalRecorders": 4,
		"driver": "osmosdr",
		"device": "rtl=0",
		"modulation": "qpsk"
	}, {
		"center": 857437700.0,
		"rate": 2000000.0,
		"error": 70,
		"gain": 80,
		"digitalRecorders": 4,
		"driver": "osmosdr",
		"device": "rtl=1",
		"modulation": "qpsk"
	}, {
		"center": 859312500.0,
		"rate": 2000000.0,
		"error": 70,
		"gain": 80,
		"digitalRecorders": 3,
		"driver": "osmosdr",
		"device": "rtl=2",
		"modulation": "qpsk"
	}],
	"systems": [{
		"control_channels": [857012500, 857437500, 858437500, 859437500],
		"type": "p25",
		"talkgroupsFile": "ChanList.csv",
		"recordUnknown": "false",
		"shortName": "bosops",
		"apiKey": "03450028d75b5e3d42ebad70f8d6122af05b4c0980e76071fad73137a06c02b6"
	}],
	"uploadServer": "https://api.openmhz.com",
	"captureDir": "audio_files"
}`

stonewalljones, 2020-04-06:
I have been having this issue too. Please let me know if you'd like any logs from me.

raygalasso, 2020-04-06:
It didn’t find the control channel on the first 3 frequencies and the 4th is on a different device.  Try putting the 4th first and see if it finds the control channel. Unfortunately all the control channels need to be covered by the same device.  

I had that issue until I found another log saying something about USB memory or buffers or something.  I don’t see that in your log though.  Maybe earlier in the log?  Also maybe try lowering the gain.

bctrainers, 2020-04-06:
> Unfortunately all the control channels need to be covered by the same device.

@robotastic - possible bug here.

kcwebby, 2020-04-06:
@Brett -- Not a Bug, a known limitation that has been discussed many
times.  It's a future enhancement.

blantonl, 2020-04-09:
The way to work around this is to create separate system entries for each of the control channels that are covered by each receiver.

In your case, your config.json would probably look like this

```
{
"sources": [{`
"center": 854912500.0,
"rate": 2000000.0,
"error": 70,
"gain": 80,
"digitalRecorders": 4,
"driver": "osmosdr",
"device": "rtl=0",
"modulation": "qpsk"
}, {
"center": 857437700.0,
"rate": 2000000.0,
"error": 70,
"gain": 80,
"digitalRecorders": 4,
"driver": "osmosdr",
"device": "rtl=1",
"modulation": "qpsk"
}, {
"center": 859312500.0,
"rate": 2000000.0,
"error": 70,
"gain": 80,
"digitalRecorders": 3,
"driver": "osmosdr",
"device": "rtl=2",
"modulation": "qpsk"
}],
"systems": [{
"control_channels": [857012500, 857437500, 858437500],
"type": "p25",
"talkgroupsFile": "ChanList.csv",
"recordUnknown": "false",
"shortName": "bosops1",
"apiKey": "xxxxxx"
},
{
"control_channels": [859437500],
"type": "p25",
"talkgroupsFile": "ChanList.csv",
"recordUnknown": "false",
"shortName": "bosops2",
"apiKey": "xxxxxx"
}],
"uploadServer": "https://api.openmhz.com",
"captureDir": "audio_files"
}
```

I'm not sure how this might impact OpenMHz uploading, but this is the workaround.

EricTendian, 2020-04-27:
For anyone experiencing this issue, I might have a long-term solution - consider trying out the changes in #334 and see if that helps.

closed 2020-05-29 by robotastic