TrunkRecorder/trunk-recorder #331: Delay of call uploading for Officer/Personnel safety

blue1sparky, 2020-04-24:
I am a super novice at coding. One item that makes me hesitant to place my local systems on OpenMhz is the possibility of certain comms being available in near real time possibly giving "bad guys" a tactical advantage. 

Would it be possible to place an item in the config file that allows an offset of x seconds (or minutes) that would allow us to "share the air" while maintaining a responsibility toward the safety of our local frontline personnel? (Or as a FF... in my case... possibly protecting myself.) 

This may also help keep local governments from being as quick to encrypt or at least help keep them from seeing this mindblowing new scanning technology as a threat to their operational security.

blantonl, 2020-04-25:
This should be done at the server side and is relatively easy to implement.  On the [Broadcastify Calls](https://www.broadcastify.com/calls) platform, we plan to restrict tactical frequencies and talkgroups to only the system broadcasters who send us call data.  

Implementing a delay on talkgroups should be a relatively simple process to happen at the centralized server level (OpenMHZ, BCFY Calls etc)

blue1sparky, 2020-04-25:
Limiting tac channel access on the system is a good start and would probably work well on county or state level trunked systems or conventional systems in large municipalitles, however in my area we have many communities that don't have a tac channel... their police and fire dispatch channels are also the tac channel. In my towns case we have 3 communities sharing the same single frequency with no tactical channel. The delay on all broadcasts, I think would help safety quite a bit. Randomizing that delay between site visits would further confound any bad actors trying to use the feeds for nefarious purposes. 

Just a thought.

blantonl, 2020-04-26:
I think that your thoughts and recommendations are items to be considered and made at individual  calls provider levels... not at the software capture level.

Other's can chime in on the applicability of this feature request to Trunk-recorder's development efforts, but this is all I've got for now.

robotastic, 2020-04-26:
@blue1sparky I am definitely supportive of adding some functionality to help with this. It might be possible on the Trunk Recorder side - I not quite sure of a good way to add it in. There could be some sort of queueing system that could delay some calls. It might be easier to do this in OpenMHz since that is in Javascript instead of C++. Unfortunately I don't have much free time these days and I am not sure when I would be able to tackle this.

If someone has thoughts on how to do this with Trunk Recorder, happy to work with them on it. On most systems the best option is to just block sensitive channels. Using trunk-player might let you stand-up more of a closed system and you could put a password on it.

porcej, 2020-04-27:
This can be done using a shell script to upload (encode-upload.sh in trunk-player) using the `sleep` command (ie `sleep 10s` to pause for 10 seconds).  [https://linuxize.com/post/how-to-use-linux-sleep-command-to-pause-a-bash-script/](https://linuxize.com/post/how-to-use-linux-sleep-command-to-pause-a-bash-script/)

robotastic, 2020-04-27:
Something like that could work! There could be a problem on busy systems if you are trying to delay for a long time because you would end up with 100s of paused threads... but I think you can increase the number of max threads.

Anyhow - this is a good idea! This wouldn't delay the upload to OpenMHz because that is handled inline, but this should work for Trunk Player. You might be able to script an upload to OpenMhz using encode-upload.sh and curl, but I am not quite sure how.

bctrainers, 2020-04-27:
Adding a sleep function into a bash script is likely the most optimal in this case, as it'd be a global delay.

Keep in mind that an individual recorder on TR will keep "recording" on an activated talkgroup until the timeout value has been met or exceeded. So in theory, long-duration transmissions, thus recordings in the TR sense, with multiple, back to back radio keyups, will continue to have delayed audio from being published as a wav file and then uploaded to where ever it may be.

At that point, you'd be needing to consider some logic in delay of publishing audio on short timed events vs long recording events.

blue1sparky, 2020-05-01:
I hadn't thought of a sleep command in the shell script. That's a great and super simple idea! I think the systems I'm working with are not so busy that it would cause huge problems with uploads. I understand that longer calls would be further delayed, but I would be OK with a longer delay if it could help protect both our public safety professionals AND our radio monitoring hobby. :-)  (I would say "scanning hobby" but trunk-recorder makes literal scanning a thing of the past.)

robotastic, 2021-04-28:
I am going to close this for now. On the TR side, I think the best option is to add a sleep function to the upload script. We might be able to do something fancier if the plugin system gets added.

I started a separate thread to track fixes on the server side here: https://github.com/robotastic/trunk-server/issues/2

closed 2021-04-28 by robotastic