TrunkRecorder/trunk-recorder #335: (error)   [MPSCS]	 Control Channel Message Decode Rate: 1/sec, count:  3

afera15, 2020-04-27:
I keep getting this when running: 

`[2020-04-27 12:01:55.009013] (error)   [MPSCS] Retuning to Control Channel: 8.527625e+08
[2020-04-27 12:01:55.009133] (info)   	 - System Source 2 - Min Freq: 8.522500e+08 Max Freq: 8.545500e+08
[2020-04-27 12:01:55.009283] (error)   [MPSCS]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2020-04-27 12:01:58.011219] (error)   [MPSCS] Retuning to Control Channel: 8.531375e+08
[2020-04-27 12:01:58.011324] (info)   	 - System Source 2 - Min Freq: 8.522500e+08 Max Freq: 8.545500e+08
[2020-04-27 12:01:58.011432] (error)   [MPSCS]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-04-27 12:02:01.013794] (error)   [MPSCS] Retuning to Control Channel: 8.527625e+08
[2020-04-27 12:02:01.013895] (info)   	 - System Source 2 - Min Freq: 8.522500e+08 Max Freq: 8.545500e+08
[2020-04-27 12:02:01.013998] (error)   [MPSCS]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-04-27 12:02:04.006183] (error)   [MPSCS] Retuning to Control Channel: 8.531375e+08
[2020-04-27 12:02:04.006272] (info)   	 - System Source 2 - Min Freq: 8.522500e+08 Max Freq: 8.545500e+08
[2020-04-27 12:02:04.006355] (error)   [MPSCS]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-04-27 12:02:07.008875] (error)   [MPSCS] Retuning to Control Channel: 8.527625e+08
[2020-04-27 12:02:07.008965] (info)   	 - System Source 2 - Min Freq: 8.522500e+08 Max Freq: 8.545500e+08
[2020-04-27 12:02:07.009051] (error)   [MPSCS]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-04-27 12:02:10.010714] (error)   [MPSCS] Retuning to Control Channel: 8.531375e+08
[2020-04-27 12:02:10.010806] (info)   	 - System Source 2 - Min Freq: 8.522500e+08 Max Freq: 8.545500e+08
[2020-04-27 12:02:10.010899] (error)   [MPSCS]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-04-27 12:02:13.012905] (error)   [MPSCS] Retuning to Control Channel: 8.527625e+08
`
My congif file:
`{
	"sources": [{
		"center": 774000000.0,
		"rate": 2400000,
		"ppm": 69.0,
		"gain": 35.0,
		"digitalLevels": 3,
		"digitalRecorders": 4,
		"driver": "osmosdr",
		"device": "rtl=0,buflen=65536",
		"modulation": "fsk4"
	},{
		"center": 851000000.0,
		"rate": 2400000,
		"ppm": 69.0,
		"gain": 35.0,
		"digitalLevels": 3,
		"digitalRecorders": 4,
		"driver": "osmosdr",
		"device": "rtl=1,buflen=65536",
		"modulation": "fsk4"
	},{
		"center": 853400000.0,
		"rate": 2400000,
		"ppm": 69.0,
		"gain": 35.0,
		"digitalLevels": 3,
		"digitalRecorders": 4,
		"driver": "osmosdr",
		"device": "rtl=2,buflen=65536",
		"modulation": "fsk4"
	}],
	"systems": [{
		"control_channels": [853137500, 852762500],
		"type": "p25",
		"shortName": "MPSCS",
		"broadcastifyApiKey": "*****",
		"broadcastifySystemId": "**",
		"audioArchive": false,
                "hideEncrypted": true
	}],
	"broadcastifyCallsServer": "https://api.broadcastify.com/call-upload",
	"controlWarnRate": 5,
	"callTimeout": 0.75
}`

bctrainers, 2020-04-27:
Not trying to sound mean, but if your RTL-SDR's have a PPM drift of 69 (!!!), you need to obtain newer/less-errored RTL-SDR's. Having a PPM that high is a tell-tale sign that the oscillator on those devices is bad. 

` Control Channel Message Decode Rate: 1/sec, count: 3` denotes that your RTLSDR array cannot 'see' the configured control channel. Be it due to drift, bad error/ppm configuration, no SDR covering that freq, or no CC is being broadcast on that frequency.

afera15, 2020-04-27:
This can be closed at it was PPM and Gain being set incorrectly

closed 2020-04-27 by afera15