TrunkRecorder/trunk-recorder #336: Generate conventional channel virtual TGID beyond the P25 TGID range (65535)

buhalo1, 2020-04-28:
Would be nice to have conventional channel virtual TGIDs beyond of P25 TGID range (65535).
100001, 100002, 100003, 100004, instead of 1, 2, 3, 4 etc.

robotastic, 2020-04-29:
Can you explain this a bit more? Does this help tell channels apart? Would the system shortname help with this? I think that limitation is in there now because of the variable size. It would be possible to change, but it would be quite a bit of work to make sure it is cleaned up everywhere.

buhalo1, 2020-04-29:
I use the rdio scanner as a front end. The subsystems inside it are divided according to their functional attribute. System A - police dept, system B - fire dept, etc. Each subsystem may contain conventional channels and truking talkgroups. The numbers of some conventional and trunking channels within the same system may coincide.

For example, in my case, talkgroups from 1 to 9 are used by the police department's communication technicians.

Surely this is not a very frequent occurrence, and if solving a problem requires a lot of coding, then maybe there is no point in doing it.

blantonl, 2020-04-29:
It is my opinion that this is not something that should be accomplished in the call capturing software, as the problem attempting to be solved is specific to one individual use case.

We handle this exact same scenario on the back end by assigning each captured system (trunked or conventional) as a unique system identifier.  For conventional systems, you can then manage each of those conventional channel captures using talkgroup slots as a subset of the unique system identifier.   You also have the added benefit of a unique key for document stores to then reference back to the call by key and timestamp.  A recommended solution might be:

Trunk Calls:
sid-talkgroupId
Example:  100-3415

Conventional Calls
c-sid-slotId
Example:  c-13-1

Couple that with the call's beginning timestamp and you get a unique key for every single call

A unique key identifer for each call can be

robotastic, 2020-05-01:
The best solution for this might be to make some changes in the upload script that sends the info to rdio scanner. I think you do some simple modifications of the JSON file and be able to swap around the talkgroup numbers based upon which system it is coming from.

closed 2021-04-28 by robotastic