TrunkRecorder/trunk-recorder #338: "audioArchive": false - metadata files remain

blantonl, 2020-04-30:
With "audioArchive" set to false metadata files still remain when uploading calls to Broadcastify.  

This results in directories full of metadata json files.

JustApu, 2020-04-30:
I noticed the same thing as I started uploading to Broadcastify. That said, I can see some value in having that data locally. I would suggest that "metadataArchive" be a separate option and not linked entirely to the "audioArchive" setting.

robotastic, 2020-05-01:
I hadn't run into this before. The `callLog` feature lets you turn off json generation, but for Broadcastify you need the json file. `audioArchive` only refers to the audio files. Coming up with a `callLogArchive` seems like the best path. This would let you selectively delete the json. I could see instance where you want the audio and not the json too. I will see if I can make this addition.

closed 2020-05-27 by robotastic

Ares4570, 2020-05-27:
After updating to the latest version that includes this, I'm now seeing an issue with duplicate call handling. BCFY generates an error (Broadcastify Metadata Upload Error: SKIPPED---ALREADY-RECEIVED-THIS-CALL) if a call has already been submitted, but this error seems to result in the call (both recording and JSON) being retained rather than deleted as they should be.

commit e9d5b678 (2020-05-27): Removing req. for callLog to be true for Broadcastify call uploads to work

Allows metadata files to be deleted after upload.

Fixes #338