TrunkRecorder/trunk-recorder #339: HACK RF ONE Control Channel Decode Error

h00fty, 2020-04-30:
I cannot seem to get a Hack RF One to decode a control channel. It gives the error

> [2020-04-30 00:44:30.002337] (error)   [sys_1]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2

I have gone over everything multiple times and I think I have it set up correctly. yet no matter what I do or adjust i get the same issue. 

Any help is greatly appreciated!

I am trying to tune into this [system](https://www.radioreference.com/apps/db/?sid=10521) using the south simulcast

here is my Config File
```
{
    "sources": [{
        "center": 853750000.0,
        "rate": 8000000.0,
        "error": 0,
        "gain": 14,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
	"device": "HackRF=687e07"
    }],
    "systems": [{
        "control_channels": [855712500,853137500,853437500,855962500],
        "type": "p25",
        "talkgroupsFile": "ChanList.csv"
	
 
   }],
	"captureDir": "/Users/josh/RECORDINGS"
} 
```

My error really is 0 when I checked in Gqrx.

And here is HackRF_info
```
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc30687e07
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00664763
```

And here is what i get when running ./recorder
```
Last login: Thu Apr 30 00:43:21 on ttys000
old-mbp:~ josh$ cd trunk-build
old-mbp:trunk-build josh$ ./recorder
[2020-04-30 00:44:16.009119] (info)   Using Config file: ./config.json
[2020-04-30 00:44:16.009773] (info)   
-------------------------------------
     Trunk Recorder
-------------------------------------
0
[2020-04-30 00:44:16.009808] (info)   
-------------------------------------
SYSTEMS
-------------------------------------
0
[2020-04-30 00:44:16.009845] (info)   

System Number: 1
-------------------------------------
1
[2020-04-30 00:44:16.009862] (info)   Short Name: sys_1
[... 107 lines cut ...]
[2020-04-30 00:44:18.735602] (info)   Call Timeout (seconds): 3
[2020-04-30 00:44:18.735624] (info)   Log to File: false
[2020-04-30 00:44:18.735639] (info)   Control channel warning rate: 10
[2020-04-30 00:44:18.735655] (info)   Control channel retune limit: 0
[2020-04-30 00:44:18.735671] (info)   Frequency format: 0
[2020-04-30 00:44:18.735686] (info)   Status as String: true
[2020-04-30 00:44:18.735701] (info)   Log Level: info
[2020-04-30 00:44:18.735736] (info)   [sys_1]	Started with Control Channel: 8.557125e+08
Decim: 80 Decim2:  4
[2020-04-30 00:44:18.736104] (info)   	 P25 Trunking two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 8000000
[2020-04-30 00:44:18.748259] (info)   	 P25 Trunking ARB - Initial Rate: 8000000 Resampled Rate: 25000 Initial Decimation: 80 System Rate: 24000 ARB Rate: 0.96
[2020-04-30 00:44:21.007318] (error)   [sys_1] Retuning to Control Channel: 8.531375e+08
[2020-04-30 00:44:21.007441] (info)   	 - System Source 0 - Min Freq: 8.497820e+08 Max Freq: 8.577180e+08
[2020-04-30 00:44:21.007541] (error)   [sys_1]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2020-04-30 00:44:24.005641] (error)   [sys_1] Retuning to Control Channel: 8.534375e+08
[2020-04-30 00:44:24.005748] (info)   	 - System Source 0 - Min Freq: 8.497820e+08 Max Freq: 8.577180e+08
[2020-04-30 00:44:24.005859] (error)   [sys_1]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-04-30 00:44:27.010629] (error)   [sys_1] Retuning to Control Channel: 8.559625e+08
[2020-04-30 00:44:27.010829] (info)   	 - System Source 0 - Min Freq: 8.497820e+08 Max Freq: 8.577180e+08
[2020-04-30 00:44:27.011005] (error)   [sys_1]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-04-30 00:44:30.002134] (error)   [sys_1] Retuning to Control Channel: 8.557125e+08
[2020-04-30 00:44:30.002235] (info)   	 - System Source 0 - Min Freq: 8.497820e+08 Max Freq: 8.577180e+08
[2020-04-30 00:44:30.002337] (error)   [sys_1]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2

```

bctrainers, 2020-04-30:
Gain values cannot be negative on the hackrf, and set the rate to 10000000 rather 8 mil. 

You may also need to obtain a txco from nooelec to fix the severe drifting issue that the hackrf has.

h00fty, 2020-05-01:
> Gain values cannot be negative on the > hackrf,

Mine is set to 14?

>and set the rate to 10000000 rather 8 >mil. 

No change

I’m beginning to think the signal is not good enough. However, I had a TRX-1 attached to the same antenna in the same location without issue. 

Would an actual scanner be better at decoding a weak signal than the SDR? 

This is all very new to me and I would love to get it working for the many benefits it has.

bctrainers, 2020-05-01:
Whatever is being passed to your TrunkRecorder from the config is reporting a gain of -14.

```
[2020-04-30 00:44:16.010625] (info)   PPM Error: 0
[2020-04-30 00:44:16.010640] (info)   Gain: -14
[2020-04-30 00:44:16.010658] (info)   IF Gain: 0
```

Also, use ifGain and bbGain in place of gain for the HackRF. Normal gain does not quite function as you'd think on it. :)

h00fty, 2020-05-01:
Thanks. I think i was messing around with different values when i captured the ./recorder response. I have it set at 14. Ive been experimenting with lots of different variables over the last 48 hours without success. 
[image]
This is what my Gqrx looks like. Im able to manually tune into voice channels and hear the traffic.

[image]
I think this is the control channel and i have adjusted the error to match

[image]
and these are the gain settings im using (updated my Config.json to match)

This is what im currently working with and the resulting errors....

```
{
    "sources": [{
        "center": 853500000.0,
        "rate": 10000000.0,
	"error": -15000,
        "gain": 14,
        "ifGain": 16,
        "bbGain": 22,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
	"device": "HackRF=687e07"
    }],
    "systems": [{
        "control_channels": [855712500],
        "type": "p25",
        "talkgroupsFile": "ChanList.csv"	
   }],
	"captureDir": "/Users/josh/RECORDINGS"
}
```

```
Last login: Thu Apr 30 17:06:45 on ttys000
old-mbp:~ josh$ cd trunk-build
old-mbp:trunk-build josh$ ./recorder
[2020-04-30 17:42:56.675027] (info)   Using Config file: ./config.json
[2020-04-30 17:42:56.675937] (info)   
-------------------------------------
     Trunk Recorder
-------------------------------------
0
[2020-04-30 17:42:56.676023] (info)   
-------------------------------------
SYSTEMS
-------------------------------------
0
[2020-04-30 17:42:56.676110] (info)   

System Number: 1
-------------------------------------
1
[2020-04-30 17:42:56.676157] (info)   Short Name: sys_1
[2020-04-30 17:42:56.676170] (info)   System Type: p25
[2020-04-30 17:42:56.676194] (info)   Control Channels: 
[2020-04-30 17:42:56.676222] (info)   8.55712e+08 
[2020-04-30 17:42:56.676252] (info)   API Key: 
[2020-04-30 17:42:56.676267] (info)   Broadcastify API Key: 
[2020-04-30 17:42:56.676284] (info)   Broadcastify Calls System ID: 0
[2020-04-30 17:42:56.676308] (info)   Upload Script: 
[2020-04-30 17:42:56.676321] (info)   Call Log: true
[2020-04-30 17:42:56.676345] (info)   Audio Archive: true
[2020-04-30 17:42:56.676359] (info)   Loading Talkgroups...
[2020-04-30 17:42:56.676436] (info)   Read 1 talkgroups.
[2020-04-30 17:42:56.676471] (info)   Talkgroups File: ChanList.csv
[2020-04-30 17:42:56.676487] (info)   Loading Unit Tags...
[2020-04-30 17:42:56.676506] (error)   Error Opening Unit Tag File: 
[... 88 lines cut ...]
[2020-04-30 17:42:59.725231] (info)   Status Server: 
[2020-04-30 17:42:59.725245] (info)   Instance Key: 
[2020-04-30 17:42:59.725259] (info)   Instance Id: 
[2020-04-30 17:42:59.725273] (info)   Broadcast Signals: false
[2020-04-30 17:42:59.725287] (info)   Default Mode: digital
[2020-04-30 17:42:59.725300] (info)   Call Timeout (seconds): 3
[2020-04-30 17:42:59.725320] (info)   Log to File: false
[2020-04-30 17:42:59.725334] (info)   Control channel warning rate: 10
[2020-04-30 17:42:59.725350] (info)   Control channel retune limit: 0
[2020-04-30 17:42:59.725366] (info)   Frequency format: 0
[2020-04-30 17:42:59.725380] (info)   Status as String: true
[2020-04-30 17:42:59.725394] (info)   Log Level: info
[2020-04-30 17:42:59.725428] (info)   [sys_1]	Started with Control Channel: 8.557125e+08
Decim: 100 Decim2:  4
[2020-04-30 17:42:59.725632] (info)   	 P25 Trunking two-stage decimator - Initial decimated rate: 100000 Second decimated rate: 25000 FA: 6250 FB: 12500 System Rate: 10000000
O[2020-04-30 17:42:59.735641] (info)   	 P25 Trunking ARB - Initial Rate: 10000000 Resampled Rate: 25000 Initial Decimation: 100 System Rate: 24000 ARB Rate: 0.96
OO[2020-04-30 17:43:02.010753] (error)   [sys_1]	There is only one control channel defined
[2020-04-30 17:43:02.010835] (error)   [sys_1]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
[2020-04-30 17:43:05.008302] (error)   [sys_1]	There is only one control channel defined
[2020-04-30 17:43:05.008369] (error)   [sys_1]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-04-30 17:43:08.006971] (error)   [sys_1]	There is only one control channel defined
[2020-04-30 17:43:08.007021] (error)   [sys_1]	 Control Channel Message Decode Rate: 1/sec, count:  3
[2020-04-30 17:43:11.009938] (error)   [sys_1]	There is only one control channel defined
[2020-04-30 17:43:11.010049] (error)   [sys_1]	 Control Channel Message Decode Rate: 0.666667/sec, count:  2
```

bctrainers, 2020-05-01:
If you're able, could you jump onto our Gitter.im chat? 

https://gitter.im/trunk-recorder/Lobby

closed 2020-06-11 by h00fty