TrunkRecorder/trunk-recorder #341: fdkaac issue

mattmac553, 2020-04-30:
`[2020-04-30 18:14:41.007855] (info)   	- Stopping P25 Recorder Num [2]	TG:       2331	Freq: 8.594625e+08 	TDMA: false	Slot: 0
sh: fdkaac: command not found
sox FAIL sox: `-' error writing output file: Broken pipe
[2020-04-30 18:14:41.222873] (error)   [jaxp25]	TG: 2331	Freq: 8.594625e+08	OpenMHz Upload Error: 
[2020-04-30 18:14:46.696743] (info)   [jaxp25]	TG:       2555	Freq: 8.569875e+08	Not Recording: `ENCRYPTED``

I redownloaded trunk-recorder on my Mac now I can not get it to run right any help?

bctrainers, 2020-04-30:
I don't use Mac, but you'll need to use `brew` to install sox, http://macappstore.org/sox/

mattmac553, 2020-04-30:
Thank You, but I am still getting: 
sh: fdkaac: command not found
[2020-04-30 19:04:36.186195] (error)   [jaxp25]	TG: 2329	Freq: 8.569375e+08	OpenMHz Upload Error: 

I have downloaded and reinstalled Fmpeg  with the FDKAAC multiple times.    what am I missing?

gofaster, 2020-04-30:
Hi.  It looks like you are missing the fdkaac command line encoder that sox is piping to.
Try [https://brewinstall.org/install-fdk-aac-encoder-on-mac-with-brew/](url)

mattmac553, 2020-04-30:
Thank you @gofaster `brew install fdk-aac-encoder` worked  and  Thank you @ @bctrainers `brew install sox`  both seemed to fix it!

closed 2020-04-30 by mattmac553