TrunkRecorder/trunk-recorder #342: Check for fdkaac and sox on startup when uploading is enabled

blantonl, 2020-05-01:
It would be good to check for working copies of sox and fdkaac when uploading is enabled for OpenMHZ and/or Broadcastify.

In Broadcastify Calls case... orphaned metadata entries (no audio) are set up to the server in cases where the user doesn't install the required prereqs... fdkaac and sox.

bctrainers, 2020-05-01:
I have been going through some wiki pages tonight tweaking some text and freshening up some package details. Some URL's are currently changing for slightly better organization.

I noticed that the MacOS page needs a bit of updating as well. I also noted an issue ticket for the same thing today.

robotastic, 2020-05-01:
This is a good idea! It will probably save a lot of heart ache. I will queue this up as a change to work on.

@bctrainers - Thanks for tidying up the wiki. I sometimes forget to make updates there.

commit faaf1162 (2020-05-27): Do not proceed with upload if audio conversion fails (fixes #342)
closed 2020-05-27 by robotastic