TrunkRecorder/trunk-recorder #343: Encrypted calls are still getting recorded and uploaded && Repeat transmissions across multiple TGs

mattmac553, 2020-05-01:
[https://openmhz.com/system/jaxp25](url)

Two part question:

A lot of encrypted calls are still getting through, any way to stop it? 

Most calls are getting stopped and it shows in the terminal:  `	Not Recording: ENCRYPTED `

But a lot are still being recorded that are obviously encrypted. hideEncrypted is set to default false.

Also I noticed with a call goes out through multiple talk groups, it records it on all, is there a way to truncate them so it does not play 4 times when ever a medical call is dispatched?

kcwebby, 2020-05-03:
I suspect the encrypted traffic that is being recorded is not encrypted at the "system" level.   Trunk-Recorder is looking for Channel Grant messages that are flagged with the Encryption tag set.   If there are switchable on/off encryption in the subscriber units (radios) that the system does not control then that flag will not get set.

This happens alot with ADP encryption, because its just a very simple crypto that requires little to no support from the trunk-system controllers.   As we move into more mainstream DES / AES where certificates and public/private keys are issued and revoked at the system level, the Encryption Flag will be set on the CC when the channel is granted to a subscriber.

Generally if you have it set to not record encrypted, and its still recording, then the control channel is not commanding encryption, just the radios on the other end.   In that case, the trunk-system is just repeating whatever it hears from the input radio to the output frequencies.

As for the multiple talkgroups, that is a much more complicated question.  There are many things that can result in multiple recordings.  Patches, recording multiple sites in the same system, etc.  Generally its a combo of things to fix it, mostly coming from your configuration, and using CSV files to ensure only one site is recording a specific talkgroup.

Hope this helps.

bctrainers, 2020-05-03:
The only three times that this will happen is of the following:

1. Poor decoding of a system (generally a decode rate of 20 or less causes Enc TG's to be recorded)
2. The talkgroup in use is not encrypted at the system level, and it is selective encryption.
3) The talkgroup in use is mixed clear and encryption. If the first transmission is encrypted, and the next is clear - it will be skipped until the timeout value is met. If the first transmission activation is in the clear, but the next key up is Encrypted, trunk recorder will record both the clear and encrypted chatter.

leee, 2020-07-26:
Just to contribute some cents:

> I suspect the encrypted traffic that is being recorded is not encrypted at the "system" level. Trunk-Recorder is looking for Channel Grant messages that are flagged with the Encryption tag set. If [...]

This is not how it works. The protection (encryption) bit in a channel grant does not stay always clear (or in your parlance, "unset", although it's always set `0` unprotected or `1` protected) regardless of if the traffic is secure and when a TG is "configured" to be used with selectable encryption.

An inbound channel grant service request and an outbound channel grant message both have the same service options field. That field contains a bit for whether or not a TG is protected or not. A subscriber can request a channel grant service to be encrypted or not, and a system can grant that with or without protection, and the transmitting subscriber follows as commanded.

You can also provision a talkgroup to be strapped clear/secure on a system level. If a subscriber then attempts to transmit in a restricted mode, it'll bonk out. But this does not translate to a "TG is always secure flag" - this simply does not exist.

> This happens alot with ADP encryption, because its just a very simple crypto that requires little to no support from the trunk-system controllers.

This is not how it works. Controllers don't manage keys. Systems dutifully transport traffic, secure or not, from site to site (whether that be say a RF site or dispatch site). Crypto material is stored locally on subscribers - not just RF subscribers, but consoles as well. By extension, that means traffic is encrypted/decrypted on a subscriber basis. Not during transport within a system.

> As we move into more mainstream DES / AES where certificates and public/private keys are issued and revoked at the system level, the Encryption Flag will be set on the CC when the channel is granted to a subscriber.

There is no public key cryptography in use for OTA traffic. It's all symmetric. If it were public key and every subscriber had their own key-pair, and you wanted to "protect" outbound traffic, your system would have to transmit `n` unique transmissions for `n` receiving SUs. If you want to "protect" inbound traffic, your system would have to mangle inbound traffic and do it live - decrypt said traffic and then transmit what needs transmitting.

---

@bctrainers' experience (points 1 and 3) has been mine as well. This software behavior is kind of a shame, as a "call" that starts encrypted but continues with other participants being in the clear does not get recorded.

At the risk of angering inode-constrained trunk-recorder installations, perhaps calls could get chunked into individual transmissions, but I'm not familiar enough to know just how much overhead this introduces (having to either instantiate a new recorder for every transmission, or interrupting a wav sink after every transmission).

> Also I noticed with a call goes out through multiple talk groups, it records it on all, is there a way to truncate them so it does not play 4 times when ever a medical call is dispatched?

Patch and MSel behavior is all sorts of fun stuff. MSel dedupe wouldn't be trivial (there are a lot of factors that come in to play from talkgroup configuration to what's physically configured at the infrastructure level - maybe base it on CC language as well as audio and call file similarity comparison)? But looks like patches is being worked on in #368.

Since this issue discusses two separate problems, the Patch/MSel topic ought to be rolled #368, the discussion of call management for selectable secure conversations can be made into a new one, and this issue be closed.

Also @kcwebby, while I have your attention - for some housekeeping of issues, I think you could close #108 and roll it into #368. Thanks!

Dygear, 2020-10-18:
```
[2020-10-17 19:48:10.005734] (info)   [ D0 ] State: inactive
[2020-10-17 19:48:10.005860] (info)   [ D1 ] State: inactive
[2020-10-17 19:48:10.005983] (info)   [ D2 ] State: inactive
[2020-10-17 19:48:10.006094] (info)   [ D3 ] State: inactive
[2020-10-17 19:48:13.210108] (info)             Not recording talkgroup 1301 (3/6 Main), priority is 9 but only 4 recorders available                                                        
[2020-10-17 19:48:34.417758] (info)   [NCPD]    TG:       1401 (               4/5 Main)        Freq: 500.875000 MHz    Not Recording: ENCRYPTED                                             
[2020-10-17 19:48:47.201953] (info)     - Starting P25 Recorder Num [0] TG:      21100 (           LongBeach PD)        Freq: 500.975000 MHz    TDMA: false     Slot: 0 Mod: false
[2020-10-17 19:48:47.202360] (info)   [NCPD]    TG:      21100 (           LongBeach PD)        Freq: 500.975000 MHz    Starting Recorder on Src: rtl=NCPD
[2020-10-17 19:48:56.258582] (info)   [NCPD]    TG:      21100 (           LongBeach PD)        Freq: 500.475000 MHz    Update Retuning - New Freq: 500.475000 MHz      Elapsed: 9s     Since update: 2s
[2020-10-17 19:49:01.217634] (info)   [NCPD]    TG:       1301 (               3/6 Main)        Freq: 501.000000 MHz    Not Recording: ENCRYPTED 
[2020-10-17 19:49:06.012950] (info)   [NCPD]    TG:      21100 (           LongBeach PD)        Freq: 500.775000 MHz    Update Retuning - New Freq: 500.775000 MHz      Elapsed: 19s    Since update: 2s
[2020-10-17 19:49:22.002038] (info)   [NCPD]    TG:      21100 (           LongBeach PD)        Freq: 500.775000 MHz    Ending Recorded Call - Last Update: 4s  Call Elapsed: 35
[2020-10-17 19:49:22.004701] (info)     - Stopping P25 Recorder Num [0] TG:      21100 (           LongBeach PD)        Freq: 500.775000 MHz    TDMA: false     Slot: 0
[2020-10-17 19:49:22.006755] (info)   Running upload script: ./MimoUpload.sh ./audio/NCPD/2020/10/17/21100-1602978527_500975000.wav &
[2020-10-17 19:49:22.353619] (info)   [NCPD]    TG:      21100 (           LongBeach PD)        Freq: 501.550000 MHz    Not Recording: ENCRYPTED
```
I have a config file that only allows for 4 digital recorders. I've set the encrypted talk groups to priority 9 (so that they will never be recorder EVEN IF it doesn't decode correctly and it thinks it's a encrypted transmission.) The problem is that sometimes, it doesn't always work. Long Beach PD is always encrypted and yet it's still tying to record that audio. I would like for it to always ignore ENCRYPTED when that's set in the talkgroups.csv file.

It's not always recognized that the channel is always encrypted. I would like that if I set the E `MODE` flag in the `Talkgroups.csv` file that it would always assume it's encrypted and is thus never recorded. Furthermore, when I set a channel `Priority` in the `Talkgroups.csv` file to greater than the number of available recorders, that it never records that channel. I put the information in the file so that I know quickly what talk groups are active on the system, and it translates their TGIDs to the `Alpha TAG` information that is easily and literally human readable. The only way to reliably ignore a channel is to simply not include it as apart of the `Talkgroups.csv` file and set the `recordUnknown` to `true` in the `config.json` file for that system.

bctrainers, 2020-10-18:
> The only way to reliably ignore a channel is to simply not include it as apart of the Talkgroups.csv file and set the recordUnknown to true in the config.json file for that system.

I assume you meant setting recordUnknown to false? Otherwise, you'll end up recording everything regardless of priorities in the csv file.

Dygear, 2020-10-18:
> I assume you meant setting recordUnknown to false? Otherwise, you'll end up recording everything regardless of priorities in the csv file.

Yep, sorry, that's exactly what I meant. Set to false.

ginomiglio, 2020-10-18:
Talkgroup priority was broken until ~2 weeks ago: #404

If you haven't pulled in new changes since October 1st, just update to the latest revision and it should work as expected. If you can't update to the latest version, setting a priority of 100 or higher will ignore the talkgroup entirely.

Dygear, 2020-10-18:
@ginomiglio According to `git status` I was up to date with master and I still had the issues. My pull request above fixes the problem outright by ignoring `E` `MODE` Talk Groups in the `talkgroup.csv` file.

PR #413 (2020-10-18): Enforce MODE flag for Encrypted in `talksgroups.csv`.
Fixes #343 by enforcing the `MODE` flag in the `Talkgroups.csv` file. If you set the `MODE` flag to `E` it will _always_ ignore that talk-group as encrypted.

Keep in mind that `Talkgroup::Talkgroup` mode is a single char, as such you can only set to `E` and not `DE`. The only other place this is checked is for `A` as in analog, so it shouldn't break anything else. Also fine with me as encrypted also implies that it must be digital.
closed 2020-10-18 by robotastic