TrunkRecorder/trunk-recorder #344: Only first transmission of conversation being recorded on SmartNet

sssmc, 2020-05-02:
I am monitoring my local SmartNet system quite successfully besides one issue. Just on analog talkgroups it appears that the first transmission of a conversation will come in fine but any following transmission are on the wrong frequency(resulting in static, audio from another talkgroup or undecoded p25 from a digital talkgroup). I don't see any errors or anything out of the ordinary on the console log, I am seeing the update returning logs for these conversations, and all the p25 digital channels work just fine. This is occurring 100% of the time for me.  

I have tried every config option that I can think of that could effect this. Here is my config file:
[config.txt](https://github.com/robotastic/trunk-recorder/files/4567180/config.txt)

Here is zip containing two audio files, one from the analog system with the issue, and one from a different digital system that broadcasts the some talkgroug, to show what I believe the transmission should sound like.  
[audio.zip](https://github.com/robotastic/trunk-recorder/files/4567174/audio.zip)

And here is the radio reference page for the system: [Here](https://www.radioreference.com/apps/db/?sid=2589)

robotastic, 2020-05-12:
Hmm... I bet I have some error in the way I am figuring out channel frequency. I haven't worked too much with  in ~150mhz region. The math that SmartNet uses is really wonky and not documented anywhere. Are you able to also monitor in GQRX and see what frequency is actaully being used and compare that to the one that TRunk Recorder picks?

robotastic, 2020-08-04:
@leee You recently dove into the Smartnet code. Did you notice anything weird in the way freq is calculated? I have never gone below 400MHz, so my math may have been off.

leee, 2020-08-04:
Nothing immediately comes to mind (which is already foggy on the old parser). I suggest to the reporter to try out the latest master and for the time being close this issue.

If this behavior continues to occur, please reopen the issue and report back. Additionally, please include:
- logs (a paste of an example offending time and a couple of seconds before and after works)
- call data (seeing what radio IDs if any are coming up in the jsons might be useful indication)

If something interesting is going on, I may want the reporter to build a copy of recorder that also logs raw OSWs to disk for analysis.

robotastic, 2020-08-04:
@sssmc Circle back if you are still having issues.

closed 2020-08-04 by robotastic