TrunkRecorder/trunk-recorder #345: Failed parsing Config: std::exception

kcwebby, 2020-05-02:
When starting recorder (fresh install of latest commit):
I've validated my config several times.  Config.test is a download one of the example configs to remove the chance that I have something wrong in my config.json file.

Anyone had any luck figuring this error out?  Of course, like all other errors, you must kill the process to exit trunk-recorder from this.

` ./recorder --config=config.test
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

[2020-05-02 16:10:46.632867] (info)   Using Config file: config.test

[2020-05-02 16:10:46.633134] (info)
-------------------------------------
     Trunk Recorder
-------------------------------------

[2020-05-02 16:10:46.633148] (info)
-------------------------------------
SYSTEMS
-------------------------------------

[2020-05-02 16:10:46.633166] (info)

System Number: 1
-------------------------------------

[2020-05-02 16:10:46.633197] (info)   Short Name: dcfems
[2020-05-02 16:10:46.633206] (info)   System Type: p25
[2020-05-02 16:10:46.633214] (info)   Control Channels:
[2020-05-02 16:10:46.633245] (info)     8.554625e+08
[2020-05-02 16:10:46.633297] (info)   API Key:
[2020-05-02 16:10:46.633313] (info)   Broadcastify API Key:
[2020-05-02 16:10:46.633321] (info)   Broadcastify Calls System ID: 0
[2020-05-02 16:10:46.633347] (info)   Upload Script:
[2020-05-02 16:10:46.633358] (info)   Call Log: true
[2020-05-02 16:10:46.633369] (info)   Audio Archive: true
[2020-05-02 16:10:46.633377] (info)   Loading Talkgroups...
[2020-05-02 16:10:46.633390] (error)   Error Opening TG File: ChanList.csv

[2020-05-02 16:10:46.633398] (info)   Talkgroups File: ChanList.csv
[2020-05-02 16:10:46.633405] (info)   Loading Unit Tags...
[2020-05-02 16:10:46.633431] (info)   Unit Tags File:
[2020-05-02 16:10:46.633441] (info)   Record Unknown Talkgroups: true
[2020-05-02 16:10:46.633451] (info)   Decode MDC: false
[2020-05-02 16:10:46.633460] (info)   Decode FSync: false
[2020-05-02 16:10:46.633469] (info)   Decode Star: false
[2020-05-02 16:10:46.633477] (info)   Decode TPS: false
[2020-05-02 16:10:46.633492] (info)   Talkgroup Display Format: Id
[2020-05-02 16:10:46.633503] (info)   Hide Encrypted Talkgroups: false
[2020-05-02 16:10:46.633517] (info)   Hide Unknown Talkgroups: false
[2020-05-02 16:10:46.633537] (info)   Minimum Call Duration (in seconds): 0
[2020-05-02 16:10:46.633561] (info)
[2020-05-02 16:10:46.633569] (info)

-------------------------------------
SOURCES
-------------------------------------

[2020-05-02 16:10:46.633592] (info)   Driver: osmosdr
[2020-05-02 16:10:46.633600] (info)   Center: 8.578000e+08
[2020-05-02 16:10:46.633617] (info)   Rate: 2048000
[2020-05-02 16:10:46.633632] (info)   Error: 1000
[2020-05-02 16:10:46.633645] (info)   PPM Error: 0
[2020-05-02 16:10:46.633655] (info)   Gain: 39
[2020-05-02 16:10:46.633667] (info)   IF Gain: 0
[2020-05-02 16:10:46.633677] (info)   BB Gain: 0
[2020-05-02 16:10:46.633687] (info)   LNA Gain: 0
[2020-05-02 16:10:46.633697] (info)   PGA Gain: 0
[2020-05-02 16:10:46.633706] (info)   TIA Gain: 0
[2020-05-02 16:10:46.633716] (info)   MIX Gain: 0
[2020-05-02 16:10:46.633726] (info)   VGA1 Gain: 0
[2020-05-02 16:10:46.633735] (info)   VGA2 Gain: 0
[2020-05-02 16:10:46.633745] (info)   Squelch: 0
[2020-05-02 16:10:46.633754] (info)   Idle Silence: false
[2020-05-02 16:10:46.633763] (info)   Digital Recorders: 4
[2020-05-02 16:10:46.633774] (info)   Debug Recorder: false
[2020-05-02 16:10:46.633782] (info)   SigMF Recorders: 0
[2020-05-02 16:10:46.633791] (info)   Analog Recorders: 0
[2020-05-02 16:10:46.633801] (info)   Modulation: qpsk
[2020-05-02 16:10:46.633811] (info)   Source Device: rtl=1
gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.11
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp
Using device #1 Realtek RTL2838UHIDIR SN: 00000001
Found Rafael Micro R820T tuner
[R82XX] PLL not locked!
[2020-05-02 16:10:47.229172] (info)   SOURCE TYPE OSMOSDR (osmosdr)
[2020-05-02 16:10:47.229211] (info)   Setting sample rate to: 2048000
[R82XX] PLL not locked!
[2020-05-02 16:10:47.348617] (info)   Actual sample rate: 2048000
[2020-05-02 16:10:47.348672] (info)   Tunning to 8.578010e+08
[2020-05-02 16:10:47.400211] (info)   Gain Stage: LNA supported values: 0 0.9 1.4 2.7 3.7 7.7 8.7 12.5 14.4 15.7 16.6 19.7 20.7 22.9 25.4 28 29.7 32.8 33.8 36.4 37.2 38.6 40.2 42.1 43.4 43.9 44.5 48 49.6
[2020-05-02 16:10:47.400253] (info)   Max Freqency: 8.587600e+08
[2020-05-02 16:10:47.400288] (info)   Min Freqency: 8.568400e+08
[2020-05-02 16:10:47.425998] (info)   Gain set to: 38.6
[2020-05-02 16:10:47.426051] (info)   Setting antenna to [RX]
Decim: 16 Decim2:  4
[2020-05-02 16:10:47.426504] (info)      P25 Recorder two-stage decimator - Initial decimated rate: 128000 Second decimated rate: 32000 FA: 6250 FB: 16000 System Rate: 2048000
[2020-05-02 16:10:47.426940] (error)   Failed parsing Config: std::exception `

The only two warnings during compile time:
`[ 62%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/main.cc.o
/home/jasonk/aprecorder/trunk-recorder/trunk-recorder/main.cc: In function ‘void unit_check()’:
/home/jasonk/aprecorder/trunk-recorder/trunk-recorder/main.cc:984:11: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
     system(shell_command);
     ~~~~~~^~~~~~~~~~~~~~~
[ 63%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/formatter.cc.o
[ 64%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/source.cc.o
[ 65%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/uploaders/uploader.cc.o
[ 66%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/uploaders/broadcastify_uploader.cc.o
[ 67%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/uploaders/openmhz_uploader.cc.o
[ 68%] Building CXX object CMakeFiles/recorder.dir/trunk-recorder/uploaders/call_uploader.cc.o
/home/jasonk/aprecorder/trunk-recorder/trunk-recorder/uploaders/call_uploader.cc: In function ‘void* upload_call_thread(void*)’:
/home/jasonk/aprecorder/trunk-recorder/trunk-recorder/uploaders/call_uploader.cc:28:9: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
   system(shell_command);
   ~~~~~~^~~~~~~~~~~~~~~
`

kcwebby, 2020-05-02:
Sorry left out the config.

`{
    "sources": [{
        "center": 857800000,
        "rate": 2048000,
        "error": 1000,
        "gain": 39,
        "debugRecorders": 0,
        "digitalRecorders": 4,
        "modulation": "qpsk",
        "driver": "osmosdr",
        "device": "rtl=1"
    }],
    "systems": [{
        "control_channels": [855462500],
        "type": "p25",
        "talkgroupsFile": "ChanList.csv",
            "shortName": "dcfems"
    }],
    "captureDir": "/home/luke/gnuradio/trunk-recorder/"
}
`

kcwebby, 2020-05-02:
This config was "fake", and not meant to represent the actual system... 
I did trace down the error to setting up the logs, and reading GNU config files.

apparently at some point, this system ran recorder (for the first time) as root(sudo), and setup some files/directories in the users folder that were owned by root.
Subsequent runs by my user, resulted in permissions errors trying to write the files, so a generic error was created.

This was fixed by removing ~/.gnuradio/ ~/gnupg ~/.gr_fftw_wisdom 

Rerunning recorder after this point allowed recorder (gnuradio) to recreate these files.   I suspect this error is coded deeper in GNURadio than Trunk-Recorder can handle, but i'll leave open for someone who knows better to review and close if the "generic error" cannot be improved.

robotastic, 2020-05-12:
That is really weird - I am not sure there is too much that can be done to improve the error. It really isn't even an error with config. It looks like it had to do with gnuradio, weird. Thanks for debugging

closed 2020-05-12 by robotastic